MichaIng/DietPi #247: DietPi-Software | Add Friendica: A Personal Social Networking Platform

Zone22, 2016-03-23:
[Friendica ](http://friendica.com/)is a communications platform for integrated social communications utilizing decentralized communications and linkage to several indie social projects - as well as popular mainstream providers.

"Currently, Friendica users can integrate contacts from Facebook, Twitter, Diaspora, GNU social, App.net, Pump.io and other services in their social streams. Communication is bi-directional wherever possible. There is also a bridge to include email contacts and RSS feeds. Additionally, connectors allow cross-posting to blog platforms like WordPress, Livejournal, Tumblr and Posterous. As the user account are spread over many servers, the addresses are composed of three parts, similar to email addresses, the user name, at "@" and the domain name of the Friendica instance." - Source wiki

[Installation](https://github.com/friendica/friendica/blob/master/INSTALL.txt)

Fourdee, 2016-03-25:
@Zone22 

Thanks for the request. I will look into this when I can.

xenfomation, 2016-05-09:
@Fourdee & @Zone22 --

I have an order of tasks I've been doing my best to contribute to, however I took 5 minutes to check out Friendica and it seems rather nice.  I have items for DietPi to correct pronto, so the point is I will circle back to this ASAP as it seems straight forward enough to test and discuss.

Zone22, 2016-05-09:
Excellent I look forward to it

xenfomation, 2016-05-17:
Still prototyping outside of testing. Manual install, size, and testing before I share the results with @Fourdee.

Thanks for the patience @zone22

closed 2017-12-30 by Fourdee