MichaIng/DietPi #248: C1+ | DietPi-System | CPU Govs | ondemand

k-plan, 2016-03-23:
@Fourdee 

thanks to C2 and RPi-Monitor, I have notices that on Odroid C1+ don't clock down in governor `ondemand ` .

Hardware Odroid C1+
Based on image: DietPi_v109_Odroid-C1-(Jessie) Version:

```
root@oDroide-C1:~# cat /DietPi/dietpi/.version
113
```
Installed Software: RAM-Log 2#, Open SSH, Pi-Hole, RPi-Monitor, LAMP
Kernel:
```
root@oDroide-C1:~# cat /proc/version
Linux version 3.10.80 (root@odroid-wheezy) (gcc version 4.7.2 (Debian 4.7.2-5) ) #1 SMP PREEMPT Sun Mar 13 15:50:53 CET 2016

root@oDroide-C1:~# uname -a
Linux oDroide-C1 3.10.80 #1 SMP PREEMPT Sun Mar 13 15:50:53 CET 2016 armv7l GNU/Linux
```

```
root@oDroide-C1:~# top
top - 23:17:16 up 2 days,  1:07,  1 user,  load average: 0,06, 0,09, 0,13
Tasks: 101 total,   2 running,  99 sleeping,   0 stopped,   0 zombie
%Cpu(s):  0,1 us,  0,0 sy,  0,0 ni, 99,9 id,  0,0 wa,  0,0 hi,  0,0 si,  0,0  st

root@oDroide-C1:~# cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors
hotplug interactive conservative ondemand performance

root@oDroide-C1:~# cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
ondemand
root@oDroide-C1:~# cat /sys/devices/system/cpu/cpu1/cpufreq/scaling_governor
ondemand
root@oDroide-C1:~# cat /sys/devices/system/cpu/cpu2/cpufreq/scaling_governor
ondemand
root@oDroide-C1:~# cat /sys/devices/system/cpu/cpu3/cpufreq/scaling_governor
ondemand

root@oDroide-C1:~# cat /sys/devices/system/cpu/cpu2/cpufreq/stats/time_in_state
96000 0
192000 0
312000 0
408000 0
504000 0
600000 0
696000 0
816000 0
912000 0
1008000 0
1104000 0
1200000 0
1296000 0
1416000 0
1488000 0
1536000 17516000
1632000 0
1728000 0

root@oDroide-C1:~# cat /sys/devices/system/cpu/cpu2/cpufreq/stats/total_trans
1
```
To test, it's a hardware issue, I changed governor `interactive ` via `dietpi-config`.

Results:
```
root@oDroid-C1_plus-test:~# cat /sys/devices/system/cpu/cpu2/cpufreq/scaling_governor
interactive

root@oDroid-C1_plus-test:~# cat /sys/devices/system/cpu/cpu2/cpufreq/stats/time_in_state
96000 34069
192000 37
312000 0
408000 0
504000 0
600000 0
696000 0
816000 0
912000 0
1008000 0
1104000 0
1200000 10
1296000 0
1416000 0
1488000 0
1536000 1692909
1632000 0
1728000 0

root@oDroid-C1_plus-test:~# cat /sys/devices/system/cpu/cpu2/cpufreq/stats/total_trans
44
```

As well tested with different Kernel and Firmware:

- DietPi_v106_Odroid-C1-(Jessie)
```
root@oDroid-C1_plus-test:~# cat /proc/version
Linux version 3.10.80 (root@odroid-wheezy) (gcc version 4.7.2 (Debian 4.7.2-5) ) #1 SMP PREEMPT Mon Nov 30 11:07:5
```

DietPi_v109_Odroid-C1-(Jessie) updated to v113:
```
root@oDroid-C1_plus-test:~# cat /proc/version
Linux version 3.10.80 (root@odroid-wheezy) (gcc version 4.7.2 (Debian 4.7.2-5) ) #1 SMP PREEMPT Sun Mar 13 15:50:53 CET 2016
```

and  tested  v113 with `apt-get dist-upgrade` and the new firmware `/boot/uImage-3.10.96`
```
root@oDroid-C1_plus-test:~# cat /proc/version
Linux version 3.10.96 (root@odroid-wheezy) (gcc version 4.7.2 (Debian 4.7.2-5) ) #1 SMP PREEMPT Mon Mar 21 07:50:12 CET 2016
```

Steady behaviour, no frequency step-down with governor `ondemand ` .

Fourdee, 2016-03-23:
@k-plan 
Excellent bug reporting :+1:

I'll spin up my C1 (not plus) and check if the same occurs.

Fourdee, 2016-03-25:
@k-plan 
Confirmed with Odroid C1. I'll pass this info onto Meveric as he maintains the Kernel.

k-plan, 2016-03-25:
@Fourdee 
thanks a lot. Looking forward to Meveric  reply. Only hint I have found so far:

- http://forum.odroid.com/viewtopic.php?f=112&t=18461

- http://forum.odroid.com/viewtopic.php?f=117&t=12057  ( :smiley: )

Fourdee, 2016-03-25:
@k-plan 

Reply
> That's a known issue. I already pointed out to others, that only the conservative governor seems to work

So its a known issue.

Interactive seems to be working, so might be a better alternative to conservative.  Conservative has a noticeable delay before scaling up.
```
root@DietPi:~# cat /sys/devices/system/cpu/cpu2/cpufreq/stats/time_in_state
96000 1409
192000 14
312000 0
408000 0
504000 0
600000 7
696000 0
816000 0
912000 0
1008000 0
1104000 0
1200000 0
1296000 0
1416000 0
1488000 0
1536000 2706
1632000 0
1728000 0
```

Fourdee, 2016-03-26:
@k-plan 
I'am going to make a patch for V114 and C1.

This will change the default CPU gov to interactive. I will also disable the ondemand option in dietpi-config.

Fourdee, 2016-03-26:
@k-plan 
Done. Patched for v114.

> General (C1) | CPU Govs: Ondemand is now disabled due to it failing to scale down frequencies. This is a known issue with the C1 kernel. Interactive is the default CPU gov and will replace ondemand. Many thanks to K-Plan for reporting this bug: https://github.com/Fourdee/DietPi/issues/248

closed 2016-03-26 by Fourdee

k-plan, 2018-01-16:
@Fourdee 

> Done. Patched for v114.

Want to check, if we can for DietPi v6 undo this patch.
https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-config#L1868-L1873
https://github.com/Fourdee/DietPi/issues/1385#issuecomment-357766293

So I have repeat this test.

Hardware Odroid C1+
Based on image: DietPi_v6.-1_Odroid-C1-(Jessie) Version:

```
root@Odroid-C1-v6:~# cat /DietPi/dietpi/.version
6
-1

root@Odroid-C1-v6:~# cat /etc/debian_version
8.10
```

Kernel:

```
root@Odroid-C1-v6:~# cat /proc/version
Linux version 3.10.107 (root@odroid-wheezy) (gcc version 4.7.2 (Debian 4.7.2-5) ) #1 SMP PREEMPT Wed Sep 13 18:48:10 CEST 2017

root@Odroid-C1-v6:~# uname -a
Linux Odroid-C1-v6 3.10.107 #1 SMP PREEMPT Wed Sep 13 18:48:10 CEST 2017 armv7l GNU/Linux
```

```
root@Odroid-C1-v6:~# top
top - 14:10:39 up 8 min,  1 user,  load average: 0.00, 0.02, 0.03
Tasks:  87 total,   2 running,  85 sleeping,   0 stopped,   0 zombie
%Cpu(s):  0.2 us,  0.3 sy,  0.0 ni, 99.5 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st

root@Odroid-C1-v6:~# cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors
hotplug interactive conservative ondemand performance

root@Odroid-C1-v6:~# cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
ondemand
root@Odroid-C1-v6:~# cat /sys/devices/system/cpu/cpu1/cpufreq/scaling_governor
ondemand
root@Odroid-C1-v6:~# cat /sys/devices/system/cpu/cpu2/cpufreq/scaling_governor
ondemand
root@Odroid-C1-v6:~# cat /sys/devices/system/cpu/cpu3/cpufreq/scaling_governor
ondemand

root@Odroid-C1-v6:~# cat /sys/devices/system/cpu/cpu0/cpufreq/stats/time_in_state
96000 68137
192000 0
312000 3
408000 3
504000 5
600000 4
696000 4
816000 5
912000 0
1008000 0
1104000 0
1200000 0
1296000 2
1416000 0
1488000 7
1536000 5955
1632000 0
1728000 0
1824000 0

root@Odroid-C1-v6:~# cat /sys/devices/system/cpu/cpu2/cpufreq/stats/time_in_state
96000 62450
192000 0
312000 3
408000 3
504000 5
600000 4
696000 4
816000 5
912000 0
1008000 0
1104000 0
1200000 0
1296000 2
1416000 0
1488000 7
1536000 5678
1632000 0
1728000 0
1824000 0

root@Odroid-C1-v6:~# cat /sys/devices/system/cpu/cpu0/cpufreq/stats/total_trans
102
root@Odroid-C1-v6:~# cat /sys/devices/system/cpu/cpu2/cpufreq/stats/total_trans
104

```

> This will change the default CPU gov to interactive from on demand (if in use). I will also disable the ondemand option in dietpi-config.

I think, with Kernel 3.10.107, we can revert in DietPi v6 to CPU gov `ondemand` for all devices and enable this option back in `dietpi-config` for Odroid C1.