MichaIng/DietPi #249: RPi Cam Autostarts when installed disabling normal Pi Camera function

JohnAllen, 2016-03-25:
Just wanted to say thank you for DietPi and also that it would be great to get some notification that installing DietPiCam (or what it was exactly called) prevents the normal function of the `raspi*` programs.

Fourdee, 2016-03-25:
@JohnAllen 
Hi John,

Thanks for the feedback.

DietPiCam is essentially https://github.com/silvanmelchior/RPi_Cam_Web_Interface thats been modified for DietPi. This includes support for optional USB drives, and works with our optimized LASP stack.
In regards to the RPi camera and DietPiCam. Are you referring to the autostart of the camera during boot which prevents other programs using it? Or something else?

JohnAllen, 2016-03-25:
When RPi Cam Web is installed, it autoruns, autostarts, which wasn't obvious to me.  When I went to use the Pi camera as usual it didn't work because there was already the RPi Cam Web using it.  It just wasn't obvoius that that would occur when I installed that software, so it would have been great to know that it would autorun and disable stuff like `raspistill`, etc.

JohnAllen, 2016-03-25:
Just a warning somewhere maybe, or not autostarting that for users like me who thought it sounded cool, but didn't necessarily want it to autorun the second I got DietPi configured and up and running.

Fourdee, 2016-03-25:
@JohnAllen 
Thanks John. 
I'll add a whiptail message that will warn the user after selecting DietPiCam for install.

Fourdee, 2016-03-26:
@JohnAllen 
Completed.
Starting from v114, this notification will show when the user selects DietPiCam for installation:
[image]

You can also use the following from terminal.
```
/etc/dietpi/dietpi-software/services/raspimjpeg.service stop
/etc/dietpi/dietpi-software/services/raspimjpeg.service start
```

closed 2016-03-26 by Fourdee

JohnAllen, 2016-03-26:
Wow, awesome.  Excellent work.  That's perfect.