MichaIng/DietPi #250: DietPi-Software | Add Huginn

Zone22, 2016-03-25:
[Huginn](https://github.com/cantino/huginn) is a system for building agents that perform automated tasks for you online. They can read the web, watch for events, and take actions on your behalf. Huginn's Agents create and consume events, propagating them along a directed graph. Think of it as a hackable Yahoo! Pipes plus IFTTT on your own server. You always know who has your data. You do.

[Instructions ](https://github.com/cantino/huginn/wiki/Running-Huginn-on-minimal-systems-with-low-RAM-&-CPU-e.g.-Raspberry-Pi)for installing and optimizing Huginn with it's required dependencies for Raspberry Pi.
These instructions were written for Pi - model B - revision 2 - 256 MB RAM and may need to be tweaked to account for higher-end hardware.

Fourdee, 2016-03-25:
@Zone22 
Thanks for the request, I will look into this when I can.

xenfomation, 2016-05-19:
Will be taking a look as there are many considerations, but I am all about automation.

Zone22, 2016-06-05:
@xenfomation 
I know you're extremely busy and have a lot on your plate. Although I'm curious as to what you think of Huginn. What attracts me to Huginn is its versatility and it's a completely self contained platform. What other options are there to consider that are similar and self hosted?

Zone22, 2016-11-21:
@Fourdee 
Maybe for 139?

For compiling Arm 64
https://github.com/cantino/huginn/issues/1586

closed 2017-12-30 by Fourdee