MichaIng/DietPi #251: DietPi-Software | Add Mumble - VOIP Server

Zone22, 2016-03-25:
Mumble is an open source voice over IP application primarily designed for use by gamers, similar to programs such as TeamSpeak and Ventrilo.

[Install Guide](http://pimylifeup.com/raspberry-pi-mumble-server/)

Fourdee, 2016-03-25:
@Zone22 
Thanks for the request, I will look into this when I can.

oshank, 2016-03-26:
I am currently running a mumble server on my B+ and its works great.

Zone22, 2016-03-26:
Yes it's trivial to install but it's nice to have things going through the dietpi software.

That being said due to the simplicity of installing Mumble I might fork and do a Pull request to included it into dietpi. It'll take a bit of work for me to digest how dietpi installs everything. We will see!

oshank, 2016-03-26:
I'm trying to find some time and patience to do the same thing. I really want to contribute to this awesome project with more then some Bitcoin.

Fourdee, 2016-03-26:
@Zone22 @oshank 
I've not documented on how to code in additional software into DietPi-Software. So if you need some pointers or have any issues, let me know.

Fourdee, 2016-04-26:
> https://www.digitalocean.com/community/tutorials/how-to-install-and-configure-mumble-server-murmur-on-ubuntu-14-04

xenfomation, 2016-04-28:
i will update my clone'd test repo AS this seems similar to the mc, mtrtiny, vim-tiny package discussion in a previous issue... i have some ideas, but will compose them before 09:00 EST 28-APRIL-2016 to see if myopia can be turned into one of the cool DietPi menus for a "most righteous means to install favorite software", etc... hrm...

Fourdee, 2016-04-28:
@xenfomation 
Sounds good, looking forward to your ideas :+1:

Fourdee, 2016-04-28:
Done, available for v117. http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=1691#p1691

closed 2016-04-28 by Fourdee