MichaIng/DietPi #252: DietPi-Software | Pi-Hole - Web UI is missing elements.

Zone22, 2016-03-25:
I noticed one reading up on how to configure Pi-Hole that the Web UI is missing the 'Update' and 'Restart' options. This can be observed when running DietP on xu4.

[image]
[image]

Fourdee, 2016-03-25:
@Zone22 
I believe they removed these options from the web UI. I cant see any references to "restart" in the index.php code. But would need to verify with the devs.

The current screenshot on their Git page look like this:
https://camo.githubusercontent.com/efe1b63217fee40926b01703710308fe0f679680/687474703a2f2f692e696d6775722e636f6d2f467765416f33362e706e67

k-plan, 2016-03-25:
Hi,

screenshot was linked in  README.md until tow day ago.  https://goo.gl/1400Fh
http://i.imgur.com/x2iMfoc.png  -  but  shows not the `branch  'master'.`

The current screenshot on their Git page shows `branch 'devel'` , **_not_** `branch 'master' .`

`Whitelist` and `Blacklist` are disabled at the moment in `branch 'master' .`

```
<!-- Query Log -->
                <li>
                    <a href="queries.php">
                        <i class="fa fa-file-text-o"></i> <span>Query Log</span>
                    </a>
                </li>
                <!-- Whitelist -->
                <!--
                <li>
                    <a href="list.php?l=white">
                        <i class="fa fa-pencil-square-o"></i> <span>Whitelist</span>
                    </a>
                </li>
                -->
                <!-- Blacklist -->
                <!--
                <li>
                    <a href="list.php?l=black">
                        <i class="fa fa-pencil-square-o"></i> <span>Blacklist</span>
                    </a>
                </li>
                -->
                <!-- Donate -->
```

Zone22, 2016-03-26:
@Fourdee
Screenshot as requested
[image] 

@k-plan 
I agree with the auto update and the prompting with DNS if it's not too much trouble.

Fourdee, 2016-03-26:
@k-plan @Zone22 

> Will be a nice add-on, installation prompt user for his preferred DNS server (not use google DNS by default)

This may be better integrated into ```dietpi-config``` under networking options. As well as allowing the user to set DNS servers manually, maybe a menu list and selection of various DNS servers?

> gravity.sh run automatically once after Pi-hole installation. It will be useful, to add this automatically to /etc/cron.daily/ (with a update log file in: /var/log/pihole-update.log) or prompt user for this.

I'am not personally keen on automating this. I've ran ```gravity.sh``` in the past and due to some of the advert block list servers being offline, it reduced the total blocked domains by 80%. I'd like to leave this down to the user which must be run manually.

Although, I can create a cronjob that will update ```gravity.sh``` if you would find it useful? I can also add that to the online documentation for users.

k-plan, 2016-03-26:
@Fourdee 

> This may be better integrated into dietpi-config under networking options. 

It's okay for me. Pihole DNS is using:   ` /etc/dnsmasq.conf`
But nothing to do with DNS settings in:  ` /etc/network/interfaces`
So please, if possible, only show this entry, if Pihole is installed.

> As well as allowing the user to set DNS servers manually, maybe a menu list and selection of various DNS servers?

Great idea.

> I'am not personally keen on automating this. I've ran gravity.sh in the past and due to some of the advert block list servers being offline, it reduced the total blocked domains by 80%.

Hmm, good spot. If you have enough listed, it will be equal. Duplicated url's will sorted out by pihole.
I personally use this as kind of "_fall-back_":  https://github.com/StevenBlack/hosts
Because sometimes things went wrong, I want to have a update log e.g. in:  `/var/log/pihole-update.log` 

> Although, I can create a cronjob that will update gravity.sh if you would find it useful?

With  `gravity.sh` output logging? Yes :+1: 
If someone has problem with updates, he can comment it out. Quite simply as write a new cron script by itself.

> I can also add that to the online documentation for PiHole users.

Great idea. Please do so.

Fourdee, 2016-04-10:
@k-plan
Implemented this into ```dietpi-config``` when selecting static DNS server. Will see about applying to PiHole.
[image]

k-plan, 2016-04-10:
@Fourdee 
have it tested. Really cool implement and a nice feature. Good to see used IPs and provider.

> Will see about applying to PiHole.

I'm curious about it. 😄

Fourdee, 2016-04-19:
I will leave the PiHole implementation out for now. I may revisit/rethink that in the future on a separate ticket.

Marked as closed, please reopen if required.

closed 2016-04-19 by Fourdee