MichaIng/DietPi #254: DietPi-Software | Add optimized installation for Tor + Wifi hotspot

Fourdee, 2016-03-26:
Route all traffic through tor, via WiFi hotspot on device.

Fourdee, 2016-03-27:
Completed. Available as an optional extra for WiFi HotSpot installation. Selecting Tor will also flag WiFi Hotspot for install.
[image]

Fourdee, 2016-03-27:
http://fuzon.co.uk/phpbb/viewtopic.php?f=8&t=5&p=1529#p1529

closed 2016-03-27 by Fourdee

msdos, 2017-04-20:
The link above is broken, had to check https://web.archive.org/web/20160831163959/http://fuzon.co.uk/phpbb/viewtopic.php?f=8&t=5&p=1529

Fourdee, 2017-04-20:
@msdos 

> The link above is broken, had to check 

Yep, we changed our domain name a while back. Fuzon.co.uk redirect was removed a few months ago. New link is:
http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=1529#p1529

msdos, 2017-04-27:
If fuzon.co.uk is still in your hands, it would be nice to redirect to dietpi.com (since the other parameters are the same it wouldn't be difficult).

If you don't want to mess up with `mod_rewrite` and such, you can use cloudflare in this domain and use their redirect feature - it's free and it works flawlessly.

https://support.cloudflare.com/hc/en-us/articles/200172286-How-do-I-perform-URL-forwarding-or-redirects-with-Cloudflare-

Fourdee, 2017-04-27:
> If fuzon.co.uk is still in your hands, it would be nice to redirect to dietpi.com (since the other parameters are the same it wouldn't be difficult).

@msdos 

fuzon.co.uk is now in use on another server, EU Odroid apt mirror for Meveric's repo:
http://fuzon.co.uk/meveric/

msdos, 2017-05-27:
Is the wifi connection `DietPi-HotSpot` hidden? If I go to `dietpi-config` it's there, but devices can't see it.

Fourdee, 2017-05-28:
@msdos 

> Is the wifi connection DietPi-HotSpot hidden? 

Not hidden, SSID is broadcasted.

Check the service/adapter for any reported issues:
```
systemctl status hostapd.service -l
iwconfig
```

msdos, 2017-05-28:
```
systemctl status hostapd.service -l
● hostapd.service - LSB: Advanced IEEE 802.11 management daemon
   Loaded: loaded (/etc/init.d/hostapd)
   Active: active (exited) since Sat 2017-05-27 22:17:24 UTC; 18h ago
  Process: 1158 ExecStart=/etc/init.d/hostapd start (code=exited, status=0/SUCCESS)

May 27 22:17:24 DietPi hostapd[1158]: Starting advanced IEEE 802.11 management: hostapd failed!
May 27 22:17:24 DietPi systemd[1]: Started LSB: Advanced IEEE 802.11 management daemon.
root@DietPi:~# iwconfig
wlan0     unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Auto  Frequency=2.412 GHz  Access Point: Not-Associated
          Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

eth0      no wireless extensions.

lo        no wireless extensions.
```

**May 27 22:17:24 DietPi hostapd[1158]: Starting advanced IEEE 802.11 management: hostapd failed!**

So if I try to run `sudo hostapd -d /etc/hostapd/hostapd.conf`:

```
root@DietPi:~# sudo hostapd -d /etc/hostapd/hostapd.conf
random: Trying to read entropy from /dev/random
Configuration file: /etc/hostapd/hostapd.conf
nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211: Remove monitor interface: refcount=0
netlink: Operstate: ifindex=3 linkmode=0 (kernel-control), operstate=6 (IF_OPER_UP)
nl80211: Set mode ifindex 3 iftype 2 (STATION)
nl80211: Failed to set interface 3 to mode 2: -19 (No such device)
nl80211 driver initialization failed.
hostapd_interface_deinit_free(0x76b2d8)
hostapd_interface_deinit_free: num_bss=1 conf->num_bss=1
hostapd_interface_deinit(0x76b2d8)
wlan0: interface state UNINITIALIZED->DISABLED
hostapd_bss_deinit: deinit bss wlan0
wlan0: AP-DISABLED
hostapd_cleanup(hapd=0x76be98 (wlan0))
hostapd_free_hapd_data: Interface wlan0 wasn't started
hostapd_interface_deinit_free: driver=(nil) drv_priv=(nil) -> hapd_deinit
hostapd_interface_free(0x76b2d8)
hostapd_interface_free: free hapd 0x76be98
hostapd_cleanup_iface(0x76b2d8)
hostapd_cleanup_iface_partial(0x76b2d8)
hostapd_cleanup_iface: free iface=0x76b2d8
```

**nl80211: Driver does not support authentication/association or connect commands**

`lsusb`

```
Bus 001 Device 004: ID 0bda:8179 Realtek Semiconductor Corp.
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9512 Standard Microsystems Corp. LAN9500 Ethernet 10/100 Adapter / SMSC9512/9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

I'm using a fresh installed DietPi, with no custom drivers. This is a raspberry 1 model with a wifi dongle. Should I be installing Realtek drivers?

Fourdee, 2017-05-29:
@msdos 

> Should I be installing Realtek drivers?

> nl80211: Failed to set interface 3 to mode 2: -19 (No such device)
nl80211 driver initialization failed.

Yep, try:
```
sed -i "/^driver=/c\driver=rtl871xdrv" /etc/hostapd/hostapd.conf
dietpi-services restart
```

If that still fails, you may need to install our compiled hostapd with support for the ```RT8188C ``` device (might be same for 8188eu). We check for this during install but only for ```RT8188C ``` :
https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software#L5003-L5008

msdos, 2017-06-14:
So, correct me if I'm wrong. From 5003-5008 you mean this:

https://github.com/Fourdee/DietPi/blob/6214cd915ef59399b214c039a643b8d045ad0856/dietpi/dietpi-software#L5003-L5008

(It's a good practice to have links that always point to a specific commit. :) Usually I click on master, then "history" in my file then "browse this repository at this point in history", so if you change the file in master the lines will always point to the correct section since the file can change)

So, I remove the condition, and `WIFIHOTSPOT_RTL8188C_DEVICE=1` will always be defined, and that's the behavior we're looking for, right? By the way, that's what I did and it's working now, thanks for the amazing help!