MichaIng/DietPi #255: Crash when starting tightvncserver on Odroid C2

djix123, 2016-03-26:
Tightvncserver crashes on Odroid C2 at start up
(this is already fixed in the Hardkernel Ubuntu 16.04 stack)

Fourdee, 2016-03-27:
```
*** Error in `Xtightvnc': free(): invalid next size (fast): 0x0000000000581a90 *    **
```
apt-get install xfonts-scaleable:
```
*** Error in `Xtightvnc': malloc(): memory corruption: 0x0000000000581a50 ***
```

@djix123 
Can you provide me with the kernel version for your Ubuntu 16.04 install please:
```uname -a```

djix123, 2016-03-27:
I'm using a kernel built following HK's wiki page instructions - built yesterday.

Fourdee, 2016-03-27:
@djix123
https://github.com/hardkernel/linux/tree/odroidc2-3.14.y ?

I'lll compile and see if it resolves the issue.

djix123, 2016-03-27:
Yes, that looks right ... this is what I used to clone it, before building:

git clone --depth 1 https://github.com/hardkernel/linux.git -b odroidc2-3.14.y

Fourdee, 2016-03-27:
@djix123 
Thanks.

Same issue with latest kernel:
```
root@DietPi:~# uname -a
Linux DietPi 3.14.29 #1 SMP PREEMPT Sun Mar 27 19:32:15 BST 2016 aarch64 GNU/Linux
```

```
root@DietPi:~# vncserver :0 -geometry 1024x600 -depth 24
*** Error in `Xtightvnc': malloc(): memory corruption: 0x0000000000581a50 ***
Couldn't start Xtightvnc; trying default font path.
Please set correct fontPath in the vncserver script.
*** Error in `Xtightvnc': malloc(): memory corruption: 0x0000000000581b30 ***
Couldn't start Xtightvnc process.
```

Really strange error, memory corruption, not helpful lol. Might be something simple like missing a package. I'll look into it.

Fourdee, 2016-03-27:
```/usr/bin/tightvncserver```
```
$fontPath = "/usr/share/fonts/X11/misc/";
```
Still gives same error.

djix123, 2016-03-27:
Here the original Ubuntu bug report ...

https://bugs.launchpad.net/ubuntu/+source/tightvnc/+bug/1429198

Fourdee, 2016-03-27:
@djix123 

> Here the original Ubuntu bug report ...

Excellent. Now it all makes sense :+1:. A bug in Tightvnc.

Jessie installs:
```
1.3.9-6.5
```

As this hasn't reached Sid/Jessie yet (https://packages.debian.org/sid/x11/tightvncserver), I'll try the 1.3.10-0ubuntu3.deb from ubuntu.

Fourdee, 2016-03-28:
@djix123 

Missing packages:
```
root@DietPi:~# dpkg -i tightvncserver_1.3.10-0ubuntu3_arm64.deb
Selecting previously unselected package tightvncserver.
(Reading database ... 53663 files and directories currently installed.)
Preparing to unpack tightvncserver_1.3.10-0ubuntu3_arm64.deb ...
Unpacking tightvncserver (1.3.10-0ubuntu3) ...
dpkg: dependency problems prevent configuration of tightvncserver:
 tightvncserver depends on libjpeg8 (>= 8c); however:
  Package libjpeg8 is not installed.
```

Attempted to copy the binary from .deb. Installed libjpeg-dev + symlink,
```
root@DietPi:~# vncserver :0 -geometry 1024x600 -depth 24
Xtightvnc: /usr/lib/aarch64-linux-gnu/libjpeg.so.8: version `LIBJPEG_8.0' not fo    und (required by Xtightvnc)
```
```
root@DietPi:~# ls -lha /usr/lib/aarch64-linux-gnu/libjpeg.so.8
lrwxrwxrwx 1 root root 37 Mar 28 14:54 /usr/lib/aarch64-linux-gnu/libjpeg.so.8 -> /usr/lib/aarch64-linux-gnu/libjpeg.so
```

Looking like a nofix.

djix123, 2016-03-28:
ok, I got it working ... I needed to install the following:

libc6_2.23-0ubuntu2_arm64.deb
libc-bin_2.23-0ubuntu2_arm64.deb
locales_2.23-0ubuntu2_all.deb
libjpeg8_8c-2ubuntu8_arm64.deb
libjpeg-turbo8_1.3.0-0ubuntu1.1_arm64.deb
tightvncserver_1.3.10-0ubuntu3_arm64.deb

... and now I have a running VNC server!

Fourdee, 2016-03-28:
@djix123 
Excellent. 
The only issue is trying to install all those Ubuntu packages on Debian Jessie. Although in theory the Ubuntu packages should work, the issue is package dependencies. I'll take another look at this tomorrow and see what's needed to get it running.

rhkean, 2016-05-17:
yeah, I agree with @Fourdee ... too many ubuntu packages.
I've confirmed this to be a problem on the Pine A64 board, also.
I think we should just install the vnc4server on arm64 systems until debian gets tightvnc_1.3.10

xenfomation, 2016-05-17:
Agreed. Even in x86 x64 Debian, I have encountered this with more and more packages (deps failing, malloc, and more).  ARGH!

rhkean, 2016-05-18:
@Fourdee 
Can you test with my testing branch on the Odroid C2.
(I got lost in the pine64 and forgot to order mine.  it should be here in a couple days, though)

Anyhow, can you test my commits into the rhkean testing branch for vnc4server on the arm64.

It starts the display on :1 instead of :0

but, should do until tightvnc on arm64 gets fixed

Fourdee, 2016-05-18:
@rhkean Yep, installing now, will post results a bit later

Fourdee, 2016-05-18:
@rhkean 
Yep, installed and running fine:
[image]

We need to 
- get this running on ```:0``` if possible
- autostart support in ```dietpi-autostart```. https://github.com/Fourdee/DietPi/blob/testing/dietpi/login#L99-L111, should just be a case of detecting ```if HW_MODEL 12``` or ``` dpkg -l | grep -ci -m1 vnc4server```, then adding the command/service to launch it.

 I'll take a look at both.

rhkean, 2016-05-18:
excellent....

Let me know if you're ready for me to issue a pull request

Fourdee, 2016-05-18:
@rhkean 

Yep, send the pull request now if you like.

> Note to self, 
during conf stage:
```
systemctl disable vncserver.service
```
During ```login```:
```vncserver start```

rhkean, 2016-05-18:
you know I added a systemd file, right?

it runs automatically

Fourdee, 2016-05-18:
@rhkean 
I just want to be consistent with the other installation (tightvnc): http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=30#p408

Because the documentation and how it runs will be different for both, I may separate the installation into tightvnc and vnc4server.

rhkean, 2016-05-18:
makes sense....

Your call.  Feel free to "fix" my code, too.  It's your creation.

I didn't spend much time delving into how you had it configured.  I should
have so that it was consistent.

Fourdee, 2016-05-18:
@rhkean 
Not your fault. Just me brainstorming :+1: 

Its not a case of fixing code, just trying to improve it. There is alot of old DietPi in there that needs improvement and/or complete rewrite. Sometimes, its only when we update specific areas that I get a chance to have a good think about that code :)

I personally feel splitting the installation is the best call. From past experience, users will always prefer the ability to choose, if you give the option to them. So in this case, vnc4server will be available for all systems, tightvnc for non arm64 systems.

Fourdee, 2016-05-22:
@rhkean @xenfomation 
Any ideas how to write to a file, that has variables, but instead of writing the variable value, write the text as is, eg:
```
echo -e "$MYVAR" > file

#I want to be able to
#cat file and have it contain
$MYVAR
```

Fourdee, 2016-05-23:
Ok, few changes to the install code, this is for both ```TIGHTVNCSERVER``` and ```VNC4SERVER```:
- VNC servers will start automatically during boot. I really liked this Rob :+1: 
- Service type is ```idle```. Similar to "$all" in LSB. Basically runs last during boot.
- VNC will start using these settings in dietpi.txt (users can change them, reboot to apply):
- Default screen is now ```:0```.
```
#VNC Server Options
dietpi_vncserver_width=1280
dietpi_vncserver_height=720
dietpi_vncserver_depth=16
dietpi_vncserver_display=0
```

I've tested, confirmed working:
C2 + VNC4SERVER
RPi3 + TIGHTVNCSERVER

I will make a note to test VNC4SERVER on VMs.

####To Do:
- Create LSB service for wheezy
- Patch existing installations with the new service
- Resolve ```no session id for pid``` during 1st vnc client connect

@rhkean 
Apologies for going over your original code, it wasn't my intention to change your work. I'll sync the code and link commit when its done.

rhkean, 2016-05-23:
@Fourdee 

1st, That's awesome!

and 2nd... Please don't apologize for going over my code....  working together makes a better product.

Fourdee, 2016-05-23:
@rhkean 
Excellent :+1: 

All done, last commit (I did a few during testing): https://github.com/Fourdee/DietPi/commit/f276539b76d6cbe661520878358afc9d235c9880

rhkean, 2016-05-23:
Totally lazy question (I could go look in the code)

Did you remove VNC as a AutoStart/AutoBoot option (since it is starting up automatically now)?

Fourdee, 2016-05-23:
@rhkean 
Yep, it used to be 6: https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-autostart#L85-L93

Patch to reset 6 to 0: https://github.com/Fourdee/DietPi/blob/testing/dietpi/patch_file#L794-L798

Fourdee, 2016-05-23:
Marking this as closed. Great work Rob :+1:

closed 2016-05-23 by Fourdee