MichaIng/DietPi #256: DietPi-Software | EmonPi - Add optional installation for RF features.

Fourdee, 2016-03-28:
> note to self:
Buy RF device (eg: temp sensor module, emonTX etc) so I can impliment/test.

Fourdee, 2016-03-28:
Continuation from https://github.com/Fourdee/DietPi/issues/210.

WolfganP, 2016-03-28:
Just for reference related to EmonPi alternate comms, the system also supports wired serial connection as per https://openenergymonitor.org/emon/node/3884 (so cheap TTL/RS232 or RS485 adapters can be used were wiring isn't an issue)

Fourdee, 2016-03-28:
@WolfganP 
Interesting, i've got two of those serial adapters at the moment but no emonTX, wonder if they can be used with the emonPi?

I've ordered a EmonTH so I can implement RF, and have some fun checking temp/humidity around the house lol :)

Fourdee, 2016-04-02:
RF receiver features (EmonTH/EmonTX) is already supported by our installation. RF transmission is untested.

closed 2016-04-19 by Fourdee