MichaIng/DietPi #257: DietPi-Software | EmonPi - Add optional installation for local EmonCMS on device

Fourdee, 2016-03-28:


Fourdee, 2016-04-02:
As our current EmonPi installation uses emoncms.org, and is extremely lightweight because of this, local EmonCMS is low priority.

xenfomation, 2016-04-30:
I have to update another case regarding encryption as it led to a fresh install.

I am taking a moment to refresh my DietPi (sorry, I just consider my PI DietPi since it is the only thing worthy enough to use!)

I will see if I can inject this in, test its installation, and make any notes for menu/CLI-based items.

xenfomation, 2016-05-02:
I see now why this is low per [https://github.com/emoncms/emoncms/blob/master/docs/RaspberryPi/readme.md](https://github.com/emoncms/emoncms/blob/master/docs/RaspberryPi/readme.md)

The wear and tear on the SD card is well documented.  IMHO, this would require a check for the use of a USB drive as well as ensuring an interactive warning menu/config screen would follow as to assist the user with USB drive configuration and Emoncms stable deployment.

Fourdee, 2016-05-11:
As per https://github.com/Fourdee/DietPi/issues/257#issuecomment-204808410, and https://github.com/Fourdee/DietPi/issues/257#issuecomment-216098099. 

A local CMS install only generates negatives:
- Excessive SD IO usage
- Requires webserver installation

As such, i've decided to close this ticket. Our current installation for EmonPi offers the lightest possible installation and works exactly the same using emoncms.org, instead of a local CMS install.

closed 2016-05-11 by Fourdee

R1CH-T, 2016-05-11:
Thanks for looking into this, guys.

As I have stated before, Open Energy Monitor utilises a low-write filesystem for this purpose, which is interesting in itself and could be extremely useful for various database applications, but I don't have a clue as to how this is implemented.

Xenfomation: you are right about the complexities of this stuff and your input is very much appreciated.
I own the hardware and if I can help at all, please let me know.
For more information about my interest in this and my ideas, please check out https://github.com/Fourdee/DietPi/issues/210

R1CH-T, 2016-05-11:
Oops.
Fourdee beat me to it, I think...
Thanks for your efforts, anyway.

Fourdee, 2016-05-11:
@R1CH-T

Only by 4 seconds lol.

Yeah sorry about this Rich. But a full blown EmonCMS installation just isn't worth it at the moment. Although offsite, http://emoncms.org offers the same viewable data, without the negatives of a local CMS installation.

R1CH-T, 2016-05-11:
No sweat!

Fourdee, 2016-05-11:
> As I have stated before, Open Energy Monitor utilises a low-write filesystem for this purpose, which is interesting in itself and could be extremely useful for various database applications, but I don't have a clue as to how this is implemented.

I'd imagine they are reducing the total times they write data to disk. Instead, opting for large data writes to disk at a lower frequency. I believe this prevents writing 4K blocks everytime < 4K of data needs to be saved, but it still needs to be saved at some point.

```
root@DietPi:~# blockdev --getbsz /dev/mmcblk0p2
4096
```

So lets say you have a file that's 1K in size. When that file is saved to disk, I believe it will write 4K to disk (the whole block size). The file will also use 4K of space on the disk (the whole block its contained in).

MichaIng, 2021-05-10:
Hey guys, just in case someone using EmonPi is subscribed to this issue:
- I'm switching our emonHub implementation back to the upstream branch, with years of new features and bugfixes. It meanwhile allows to use any config directory: https://github.com/MichaIng/DietPi/pull/4357
- I recognised that data can be sent via MQTT (Mosquitto) as well, additionally or as alternative to emonCMS. And we provide a Mosquitto install option as well. Shall we install Mosquitto as a dependency together with emonCMS or at least give users the choice when emonHub is selected for install?
- And I'm rethinking to add the emonCMS install option. There are other reasons why users might have a webserver installed already, so that it's no much overhead anymore. The regular disk writes should be solved via Redis RAM-based database, if I'm not mistaken, which is available via dietpi-software as well.

Last but not least, if someone is able to test the change (above linked pull request) with an actual emonPi, that would be great. Install process and service starts fine, but without the hardware, I cannot test data collection and emoncmg.org forwarding connection itself.