MichaIng/DietPi #259: DietPi-Software | Kodi install as an unprivileged user

Zone22, 2016-03-28:
All the programs installed by DietPi Software are installed by root with the exceptions of suff that runs on top of Apache. Although It's best practice to install programs as an unprivileged user some exceptions can be made.  Even given what I've said thus far I cannot stress enough that Kodi should be installed as an unprivileged user. Kodi has ability to use powerful add-ons and plug-ins. Kodi runs as root and so do it add-ons and plug-ins.  This presents a significant security risk for anyone running Kodi along with any server software containing sensitive or personal information. 

It's not enough to say simply install plug-ins that are trusted when self posting your personal data. I know this request would complicates file permissions and the complexity of DietPi scripts.

Fourdee, 2016-03-28:
@Zone22 

> It's not enough to say simply install plug-ins that are trusted. I know this request would complicates file permissions and the complexity of DietPi scripts.

You are correct, this would vastly increase dev time, but this is not the reason why we only use root (minus www-data).

You are probably not going to like my answer, but I want to be honest:
Users who are security conscience, will always apply their security needs to the OS, regardless of what DietPi does. The whole "you shouldn't run as root" is non-sense, more often than not, commented by users who simply dont understand that running sudo before every command is exactly the same as running as root. You are more likley to do damage by running ```sudo install.sh``` than by a Kodi addon.

Kodi and Linux are completely opensource. So if any "ill intention" code exists in any project (or Kodi addon), the whole world will be able to see it and it doesn't take long for the word to get around.

I am not willing to bubble wrap Kodi (or even DietPi) for our users. If a user has a security requirement or concerns of what certain software "might, could" and should not do, they are more than capable of applying a level of security they require.

Please dont take the above personally, its only my opinion.

Zone22, 2016-03-28:
I appreciate you being upfront with your opinion and striving to be honest.

Fourdee, 2016-03-29:
@Zone22 
I'l mark this as closed. Please reopen if required.

closed 2016-03-29 by Fourdee