MichaIng/DietPi #260: C1+ | DietPi-System | Missing/invisible cursor on monitor/display

k-plan, 2016-03-29:
@Fourdee 

System: Odroid C1+
Version: DietPi v114 based on v109

Software: with or without xserver/desktop

Problem: Missing/invisible cursor on monitor/display (don#t know, since fb install??)

```
root@oDroid-C1_plus-test:~# fbset -v
Linux Frame Buffer Device Configuration Version 2.1 (23/06/1999)
(C) Copyright 1995-1999 by Geert Uytterhoeven

Opening frame buffer device `/dev/fb0'
Using current video mode from `/dev/fb0'

mode "1280x1024"
    geometry 1280 1024 1280 2048 16
    timings 0 0 0 0 0 0 0
    rgba 5/11,6/5,5/0,0/0
endmode
```
In a ssh terminal it works without problems.

Fourdee, 2016-03-29:
@k-plan 

Can you try the following:
```
#via ssh
setterm -cursor on &> /dev/tty0
setterm -cursor on &> /dev/tty1

#locally
setterm -cursor on

#Just to be sure
setterm -cursor on &> /dev/fb0
setterm -cursor on &> /dev/fb1
```

I think its a known issue with the kernel/framebuffer , but If the above doesn't do it, let me know and I'll contact meveric.

k-plan, 2016-03-30:
@Fourdee 
sorry, was too late last night ....

> ```
> #via ssh
> setterm -cursor on &> /dev/tty0
> setterm -cursor on &> /dev/tty1
> 
> #locally
> setterm -cursor on
> 
> #Just to be sure
> setterm -cursor on &> /dev/fb0
> setterm -cursor on &> /dev/fb1
> ```
> 

All tested, also with:
`root@oDroid-C1_plus-test:~# setterm --cursor on &> /dev/tty0`
locally and remote. No effects - no cursor in cmd or editor at the monitor/display.

Will have a look later, if it works on any of my old backups.

> I think its a known issue with the kernel/framebuffer , but If the above doesn't do it, let me know and I'll contact meveric.

:cry:

k-plan, 2016-03-31:
@Fourdee 

> Will have a look later, if it works on any of my old backups.

Testing:
```
root@oDroide-C1:~# cat /DietPi/dietpi/.version
109

root@oDroide-C1:~# cat /proc/version
Linux version 3.10.80 (root@odroid-wheezy) (gcc version 4.7.2 (Debian 4.7.2-5) ) #1 SMP PREEMPT Wed Jan 20 16:18:02 CET 2016

root@oDroide-C1:~# fbset -v
Linux Frame Buffer Device Configuration Version 2.1 (23/06/1999)
(C) Copyright 1995-1999 by Geert Uytterhoeven

Opening frame buffer device `/dev/fb0'
Using current video mode from `/dev/fb0'

mode "1280x1024"
    geometry 1280 1024 1280 2048 16
    timings 0 0 0 0 0 0 0
    rgba 5/11,6/5,5/0,0/0
endmode

```
No effects - no cursor in cmd or editor at the monitor/display.

:no_mouth:

Fourdee, 2016-05-17:
@k-plan 
I'll mark this old ticket as closed as we dont maintain the kernel for the C1. It may be an issue with the kernel source, but the only way to know is try the ARMbian image and see if you can replicate the same issues: http://www.armbian.com/odroid-c1/

closed 2016-05-17 by Fourdee

k-plan, 2016-05-18:
@Fourdee 

Yes, I had tried this, the first preview (won't boot up) and last beta (no Display output).
Have now realise, there is now Armbian 5.10 out. Will give it a shot, if I find spare time ... :no_mouth: 

:laughing:

Fourdee, 2016-05-18:
@k-plan 

>  first preview (won't boot up) and last beta (no Display output).

Doesnt sound good lol. Would love to see the results of 5.1 if/when you get time to test it.

k-plan, 2016-05-22:
@Fourdee 

> Would love to see the results of 5.1 if/when you get time to test it.

really the very last test.

```
uname -a
Linux odroidc1 3.10.101-odroidc1 #17 SMP PREEMPT Sat Apr 30 21:27:36 CEST 2016 armv7l

```

Did not get any boot.ini config on work, with my `setenv m "sxga"         	# 1280x1024` monitor connected with dvi-2-hdmi cable.
So, no monitor output ... 

Fiddling around with these crappy URAT-2-serial apter to get some output.

```
Starting kernel ...

[    6.684940@0] hdmi: Fixing to HDMI Mode
[    6.685018@0] hdmi: Sink is HDMI device
[    6.687004@0] hdmi: No sink attached
Loading, please wait...
Begin: Loading essential drivers ... done.
```

.. and then:

```
[  244.430171@0] Kernel panic - not syncing: hung_task: blocked tasks
[  244.430834@0] CPU: 0 PID: 43 Comm: khungtaskd Not tainted 3.10.101-odroidc1 #17
[  244.438151@0] [<c0014a58>] (unwind_backtrace+0x0/0xf4) from [<c0011a7c>] (show_stack+0x10/0x14)
[  244.446846@0] [<c0011a7c>] (show_stack+0x10/0x14) from [<c065a2d8>] (panic+0xa0/0x1f0)
[  244.454735@0] [<c065a2d8>] (panic+0xa0/0x1f0) from [<c00927c4>] (watchdog+0x244/0x294)
[  244.462635@0] [<c00927c4>] (watchdog+0x244/0x294) from [<c004ad88>] (kthread+0xb0/0xb4)
[  244.470628@0] [<c004ad88>] (kthread+0xb0/0xb4) from [<c000dce0>] (ret_from_fork+0x14/0x34)
[  244.478874@3] CPU3: stopping
[  244.481652@3] CPU: 3 PID: 206 Comm: systemd-journal Not tainted 3.10.101-odroidc1 #17
[  244.489548@3] [<c0014a58>] (unwind_backtrace+0x0/0xf4) from [<c0011a7c>] (show_stack+0x10/0x14)
[  244.498242@3] [<c0011a7c>] (show_stack+0x10/0x14) from [<c0013320>] (handle_IPI+0xd4/0x17c)
[  244.506572@3] [<c0013320>] (handle_IPI+0xd4/0x17c) from [<c0008470>] (gic_handle_irq+0x58/0x5c)
[  244.515262@3] [<c0008470>] (gic_handle_irq+0x58/0x5c) from [<c000d9bc>] (__irq_usr+0x3c/0x60)
[  244.523765@3] Exception stack(0xedef7fb0 to 0xedef7ff8)
[  244.528924@3] 7fa0:                                     be8e29c0 be8e29bc 00000000 000481a8
[  244.537311@3] 7fc0: 0000000e 00000000 00001608 00000000 00003600 00000000 00000036 00000000
[  244.545646@3] 7fe0: be8e29bc be8e28d0 00000004 0002d318 600f0030 ffffffff
[  244.552394@2] CPU2: stopping
[  244.555203@2] CPU: 2 PID: 0 Comm: swapper/2 Not tainted 3.10.101-odroidc1 #17
[  244.562392@2] [<c0014a58>] (unwind_backtrace+0x0/0xf4) from [<c0011a7c>] (show_stack+0x10/0x14)
[  244.571099@2] [<c0011a7c>] (show_stack+0x10/0x14) from [<c0013320>] (handle_IPI+0xd4/0x17c)
[  244.579430@2] [<c0013320>] (handle_IPI+0xd4/0x17c) from [<c0008470>] (gic_handle_irq+0x58/0x5c)
[  244.588119@2] [<c0008470>] (gic_handle_irq+0x58/0x5c) from [<c000d800>] (__irq_svc+0x40/0x70)
[  244.596622@2] Exception stack(0xef2a1f98 to 0xef2a1fe0)
[  244.601781@2] 1f80:                                                       00000002 00000000
[  244.610169@2] 1fa0: 000096ea 00000000 c0952550 c0669038 ef2a0000 c09b4420 ef2a0000 c09b4420
[  244.618506@2] 1fc0: ef2a0000 ef2a0000 00000000 ef2a1fe0 c000eff4 c000eff8 60000013 ffffffff
[  244.626845@2] [<c000d800>] (__irq_svc+0x40/0x70) from [<c000eff8>] (arch_cpu_idle+0x28/0x2c)
[  244.635271@2] [<c000eff8>] (arch_cpu_idle+0x28/0x2c) from [<c006e4b8>] (cpu_startup_entry+0xf8/0x154)
[  244.647670@2] [<c006e4b8>] (cpu_startup_entry+0xf8/0x154) from [<00856e04>] (0x856e04)
[  244.655640@1] CPU1: stopping
[  244.659089@1] CPU: 1 PID: 0 Comm: swapper/1 Not tainted 3.10.101-odroidc1 #17
[  244.665800@1] [<c0014a58>] (unwind_backtrace+0x0/0xf4) from [<c0011a7c>] (show_stack+0x10/0x14)
[  244.674617@1] [<c0011a7c>] (show_stack+0x10/0x14) from [<c0013320>] (handle_IPI+0xd4/0x17c)
[  244.683042@1] [<c0013320>] (handle_IPI+0xd4/0x17c) from [<c0008470>] (gic_handle_irq+0x58/0x5c)
[  244.691823@1] [<c0008470>] (gic_handle_irq+0x58/0x5c) from [<c000d800>] (__irq_svc+0x40/0x70)
[  244.700425@1] Exception stack(0xef29ff98 to 0xef29ffe0)
[  244.705693@1] ff80:                                                       00000001 00000000
[  244.714160@1] ffa0: 00006a06 00000000 c0952550 c0669038 ef29e000 c09b4420 ef29e000 c09b4420
[  244.722544@1] ffc0: ef29e000 ef29e000 00000000 ef29ffe0 c000eff4 c000eff8 600d0013 ffffffff
[  244.730942@1] [<c000d800>] (__irq_svc+0x40/0x70) from [<c000eff8>] (arch_cpu_idle+0x28/0x2c)
[  244.739434@1] [<c000eff8>] (arch_cpu_idle+0x28/0x2c) from [<c006e4b8>] (cpu_startup_entry+0xf8/0x154)
[  244.748714@1] [<c006e4b8>] (cpu_startup_entry+0xf8/0x154) from [<00856e04>] (0x856e04)

```

Just my 2 Cent:

Sorry, nothing for me any more ....
Igor do great work for Allwinner devices, but for Odroid, he as to spend some more time and work.
But without me ... at the moment only waste of time nerves. Sorry 👎