MichaIng/DietPi #261: PINE64 | Add support and optimizations in DietPi

rhkean, 2016-03-29:
Add [PINE64](http://pine64.com/) support and optimizations to DietPi

(you can assign this to me)

Fourdee, 2016-03-29:
@rhkean 
Done :+1: 

Mines arriving in May, i think lol.

Which Debian Jessie image base will you be using for it?

Some notes/suggestions:
- dietpi-obtain_hw_model
Assign a ```HW_MODEL``` of 40. We will reserve 40-50 for Pines (eg: RPi are 0-10 etc)

- dietpi-config
Might take a while to implement Pine and its ```HW_MODEL```. Will need to go through each option and verify if ```HW_MODEL 40``` will use it, then add the required code to support it (eg: change display resolution)

- dietpi-software
Same as above, need to go through all the installations and verify they work. If not, patch code accordingly.

rhkean, 2016-03-29:
@Fourdee , 
Don't hold your breath on a May delivery.  I was supposed to have mine in February... still don't have it.  :disappointed: 

I'm not certain which base I'll use yet... a lot depends on which (if any) are available, tested, and supported by PINE when I start.   At the moment, in the forums, it doesn't look like they're planning on supporting debian... just Ubuntu.  But, the comments are suggesting a look at hardkernel images.  So, I will probably start with the C2 images....

k-plan, 2016-04-25:
@rhkean 

> it doesn't look like they're planning on supporting debian... just Ubuntu

[Debian Linux with Mate GUI 20160422](http://wiki.pine64.org/index.php/Pine_A64_Software_Release#Debian_Linux_with_Mate_GUI_20160422_by_lenny.raposo_with_Longsleep_kernel)

rhkean, 2016-04-25:
cool!
now, if I can just get my board.. :cry:

rhkean, 2016-04-27:
got my board tonight...  woot!

xenfomation, 2016-04-27:
${somuch}envy  On the positive @rhkean, I am testing 3+ "additional packages" to the dietpi-sotware utility instead of throwing 60 in!  enjoy the board!!

rhkean, 2016-04-27:
don't be too envious...  it's 2 1/2 months late and missing the screws to mount the board in on the case.  😭

rhkean, 2016-04-27:
wow.... 1.1GB zip file, unzips to 8GB, writes a 2.3GB partition.  😃

Fourdee, 2016-04-27:
@rhkean 

> got my board tonight... woot!

woop woop :+1:. When you've had time to test it, you will have to let us know what you think of it.

> wow.... 1.1GB zip file, unzips to 8GB, writes a 2.3GB partition.

lol. I feel the pain from your SD card :)

rhkean, 2016-04-29:
well, this is going to take a while.

- The only debian image is flaky.  
- I can't tell at this point if the NIC problems that I'm encountering are image or driver related
- if the wifi/bluetooth addon board is plugged in, the system starts treating the gigabit Ethernet like a red-headed step child.

I might just build it from scratch... not sure, but it definitely won't be up to DietPi standards for a couple versions.
sorry

Fourdee, 2016-04-29:
@rhkean 

> well, this is going to take a while.

We can always release it under alpha/beta if its "semi-working", this will be your call. I did that for the Odroid C2 whilst we wait for X11 GPU drivers and gives me a chance to iron out some of those "arm64 issues".

> I might just build it from scratch... not sure, but it definitely won't be up to DietPi standards for a couple versions.

Probably best to start from scratch and compile the kernel. I'd focus on getting the device working stable as a headless system first. GPU/VPU support (eg: Kodi, desktop) could be something to be added at a later date.

When mine turns up (hopefully this year lol), i'll jump in aswell.

rhkean, 2016-04-29:
I got an email from backerkit.com with my tracking number...

2 days AFTER my package arrived

Fourdee, 2016-04-29:
lmao. Epic 1st release for Pine :+1:

rhkean, 2016-04-30:
I just got this to boot into a simple kernel/busybox environment 

And connected via UART from my Raspberry Pi

http://forum.pine64.org/showthread.php?tid=99&pid=7625#pid7625

rhkean, 2016-04-30:
blarg!!!!
when did "minimal" come to mean 1.1GB???
http://forum.pine64.org/showthread.php?tid=87

Fourdee, 2016-04-30:
@rhkean 
> when did "minimal" come to mean 1.1GB???

lol. Clearly the definition of minimal, varies depending on the person claiming it :)

rhkean, 2016-05-01:
@Fourdee ,
don't know if it's gonna boot yet, but I was too excited not to share...

http://dietpi.com/phpbb/viewtopic.php?p=1698#p1698

rhkean, 2016-05-01:
@Fourdee 
how big is your C2 image?

k-plan, 2016-05-01:
> how big is your C2 image?

DietPi_v115_Odroid-C2-(Jessie_arm64).img = 644,9 MB

Burned-to-SD-Card:  boot = 42,3 MB  +  rootfs = 454,8 MB

rhkean, 2016-05-01:
I've got a bootable rootfs at 465MB on the Pine.

I did a 
dpkg --get-selections | grep -v deinstall

on the RPi and the Pine 64 after running the debian install, then deleted the extra packages

Just gotta put dietpi on the image now, but need sleep.

Fourdee, 2016-05-01:
@rhkean 

Steps to convert a standard Debian Jessie image into DietPi. Make sure the following files exist. Grab these from the testing branch:
```/boot/```config.txt
```/boot/```dietpi.txt
```/boot/dietpi/``` All files

Also, backup your current image (eg dd > file), before you start this process, just incase. These are notes I work from, just go through each line/command and let me know if you have any issues.
Also, most of these are copy/paste commands, but keep an eye out for manual entries (eg: nano file).

```
#-------------------------------------------------------------------------
//DietPi on a fresh install
#-------------------------------------------------------------------------

#Jessie , unified apt sources.
rm /etc/apt/sources.list.d/deb-multimedia.list

cat << _EOF_ > /etc/apt/sources.list
deb http://ftp.debian.org/debian jessie main contrib non-free
deb http://ftp.debian.org/debian jessie-updates main contrib non-free
deb http://security.debian.org jessie/updates main contrib non-free
deb http://ftp.debian.org/debian jessie-backports main contrib non-free
_EOF_

#Remove following Jessie
apt-get purge dialog dhcpcd5 libsqlite* libxapian22 lua5.1 netcat-* make makedev ncdu plymouth openresolv shared-mime-in* tcpd strace tasksel* wireless-* xdg-user-dirs triggerhappy python* v4l-utils traceroute xz-utils ucf xauth zlib1g-dev xml-core aptitude* avahi-daemon rsyslog logrotate man-db manpages vim vim-common vim-runtime vim-tiny mc mc-data
apt-get autoremove --purge -y

#install
echo -e "CONF_SWAPSIZE=0" > /etc/dphys-swapfile
apt-get install -y ifmetric fbset wpasupplicant resolvconf bc dbus bzip2 psmisc bash-completion cron whiptail sudo ntp ntfs-3g dosfstools parted hdparm pciutils usbutils zip htop wput wget fake-hwclock dphys-swapfile curl unzip ca-certificates console-setup console-data console-common keyboard-configuration wireless-tools wireless-regdb crda --no-install-recommends

#firmware
apt-get install firmware-realtek firmware-ralink firmware-brcm80211 firmware-atheros -y --no-install-recommends

#DIETPI STUFF
#Create folders
mkdir /DietPi
mkdir -p /mnt/usb_1
mkdir -p /mnt/samba
mkdir -p /mnt/ftp_client
echo -e "Samba client can be installed and setup by DietPi-Config.\nSimply run: dietpi-config 8" > /mnt/samba/readme.txt
echo -e "FTP client mount can be installed and setup by DietPi-Config.\nSimply run: dietpi-config 8" > /mnt/ftp_client/readme.txt

/boot/dietpi/dietpi-logclear 2

#FSTAB
cp /boot/dietpi/conf/fstab /etc/fstab

#setup dietpi service
echo 1 > /boot/dietpi/.install_stage
cp /boot/dietpi/conf/dietpi-service /etc/init.d/dietpi-service
chmod +x /etc/init.d/dietpi-service
update-rc.d dietpi-service defaults 00 80
service dietpi-service start

#Cron jobs
cp /DietPi/dietpi/conf/cron.daily_dietpi /etc/cron.daily/dietpi
chmod +x /etc/cron.daily/dietpi
cp /DietPi/dietpi/conf/cron.hourly_dietpi /etc/cron.hourly/dietpi
chmod +x /etc/cron.hourly/dietpi

#Crontab
cat << _EOF_ > /etc/crontab
#Please use dietpi-cron to change cron start times
SHELL=/bin/sh
PATH=/usr/local/sbin:/usr/local/bin:/sbin:/bin:/usr/sbin:/usr/bin

# m h dom mon dow user  command
[... 150 lines cut ...]
systemctl daemon-reload

#Shutdown SSH/Dropbear before reboot
cat << _EOF_ > /etc/systemd/system/kill-ssh-user-sessions-before-network.service
[Unit]
Description=Shutdown all ssh sessions before network
DefaultDependencies=no
Before=network.target shutdown.target

[Service]
Type=oneshot
ExecStart=/usr/bin/killall sshd && /usr/bin/killall dropbear

[Install]
WantedBy=poweroff.target halt.target reboot.target
_EOF_
systemctl enable kill-ssh-user-sessions-before-network
systemctl daemon-reload

#set en_GB.UTF8
dpkg-reconfigure tzdata
dpkg-reconfigure keyboard-configuration #Keyboard must be plugged in for this to work!
dpkg-reconfigure locales

```

rhkean, 2016-05-01:
[image]

Fourdee, 2016-05-01:
@rhkean 
Whoooooohoo, excellent work, nice one! :+1:

k-plan, 2016-05-01:
> _Created by:  Daniel Knight_

:flushed: No:

_Created by:  rhkean (Rob) and Daniel Knight_

👬   

:wink:

Fourdee, 2016-05-01:
@k-plan 

If Rob continues to contribute, it won't be long before I add his name in the banner :)

Fourdee, 2016-05-01:
@rhkean 

You will also need to install and configure netplug on the image, as per: https://github.com/Fourdee/DietPi/issues/273#issuecomment-216048131

Run these commands:
```
#netplug. Resolves issues with unplugging eth and breaking all connections, when WiFi is also active: https://github.com/Fourdee/DietPi/issues/273#issuecomment-215996025
apt-get install netplug -y
cat << _EOF_ > /etc/netplug/netplugd.conf
eth*
wlan*
_EOF_
```

rhkean, 2016-05-01:
but, 

- it's not running dietpi-software on 1st login (not sure what I overlooked)
- there are known issues with the NIC.  It has to be initialized at 100Mbps instead of Gbps or it hangs.  I plugged it into a 100mbps hub, but there's a config tweak that I need to look into
- it's also taking too long to initialize and the obtain_network_details script is running before the nic has an ip address.  I stuck a loop in the code to get the banner to work on login
- the banner isn't displaying on boot (from rc.local), but does display on login
[image]

rhkean, 2016-05-02:
HA!!!!
it's not autorunning because of the

> echo 1 > /boot/dietpi/.install_stage 

I probably shouldn't have stayed up until 6:00am this morning.... 😆

rhkean, 2016-05-02:
so, I'm pretty sure that the reason the banner doesn't show up on the hdmi output is related to the way the TTY's are laid out with @longsleep 's kernel.

Unfortunately, that seems to have gotten really bad after I ran /DietPi/dietpi/finalise on my image.  When I booted the finalized image, I didn't see anything on the output but the dmesg's.

I'll hook the Pi + minicom back up and see if there's any more info that comes over the UART.

rhkean, 2016-05-02:
Now, I have to figure out how to redirect the output to /dev/console
[image]

rhkean, 2016-05-03:
SUCCESS!!!!!!!!!!!!
[image]
now, to patch DietPi

rhkean, 2016-05-06:
@Fourdee 
this is exhausting.... how did you do this for so many other boards

Fourdee, 2016-05-06:
@rhkean 
Persistence lol

Fourdee, 2016-05-06:
Sorry Rob, probably wasnt much help.

So the main things when adding another device to DietPi:
 - ```dietpi-obtain_hw_model``` = Add the device. As Pine is a new manufacture, ```HW_MODEL=40``` would be a good number.
 - ```dietpi-config``` This should work off the bat, but check (find) all references to ```HW_MODEL```. You'll need to implement stuff like display resolution if supported by the device.
- ```dietpi-software``` The big one, again its probably best to check for all references to ```HW_MODEL``` and add for Pine as needed.

Theres alot to do for a new device, so if need any assistance, just let me know. Although, I'am still waiting for my Pine so I wont be able to test it lol :(.

rhkean, 2016-05-06:
@Fourdee 
no worries there.... that was funny.

really, that was just my way of saying, "I'm still here.  I didn't abandon the task."

The biggest challenge is that there isn't an official (or supported, for that matter) build to fall back on... Everything is in flux.  

This will definitely go up as alpha when I'm done with it... Which I'm hoping to complete this weekend.

Fourdee, 2016-05-07:
@rhkean 

> This will definitely go up as alpha when I'm done with it... Which I'm hoping to complete this weekend.

No worries and no rush.
The Pine is a completely new board manufacture which comes with the downside of starting from scratch and lots of debugging (by the looks of it). You are doing a cracking job, so If you need to release it as Alpha, not an issue, we can always patch the system in future releases as needed.

rhkean, 2016-05-09:
@Fourdee @k-plan 

Need clarification
are the following lines in /etc/network/interfaces supposed to be commented out or uncommented with the the latest netplug additions?

```
#allow-hotplug eth0
#allow-hotplug wlan0

```
thank you

Fourdee, 2016-05-09:
@rhkean 

When you finalize (prep system to be run for the 1st time at next reboot) the image with ```/DietPi/dietpi/finalise``` , both get disabled. One will get enabled during 1st run setup based on user settings in dietpi.txt

https://github.com/Fourdee/DietPi/blob/master/dietpi/finalise#L97-L99 | https://github.com/Fourdee/DietPi/blob/master/dietpi/conf/network_interfaces

Fourdee, 2016-05-09:
@rhkean 

> Need clarification
are the following lines in /etc/network/interfaces supposed to be commented out or uncommented with the the latest netplug additions?

They have the same effect as using ```auto eth0```. So you must enable at least one to allow the adapter to be active.

Fourdee, 2016-05-09:
@rhkean 

When you finalize (prep system to be run for the 1st time at next reboot) the image with ```/DietPi/dietpi/finalise``` , both get disabled. One will get enabled during 1st run setup based on user settings in dietpi.txt

https://github.com/Fourdee/DietPi/blob/master/dietpi/finalise#L97-L99 | https://github.com/Fourdee/DietPi/blob/master/dietpi/conf/network_interfaces

If I havn't explained the finalize script to you, I apologise and please let me know.

k-plan, 2016-05-09:
@rhkean 

> Need clarification
are the following lines in /etc/network/interfaces supposed to be commented out or uncommented with the the latest netplug additions?

it is not direct associated with `netplug`. (or `ifmetric`)
https://github.com/Fourdee/DietPi/issues/305

But because it  is no skin off one's nose, it's stay in.
https://github.com/Fourdee/DietPi/issues/273#issuecomment-216053065

----
To explain it short:

_We have now the possibility to have two active network interface.
Each of them has a default route with a gateway. 
If both interfaces are active, which one will be used?
So we need a `metric` and someone who exploit this. ( `ifmetric` )
So the routing table will be build to use only one gateway with one default route._

_But if one of the interfaces goes down or will be plugged out, routing table will not be rebuild automatically, because only the kernel knows, what's happened.
So we needed `netplug` to watch the interface connectivity.
It something happened with the interfaces `netplug` will trigger a ` ifdown` and `ifup`, which rebuild routing table correct with currently valid routes._

_`allow-hotplug` brings only a preconfigured interface up in a running system, without `reboot` or `service networking restart`_

rhkean, 2016-05-10:
I have the following:
- a bootable, debian, DietPi image for the Pine A64+ with arm64 kernel
- updated obtain_hw_info to properly identify the Pine
- modified dietpi-ramdisk script to handle the pine's /boot/uEnv.txt file
- the image is 811MB (125MB in 7z format)

What is not completed
- debugged and modified dietpi-config
- debugged and modified dietpi-software
- I have not finalized the image yet, either
- I have not addressed the Pine's wifi/bt addon board yet

xenfomation, 2016-05-10:
Impressive!  Looks like I need to get a Pine as well.  DietPi -- SBC thirst quencher.

rhkean, 2016-05-14:
@Fourdee 
I need a place to upload the image.

Fourdee, 2016-05-15:
@rhkean o0o0o0o0o0o0o00o0 :dancer: 

Sent you email with FTP details :+1:

k-plan, 2016-05-17:
_The DietPi image for Pine A64 has been created and implemented by Rhkean._

[image]
**DietPi for Pine A64:** _**Give your Pine A64 some lightweight justice**_

###### ALPHA notes and Download:
For all up to date information and downloads for this release, please go to: 
[forum.pine64.org - DietPi for PineA64](http://forum.pine64.org/showthread.php?tid=1068)

Fourdee, 2016-05-17:
@rhkean 
As your Pine image is now released :+1: , you can close this ticket at your leisure. 

If you have outstanding jobs/fixes that you plan to add to the Pine, probably best to create a new ticket for each specific job, or whatever works for you.

Fourdee, 2016-05-17:
@rhkean 
On the social media band wagon lol:
https://twitter.com/DietPi_/status/732630498183368704

Zone22, 2016-06-05:
Looks like my pine64 just arrived!

Fourdee, 2016-06-14:
Mine turned up today :dancer:. @rhkean excellent work on this image :+1:. I'am thinking we should change this to Beta? I cant see any stability issues so far.

rhkean, 2016-06-14:
:fireworks: 
I don't have any objections to beta.

* no bluetooth support for the wifi/bt add-on board
* I haven't tested everything

Fourdee, 2016-06-14:
@rhkean

> no bluetooth support for the wifi/bt add-on board

I didn't order one of those so I'am unable to test :(. Have you had a chance to try with the latest longsleep kernel and the addon board?
```
ifconfig -a
```

> I haven't tested everything

Its impossible to test everything in every possible combination hehe. I'll check Kodi tomorrow for sound and GPU test.

rhkean, 2016-06-14:
> I didn't order one of those so I'am unable to test :(. 

you'll have the [wifi/bt board](https://shop.pine64.com/collections/modules/products/wifi-bluetooth-4-0-module) in a couple days.

> Have you had a chance to try with the latest longsleep kernel and the addon board?

Yes, wifi portion of this[ wifi/bt add-on board](https://shop.pine64.com/collections/modules/products/wifi-bluetooth-4-0-module) works with v118.  Mine's at home on the wired network at the moment, so I'll get you the `ifconfig -a` output tonight.

the `dietpi-config -> update firmware` option will update the u-boot and kernel to longsleep's latest versions.

Fourdee, 2016-06-15:
> you'll have the wifi/bt board in a couple days.

Legend :+1: 

Kodi won't run, i'll look into it.

Fourdee, 2016-06-27:
As Kodi issue has its own ticket: https://github.com/Fourdee/DietPi/issues/380. I'll mark this as closed, great work on this @rhkean :+1:

closed 2016-06-27 by Fourdee