MichaIng/DietPi #263: OPi PC unstable.

Fourdee, 2016-03-30:
> https://twitter.com/ewan2395/status/714738230629900289

This is to resolve the comminuty known issue with the 1.5ghz clock and high voltage issues:
http://www.orangepi.org/orangepibbsen/forum.php?mod=viewthread&tid=785&fromuid=29411

Fourdee, 2016-03-30:
Attempting ARMbian kernel to replace Loboris: https://github.com/igorpecovnik/lib/issues/250#issuecomment-203021008

Fourdee, 2016-03-30:
Notes:
```
rwxrwxrwx  1 root root   24 Mar  8 20:20 script.bin -> /boot/bin/orangepipc.bin
lrwxrwxrwx  1 root root   21 Mar  8 20:19 zImage -> vmlinuz-3.4.110-sun8i
```

Fourdee, 2016-03-30:
Community fix by Thomas Kaiser for vcore/clocks will be applied during v115 patch: https://github.com/igorpecovnik/lib/issues/250#issuecomment-203502587

closed 2016-03-30 by Fourdee

rhkean, 2016-03-30:
now if I can just figure out ho-w to get a 3.15 or greater kernel on the Odroid-XU4, so that we can have UAS (USB Attached SCSI) support