MichaIng/DietPi #264: Configure WLAN0 as Access Point while keeping ETH0

spiderkeys, 2016-03-31:
Is there currently a way to enable Wifi in access point mode while still leaving ethernet enabled? When I try to enable wifi in dietpi-config, it says:

```
(NOTICE) This will disable Ethernet
```
For our application, our main connection to the Pi is via ethernet, but we also want to give the option for clients to connect to it over wifi. The ethernet is not connected to a router that reaches out to the internet, but instead is used by another client computer to connect to the Pi. Perhaps a bit of a strange case, but I found it odd that I couldn't enable both interfaces at the same time in the config tool. Let me know if there is a way to accomplish this that I'm not seeing, or if there is a better way of going about it.

Thanks

WolfganP, 2016-04-01:
I was thinking about the same when started playing with DietPi. Maybe change the concept of exclusive LAN connection (Eth OR Wifi) to a priority based one (use preconfigured wifi credentials if wired ethernet not present)

Fourdee, 2016-04-01:
@spiderkeys 

> Is there currently a way to enable Wifi in access point mode while still leaving ethernet enabled? When I try to enable wifi in dietpi-config, it says:

I believe Installing WiFi hotspot from ```dietpi-software``` will allow you to achieve this. Once installed, the ```dietpi-config``` networking options (wifi) will be converted to SSID/Key assignment. More info: http://fuzon.co.uk/phpbb/viewtopic.php?f=8&t=5&p=1207#p1207

@WolfganP 
> I was thinking about the same when started playing with DietPi. Maybe change the concept of exclusive LAN connection (Eth OR Wifi) to a priority based one (use preconfigured wifi credentials if wired ethernet not present)

Yep I agree. At the very least, DietPi should allow for both eth and wlan to be enabled at the same time (excluding WiFi Hotspot). If you could create a ticket for this feature request, i'd be grateful.

spiderkeys, 2016-04-01:
@Fourdee 
After installing wifi hotspot, I do indeed have those options. However, something else seems to be amiss, as the Rpi3 is not actually bringing up a Wifi access point (can't detect one with a scanning app or any of my devices) despite all of the following looking ok:

In dietpi-config, I configured the SSID, channel, and password info. At the top of dietpi-config/network, it says:
```
Wifi: Available | Enabled | Wifi Hotspot Mode
```
Running ifconfig, seems to match whats in /etc/network/interfaces:
```
wlan0     Link encap:Ethernet  HWaddr b8:27:eb:85:18:cc  
          inet addr:192.168.42.1  Bcast:192.168.42.255  Mask:255.255.255.0
          inet6 addr: fe80::ba27:ebff:fe85:18cc/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:93 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:17146 (16.7 KiB)
```

/etc/network/interfaces:
```
#/etc/network/interfaces
#Please use DietPi-Config to modify network settings.

# Local
auto lo
iface lo inet loopback

# Ethernet
auto eth0
iface eth0 inet dhcp
address 192.168.0.100
netmask 255.255.255.0
gateway 192.168.0.1
#dns-nameservers 8.8.8.8 8.8.4.4

# Wifi
auto wlan0
iface wlan0 inet static
address 192.168.42.1
netmask 255.255.255.0
#gateway 192.168.0.1
#wireless-essid FuzonWifi
#wireless-key abcde12345
#wireless-mode Managed
wireless-power off
#wpa-ssid FuzonWifi
#wpa-psk abcde12345
dns-nameservers 8.8.8.8 8.8.4.4

# IP tables
up iptables-restore < /etc/iptables.ipv4.nat
```

Running iwconfig, looks about right as well:
```
wlan0     IEEE 802.11bgn  ESSID:off/any  
          Mode:Managed  Access Point: Not-Associated   Tx-Power=1496 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
```

The only thing that seems strange is the dmesg output regarding the brcm drivers:
```
[    3.931217] brcmfmac: brcmf_sdio_drivestrengthinit: No SDIO Drive strength init done for chip 43430 rev 1 pmurev 24
[    3.945437] usbcore: registered new interface driver brcmfmac
[    4.120122] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: Dec 15 2015 18:10:45 version 7.45.41.23 (r606571) FWID 01-cc4eda9c
[    4.142915] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
[    4.310208] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
[    4.310257] cfg80211: World regulatory domain updated:
[    4.310269] cfg80211:  DFS Master region: unset
[    4.310281] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[    4.310296] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[    4.310308] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[    4.310321] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
[    4.310336] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[    4.310351] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[    4.310363] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
[    4.310375] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
[    4.310388] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
[    4.355022] bcm2835-rng 3f104000.rng: hwrng registered
[    4.400651] systemd-journald[90]: Received request to flush runtime journal from PID 1
[    4.938953] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[    4.939235] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    6.573935] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[    6.574917] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xC5E1
[    7.676817] random: nonblocking pool is initialized
[   20.632848] brcmfmac: power management disabled
[   20.706132] brcmfmac: brcmf_add_if: ERROR: netdev:wlan0 already exists
[   20.706151] brcmfmac: brcmf_add_if: ignore IF event
[   20.711638] brcmfmac: power management disabled
```
Particularly the lines:
```
[    3.931217] brcmfmac: brcmf_sdio_drivestrengthinit: No SDIO Drive strength init done for chip 43430 rev 1 pmurev 24

[    4.120122] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: Dec 15 2015 18:10:45 version 7.45.41.23 (r606571) FWID 01-cc4eda9c
[    4.142915] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
[    4.310208] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code

[   20.632848] brcmfmac: power management disabled
[   20.706132] brcmfmac: brcmf_add_if: ERROR: netdev:wlan0 already exists
[   20.706151] brcmfmac: brcmf_add_if: ignore IF event
[   20.711638] brcmfmac: power management disabled
```

Any ideas? Seems like there might be a driver issue...

spiderkeys, 2016-04-01:
Sorted the problem out. After making a guess that "WifiHotspot" was really just hostapd, I manually ran hostapd /etc/hostapd/hostapd.conf and got an error message back saying the password I had set was invalid. Maybe at some point some point that validation can added into dietpi-config. Also, perhaps it might be a good idea to put in dietpi-software the name of the actual software, like "WifiHotspot (hostapd)". Not a biggie, but just helps with knowing everything that DietPi is doing/wrapping for when non-obvious issues pop up.

Looking forward to the ability to also put wifi into client mode with Ethernet as mentioned by @WolfganP as that is also one of the configurations for our product :)

closed 2016-04-01 by spiderkeys