MichaIng/DietPi #265: Configure Wifi as fallback LAN for Ethernet link not connected at boot

WolfganP, 2016-04-01:
As per https://github.com/Fourdee/DietPi/issues/264

As a basic connectivity model for DietPi, change the concept of current exclusive LAN connection (Eth OR Wifi) to a priority based one (activate wifi with preconfigured credentials if wired ethernet not present at boot)

It may later be expanded to simultaneous connectivity / both links active model if worthy.

spiderkeys, 2016-04-01:
Would also like to support the notion of providing simultaneous connectivity/bridging the two connections, regardless of whether or not the wifi is infrastructure, adhoc, or AP mode.

Fourdee, 2016-04-02:
@WolfganP 
Thanks for creating the request. As v115 is nearly completed, I'll add this to v116 and start work on it after v115 is released.

@spiderkeys 
Noted :+1: . We should be able to make that possible.

k-plan, 2016-04-04:
Hi,

nice feature and can be really helpful.
Which USB Wifi adapter (chip set) will be supported in DietPi  out-of-the box  adhoc or AP mode?

Tested five different units (chips: 1x TI, 2x ralink, 1x realtek, 1x atheros), not only one will work.

rhkean, 2016-04-04:
probably best to let the kernel detect, then test for presence of wlan0.

WolfganP, 2016-04-08:
From what I read at http://raspberrypi.stackexchange.com/questions/8851/setting-up-wifi-and-ethernet it seems that declaring both interfaces and keeping the proper credentials at wpa_supplicant.conf will work (plus disabling the hotplugging capability).
And obviously not making the selection of interfaces exclusive at dietpi-config...

Fourdee, 2016-04-08:
@rhkean @WolfganP 

> probably best to let the kernel detect, then test for presence of wlan0.

Yep, thats the plan.
Basically we just need to allow the user to enable and configure both adapters at the same time, and support it throughout DietPi scripts (eg: dietpi-config). Then we will probably let the kernel decide which one is actually used.

I'am aiming to start work on this later tonight and tomorrow. My son is staying at his nans for the night, so I will get some time to really dig into this :).

WolfganP, 2016-04-08:
Great!

Fourdee, 2016-04-09:
Done. Wasn't as bad as I thought. Did most of the code work when I implemented the WiFi hotspot support in ```dietpi-config```

- You can now enable and configure both Ethernet and WiFi adapters at the same time.
- You can also disable either adapter from ```dietpi-config```
- I've also rearranged the ```dietpi-config``` menus to split Networking into Adapters and NAS/Misc.

We will let the kernel decide which connection is being utilized, when both are active.

Fourdee, 2016-09-18:
Removal of ifmetric to fix: https://github.com/Fourdee/DietPi/issues/515#issuecomment-247798038
New system required.

Fourdee, 2016-12-19:
Bonding: http://dietpi.com/phpbb/viewtopic.php?f=9&t=340&start=20#p4417
Does the following:
- Allows for fallback to ```wlan0``` when ```eth0``` is unplugged with working route
- ```wlan0``` works if ```eth0``` was unplugged during boot

closed 2017-12-30 by Fourdee

sibero80, 2019-03-30:
Hello Everyone, greetings from Colombia!

Currently my Diet-pi is setup with the Hotspot package, using a Wifi dongle (**wlan0**) and Ethernet (**eth0**) as internet source.

I'm interested on adding a second WiFi adapter (**wlan1**) as I wish to be able to switch to **wlan1** if internet **eth0** gets disconnected. (or maybe bonding both **eth0** and **wlan1** )

This is my current setup:

```
#/etc/network/interfaces
#Please use DietPi-Config to modify network settings.

# Local
auto lo
iface lo inet loopback

# Ethernet
allow-hotplug eth0
iface eth0 inet static
address 192.168.0.18
netmask 255.255.255.0
gateway 192.168.0.1
dns-nameservers 8.8.8.8

# Wifi Hotspot
allow-hotplug wlan0
iface wlan0 inet static
address 192.168.42.1
netmask 255.255.255.0
#gateway 192.168.0.1
#dns-nameservers 8.8.8.8 8.8.4.4

# IP tables
up iptables-restore < /etc/iptables.ipv4.nat
```

From your previous post:
1. Do yo thing the following setup would do the trick? (My second Wifi dongle has not arrived, so cannot test right now) 
2. How can I set up DHCP in **wlan1** in a way that does not require the IP Adress, Netmask and Gateway to be specified?

```
#/etc/network/interfaces
#Please use DietPi-Config to modify network settings.

# Local
auto lo
iface lo inet loopback

# Ethernet
allow-hotplug eth0
iface eth0 inet static
address 192.168.0.18
netmask 255.255.255.0
gateway 192.168.0.1
dns-nameservers 8.8.8.8
    bond-master bond0
    bond-primary eth0
    bond-mode active-backup

# Wifi Hotspot
allow-hotplug wlan0
iface wlan0 inet static
address 192.168.42.1
netmask 255.255.255.0
#gateway 192.168.0.1
#dns-nameservers 8.8.8.8 8.8.4.4

# IP tables
up iptables-restore < /etc/iptables.ipv4.nat

# Wifi Fallback
allow-hotplug wlan1
iface wlan0 inet dhcp
address 192.168.0.100
netmask 255.255.255.0
gateway 192.168.0.1
wireless-essid WIFI_SSID
wireless-key WIFI_KEY
wireless-mode Managed
wireless-power off
wpa-ssid WIFI_SSID
wpa-psk WIFI_KEY
#dns-nameservers 8.8.8.8 8.8.4.4
    bond-master bond0
    bond-primary eth0
    bond-mode active-backup

# Define bond
auto bond0
iface bond0 inet dhcp
    bond-slaves none
    bond-primary eth0
    bond-mode active-backup
    bond-miimon 100
```

Thank you!