MichaIng/DietPi #267: Add netdata (realtime perfomance monitoring) to dietpi

rhkean, 2016-04-04:
feature suggestion:
Readme.md:  https://github.com/firehol/netdata/blob/master/README.md

http://fuzon.co.uk/phpbb/viewtopic.php?f=9&t=354

rhkean, 2016-04-04:
This is my first full add-on feature (not just a code tweak to @Fourdee 's code), so figured I'd start small... (since this will require modifications to the whiptail menus and such)

I'll probably lean on you guys more for review and testing....

Fourdee, 2016-04-04:
@rhkean 
Nice :+1: Looking forward to seeing this.

k-plan, 2016-04-04:
@rhkean 

Cool, assistance for Fourdee.  :+1: 

> I'll probably lean on you guys more for review and testing....

If you have something to test, give it to us.  :smile:

Fourdee, 2016-04-19:
@rhkean 
Excellent work on implementing this, top stuff :+1: 

Ticket closed, marked as completed.

closed 2016-04-19 by Fourdee