MichaIng/DietPi #269: Kodi slow after running idle and libcec3 not working

hydrouk, 2016-04-05:
I have noticed Kodi starts to run really slow after a while idle and needs a reboot to get back to speed and libcec is not working on multiple TV's and receivers.

Fourdee, 2016-04-05:
@hydrouk 

Thanks for the report, what device are you using. Are you running Kodi from console/desktop or autoboot?

hydrouk, 2016-04-05:
XU4 device and its auto start from the dietpi settings

Fourdee, 2016-04-06:
@hydrouk 
Thanks for the info.

> libcec is not working on multiple TV's and receivers.

Unfortunately, I do not own a CEC compatible device, so I am unable to assist you regarding that issue. I have a new CEC TV on my amazon wishlist, its been there a while lol :)

> Kodi starts to run really slow after a while idle and needs a reboot to get back to speed

I did a quick search through the XU4 Odroid forums and couldnt find similar reports, so I will need to setup my XU4 and try to replicate this issue.

Could you tell me the following:
- When you say slow, are you referring to the menu interface, video playback, or everything?
- How long does it roughly take for this "slowness" to start appearing?
- What kind of video media are you playing, .avi .mkv, youtube streams etc?

When you notice Kodi is running slow, can you exit and provide (upload/paste) the following logfiles:
- ```/root/.kodi/temp/kodi.log```
- ```/var/log/Xorg.0.log```

hydrouk, 2016-04-09:
Not getting anything in the logs that looks wrong, unplugged the cloudshell board to eliminate that from the stability issue, gonna run for 24 hours now and see if the lag disappears

hydrouk, 2016-04-09:
I don't have any lag now since disconnecting the cloudshell PCB, i think it might have something to do with that, i have a 2TB drive installed with 300GB+ of music and some video files but i do not have library mode activated, I was also getting video lag before, it was dropping frames

Fourdee, 2016-04-10:
@hydrouk 

> UPDATE
> Lag is still present but not taking as long to recover, i will post logs when i get home but i do not see anything to cover this lag myself, not sure its Kodi at all.

Yep, seems strange removing the cloudshell would have a positive effect on the issue. It could indicate a power issue, but I'll wait for your response regarding the logs.

The cloudshell LCD screen is enabled in modules by default on the DietPi XU4 image. Although plugging in a HDMI cable will override the output, it may be worth completely disabling the LCD screen in modules to check for any improvements.

Disable Cloudshell LCD:
```
rm /etc/modprobe.d/odroid-cloudshell.conf
sed -i '/spi_s3c64xx/d' /etc/modules
sed -i '/fbtft_device/d' /etc/modules
```

Enable Cloudshell LCD (if you need to reactivate it):
```
#Cloudshell (Enable LCD screen)
echo "options fbtft_device name=hktft9340 busnum=1 rotate=270" > /etc/modprobe.d/odroid-cloudshell.conf
echo "spi_s3c64xx" >> /etc/modules
echo "fbtft_device" >> /etc/modules
```

hydrouk, 2016-04-14:
I have an update!

I did an aptitude update today and it had a new kernel so i applied that and since then Kodi has been running perfectly and i can live with HDMI-CEC if Kodi continues to run as nice as it does.  I have done nothing else and the USB3 cloudshell interface and drive is connected again.  Thanks Fourdee

Fourdee, 2016-04-15:
@hydrouk 
Excellent, Meveric must of compiled the new kernel which fixes CEC. Thanks for letting us know its working :)

hydrouk, 2016-04-15:
Doh! forgot to test HDMI-CEC and it's still not working but Kodi is a lot
more stable, sorry about that

Fourdee, 2016-04-15:
@hydrouk 
Ah thanks for letting me know. 

Not sure what I can do to help regarding the CEC as I have no CEC capable TV's to even test it. I can only suggest searching the XU4 Odroid forums for other CEC users, maybe they have experienced a similar issue? http://forum.odroid.com/viewforum.php?f=92

hydrouk, 2016-04-15:
I think it's a small thing compared to the Kodi issue I had but it would be
nice to have CEC working some day. Thanks for your help and of course
DietPi

rhkean, 2016-04-15:
@hydrouk 

To confirm... CEC is not working when attempting to use it to control the kodi installed by DietPi on an Odroid-XU4

correct?

hydrouk, 2016-04-15:
Indeed

rhkean, 2016-04-15:
can you try this from the BASH prompt:

`apt-get install libcec`

thanks

hydrouk, 2016-04-15:
I did have it installed but it fails to work as it should, my receiver gets
some data over it but nothing is received and it locks up Kodi a bit and
also crashed my receiver if on over night. I had a previous problem where
Kodi was laggy the next day but that had nothing to do with CEC and was
fixed by the latest kernel.

rhkean, 2016-04-15:
well, I hadn't used the DietPi kodi install before, b/c it was 2 version behind and the databases were compatible.  But, I see the meverick has updated it to version 16, so I'll try to test some this weekend.

I have a suggestion for you, though... (actually 2)
1.  take the receiver out of the setup and see it your problem goes away.
2.  send me a screenshot of your CEC configuration.... there's an "ID" option that comes into play when dealing with chained devices.  (I can't recall the config option off the top of my head and I can't look b/c I'm at work)

hydrouk, 2016-04-15:
Will do thanks for your help

rhkean, 2016-04-16:
testing update:
Odroid-XU4
new install:  v116 (testing)
selection: kodi
autostart: kodi

result:
rebooted into kodi, settings->system->input -> no CEC input options
ssh to machine, apt-get install libcec, reboot

result:
rebooted into kodi, automatically initialized CEC

hydrouk, 2016-04-16:
Today i decided enough is enough and flashed the v109 image and let it
upgrade and installed the bits i wanted, when Kodi started i did as you
instructed but still the same deal however i had not ventured in to the
peripherals menu before so i disabled everything and just left it Enabled,
rebooted and boom it connected to my Onkyo receiver. I guess what we
learned here is all the fancy bits do not work with Onkyo receivers and
maybe that should be put on a sticky but maybe written better than i just
did. I really appreciate the help *Fourdee
<https://github.com/Fourdee>* & *rhkean
<https://github.com/rhkean>* because without you i would still be moaning
at my DietPi installation heh :) I hope this does get put somewhere because
it was frustrating for me and should be easy for another.

rhkean, 2016-04-16:
Onkyo's are evil
not as evil as Harmon Kardon receivers.
I'm just glad we could help get it working.

hydrouk, 2016-04-16:
Yea it does appear they like to ignore major parts of the HDMI-CEC protocol
but yea we cracked it!

joaofl, 2016-09-26:
> I have noticed Kodi starts to run really slow after a while idle and needs a reboot to get back to speed

The same is happening here with my XU4. But noting that I don't use cloudshell, neither CEC. 
The issue is the same: interface lack of responsiveness after some time idle.

This means: a send arrow commands, and the interface reacts sometimes 1 minute later. But if u send many commands, they actually get buffered, and are executed suddenly all at once.
It take a while to navigate to a file, but once it loads, it plays smoothly, unless again you need to interact with the interface.

After rebooting it works fine for a while.

For some time I thought it was the wireless keyboard I use, but tested somewhere else and it is fine.

Xorg.0.log is empty

`root@odroid:~# cat /root/.kodi/temp/kodi.log
17:40:21 T:3004067840  NOTICE: special://profile/ is mapped to: special://masterprofile/
17:40:21 T:3004067840  NOTICE: -----------------------------------------------------------------------
17:40:21 T:3004067840  NOTICE: Starting Kodi (16.1 Git:2016-04-24-cdb7704-dirty). Platform: Linux ARM (Thumb) 32-bit
17:40:21 T:3004067840  NOTICE: Using Release Kodi x32 build
17:40:21 T:3004067840  NOTICE: Kodi compiled Apr 24 2016 by GCC 4.9.2 for Linux ARM (Thumb) 32-bit version 3.16.7 (200711)
17:40:21 T:3004067840  NOTICE: Running on Debian GNU/Linux 8 (jessie), kernel: Linux ARM 32-bit version 3.10.103+
17:40:21 T:3004067840  NOTICE: FFmpeg statically linked, version: 2.6.8
17:40:21 T:3004067840  NOTICE: WARNING: unsupported ffmpeg version detected
17:40:21 T:3004067840  NOTICE: Host CPU: ARMv7 Processor rev 3 (v7l), 8 cores available
17:40:21 T:3004067840  NOTICE: ARM Features: Neon enabled
17:40:21 T:3004067840  NOTICE: special://xbmc/ is mapped to: /usr/local/share/kodi
17:40:21 T:3004067840  NOTICE: special://xbmcbin/ is mapped to: /usr/local/lib/kodi
17:40:21 T:3004067840  NOTICE: special://masterprofile/ is mapped to: /root/.kodi/userdata
17:40:21 T:3004067840  NOTICE: special://home/ is mapped to: /root/.kodi
17:40:21 T:3004067840  NOTICE: special://temp/ is mapped to: /root/.kodi/temp
17:40:21 T:3004067840  NOTICE: The executable running is: /usr/local/lib/kodi/kodi.bin
17:40:21 T:3004067840  NOTICE: Local hostname: odroid
17:40:21 T:3004067840  NOTICE: Log File is located: /root/.kodi/temp/kodi.log
17:40:21 T:3004067840  NOTICE: -----------------------------------------------------------------------
17:40:21 T:3004067840   ERROR: DBus: Error org.freedesktop.DBus.Error.ServiceUnknown - The name org.freedesktop.UPower was not provided by any .service files
17:40:21 T:3004067840  NOTICE: load settings...
17:40:21 T:3004067840  NOTICE: Found 1 Lists of Devices
17:40:21 T:3004067840  NOTICE: Enumerated PULSE devices:
17:40:21 T:3004067840  NOTICE:     Device 1
17:40:21 T:3004067840  NOTICE:         m_deviceName      : Default
17:40:21 T:3004067840  NOTICE:         m_displayName     : Default
17:40:21 T:3004067840  NOTICE:         m_displayNameExtra: Default Output Device (PULSEAUDIO)
17:40:21 T:3004067840  NOTICE:         m_deviceType      : AE_DEVTYPE_PCM
17:40:21 T:3004067840  NOTICE:         m_channels        : FL,FR
17:40:21 T:3004067840  NOTICE:         m_sampleRates     : 5512,8000,11025,16000,22050,32000,44100,48000,64000,88200,96000,176400,192000,384000
17:40:21 T:3004067840  NOTICE:         m_dataFormats     : AE_FMT_U8,AE_FMT_S16NE,AE_FMT_S24NE3,AE_FMT_S24NE4,AE_FMT_S32NE,AE_FMT_FLOAT
17:40:21 T:3004067840  NOTICE:     Device 2
17:40:21 T:3004067840  NOTICE:         m_deviceName      : alsa_output.usb-ASUS_Xonar_U7_Echelon_Ed.-00-Ed.analog-stereo
17:40:21 T:3004067840  NOTICE:         m_displayName     : Xonar U7 Echelon Ed. Analog Stereo
17:40:21 T:3004067840  NOTICE:         m_displayNameExtra: Xonar U7 Echelon Ed. Analog Stereo (PULSEAUDIO)
17:40:21 T:3004067840  NOTICE:         m_deviceType      : AE_DEVTYPE_PCM
17:40:21 T:3004067840  NOTICE:         m_channels        : FL,FR
17:40:21 T:3004067840  NOTICE:         m_sampleRates     : 5512,8000,11025,16000,22050,32000,44100,48000,64000,88200,96000,176400,192000,384000
17:40:21 T:3004067840  NOTICE:         m_dataFormats     : AE_FMT_U8,AE_FMT_S16NE,AE_FMT_S24NE3,AE_FMT_S24NE4,AE_FMT_S32NE,AE_FMT_FLOAT
17:40:21 T:3004067840  NOTICE:     Device 3
17:40:21 T:3004067840  NOTICE:         m_deviceName      : alsa_output.platform-sound.13.analog-stereo
17:40:21 T:3004067840  NOTICE:         m_displayName     : odroid-audio Analog Stereo
17:40:21 T:3004067840  NOTICE:         m_displayNameExtra: Analog Output (PULSEAUDIO)
17:40:21 T:3004067840  NOTICE:         m_deviceType      : AE_DEVTYPE_PCM
17:40:21 T:3004067840  NOTICE:         m_channels        : FL,FR
17:40:21 T:3004067840  NOTICE:         m_sampleRates     : 5512,8000,11025,16000,22050,32000,44100,48000,64000,88200,96000,176400,192000,384000
17:40:21 T:3004067840  NOTICE:         m_dataFormats     : AE_FMT_U8,AE_FMT_S16NE,AE_FMT_S24NE3,AE_FMT_S24NE4,AE_FMT_S32NE,AE_FMT_FLOAT
17:40:21 T:3004067840  NOTICE: No settings file to load (special://xbmc/system/advancedsettings.xml)
17:40:21 T:3004067840  NOTICE: No settings file to load (special://masterprofile/advancedsettings.xml)
17:40:21 T:3004067840  NOTICE: Default DVD Player: dvdplayer
17:40:21 T:3004067840  NOTICE: Default Video Player: dvdplayer
17:40:21 T:3004067840  NOTICE: Default Audio Player: paplayer
17:40:21 T:3004067840  NOTICE: Disabled debug logging due to GUI setting. Level 0.
[... cut ...]
00:44:33 T:2768692144  NOTICE: Closing stream player 3
00:44:34 T:3004060592  NOTICE: PulseAudio: Opened device Default in pcm mode with Buffersize 199 ms
00:44:34 T:3004067840  NOTICE: CDVDPlayer::CloseFile()
00:44:34 T:3004067840  NOTICE: DVDPlayer: waiting for threads to exit
00:44:34 T:3004067840  NOTICE: DVDPlayer: finished waiting
00:44:34 T:3004067840  NOTICE: CDVDPlayer::CloseFile()
00:44:34 T:3004067840  NOTICE: DVDPlayer: waiting for threads to exit
00:44:34 T:3004067840  NOTICE: DVDPlayer: finished waiting`

hydrouk, 2016-09-26:
To add I never resolved this and stopped using Kodi but would love to be able to again.

Fourdee, 2016-09-26:
@joaofl @hydrouk 

Hi guys,

Thanks for letting us know 👍 

I'll need to run some tests to try and replicate. Some questions:
- What media are you guys playing, h264?
- Roughly how long before you can see the latency?

> ERROR: CDVDMsgGeneralSynchronize - timeout 

note to self: Seems to crop up alot, dropping frames, need to look into it.

hydrouk, 2016-09-26:
I play all kinds of media from local and remote locations

joaofl, 2016-09-26:
like  hydrouk sayd, I also play all kinds of media, and it doesn't seem to
be related to that. But I'll take a look.

I would say that in 2hrs, you can already feel it. For example, after
watching  a movie.

I notice that very often, CPU usage is around 10 to 25%, which means that
its taking almost 100% of one of the 8 cores. Seems to be some overflow, or
some infinite loop somewhere, which was not present in previous version of
Kodi used in Dietpi.

joaofl, 2016-09-26:
@Fourdee 
Just to confirm the last media I played is coded H264 MPEG 4 AVC

joaofl, 2016-09-26:
@Fourdee 
Forgot to mention another issue/symptom I'm having, which I'm not sure it is related to that or note.

Sometimes, in the middle of the video playback, the image simply freezes, but the audio and subtitles playback continues. To solve the issue I have to rewind before the freezing point, and play it again. Not aways works. It seems that I have to rewind enough to force some re-caching.. Otherwise it will freeze at the same point. 

Maybe related to hardrive spining-on/off? Not sure.

Let me know if I can help on anything else.

joaofl, 2016-10-25:
Adding: If I go and check the system log, thats the error the video playback dumps:

```
[ 1885.956157] [c4] [d:0, c:0] s5p_mfc_open:1629: NORMAL instance is opened [0:1]
[ 1885.957470] [c4] [d:0] s5p_mfc_alloc_firmware:122: Port for DRM F/W : 0xafc00000
[ 1885.959896] [c4] [d:0] s5p_mfc_init_memctrl:304: [1] Base Address : 10000000
[ 1885.961288] [c4] [d:0] mfc_init_hw:426: MFC v8.0, F/W: 14yy, 01mm, 13dd (D)
[ 1885.962077] [c4] [d:0, c:0] s5p_mfc_open:1725: MFC open completed [0:1] dev = ddccb010, ctx = c285c000
[ 1886.813863] [c4] [d:0, c:0] vidioc_s_fmt_vid_out_mplane:1401: Dec input codec(0): H264 Encoded Stream
[ 1886.827213] [c3] [d:0, c:0] vidioc_s_fmt_vid_cap_mplane:1358: Dec output pixelformat : 4:2:0 2 Planes Y/CbCr
[ 1886.828444] [c3] [d:0, c:0] s5p_mfc_dec_calc_dpb_size:708: SEQ Done: Movie dimensions 1920x1088, buffer dimensions: 1920x1088
[ 1886.899668] [c5] [d:0, c:0] s5p_mfc_dec_calc_dpb_size:708: SEQ Done: Movie dimensions 1920x1088, buffer dimensions: 1920x1088
[ 1887.927093] [c1] [d:0, c:0] s5p_mfc_handle_frame_new:681: Warning for displayed frame: 160
[ 1890.458424] [c1] [d:0, c:0] s5p_mfc_handle_frame_new:681: Warning for displayed frame: 160
[ 5521.046819] [c3] [d:0, c:0] s5p_mfc_release:1883: MFC driver release is called [0:1], is_drm(0)
[ 5521.047830] [c3] s5p_mfc_dev_release:1801: power off
[ 5521.053068] [c3] [d:0] s5p_mfc_release:2037: mfc driver release finished [0:0], dev = ddccb010
```

hydrouk, 2017-05-18:
Is there any update on this at all? i am kinda sad i cannot use my XU4 for Kodi when the C2 is running it fine. Thanks

Fourdee, 2017-05-19:
@hydrouk 

4.9 kernel update may provide a fix, however, 4.9 is still in testing:
https://github.com/Fourdee/DietPi/issues/926

MichaIng, 2018-04-06:
Is kernel 4.14 stable now on Odroids? Can't exactly assure from here, but they are talking about "release" at least: https://forum.odroid.com/viewtopic.php?f=146&t=28895

Fourdee, 2018-04-06:
@MichaIng 

> Is kernel 4.14 stable now on Odroids?

Unsure and untested. Meveric has installed 4.14 for his XU4 stretch image, so, ours will be that by default aswell.

We'll probably leave Jessie image running 3.x.

Fourdee, 2018-07-09:
Ticket cleanup.

If you experience the issue mentioned in this ticket, please reopen.

closed 2018-07-09 by Fourdee