MichaIng/DietPi #271: General | Add support for 2+ USB drives

Fourdee, 2016-04-06:


xenfomation, 2016-04-24:
As of 18:00 EST, I am will pull this down as I am quite interested to take an approach to solve via /dev/disk-by-uuid/, etc!

xenfomation, 2016-04-27:
Update -- I have my third Pi running and will be investigating this/testing before end of week.

Fourdee, 2016-04-27:
@xenfomation 
Excellent :+1: Keep us updated when you can :)

xenfomation, 2016-04-28:
Will do @Fourdee !!  Thanks again!

xenfomation, 2016-05-07:
This is still in progress, however [http://dietpi.com/phpbb/viewtopic.php?f=11&t=40](http://dietpi.com/phpbb/viewtopic.php?f=11&t=40)1 / [https://github.com/Fourdee/DietPi/issues/271](https://github.com/Fourdee/DietPi/issues/271) has grabbed my time and attention.

The high-level is to ensure that upon USB installation options, the process is seamless for the user, but tracks unique USB identifiers.  In addition, considerations need to be contained within a whiptail menu for informing the admin as well as cache mechanisms, boot/fstab mounts, etc.

xenfomation, 2016-05-12:
As an update, I am still working on this to ensure from init to reboot, the two usb drives remain useful!

Fourdee, 2016-09-07:
Note to self:
Create a system that allows 
- uses to add 1 USB drive at a time
- Have that drive formatted if needed
- Use ```UUID``` to replace ```fstab``` entry

Max 5 drives with new system.

Fourdee, 2016-09-08:
[image]
[image]

k-plan, 2016-09-08:
@Fourdee 

looks very good! :+1: 

Can we get R/W support for FAT16, FAT32, NTFS, EXT2/3/4 and HFS+ ?

Fourdee, 2016-09-08:
@k-plan 

> Can we get R/W support for FAT16, FAT32, NTFS, EXT2/3/4

Already supported on our image by default.

> HFS+

What filesystem is this? lol

Fourdee, 2016-09-08:
@k-plan 
Initial code is done, its can be run with:
```
/DietPi/dietpi/dietpi-external_drive_setup 2
```
Going to need lots and lots of testing and debugging with various drives and combinations.

Fourdee, 2016-09-08:
Notes to self:
- after format UUID is changed.

k-plan, 2016-09-08:
@Fourdee 

> What filesystem is this? lol

:smiley:   [hfsplus](https://en.wikipedia.org/wiki/HFS_Plus)

```
:~# apt-get install hfs
hfsplus         hfsprogs        hfsutils        hfsutils-tcltk

```

 But with usb-automount?  https://github.com/Fourdee/DietPi/issues/501#issuecomment-245317621

```
:~# mkfs.
mkfs.bfs      mkfs.ext2     mkfs.ext4     mkfs.f2fs     mkfs.minix    mkfs.ntfs
mkfs.cramfs   mkfs.ext3     mkfs.ext4dev  mkfs.fat      mkfs.msdos    mkfs.vfat

But anyway. Please finish your awesome work first, so I will be able to test it. Or do we need a new image version  for this lovely  feature?
```

Fourdee, 2016-09-08:
> apt-get install -y hfsplus

:+1: 
I'll probably add this to our image by default to enable hfs support.

Fourdee, 2016-09-08:
Ok i'll add hfs+ support to all DietPi images by default:
[image]

k-plan, 2016-09-08:
@Fourdee 

> I'll probably add this to our image by default to enable hfs support.

Thanks, but you don't have to do that only for me. :smile: 

> Ok i'll add hfs+ support to all DietPi images by default:

Yes, I think it's a good decision, where is a world aside from Windows and Linux. :wink: 

------------------

Quick and dirty test. It works great so far. :+1: 

Only downside: the boot up time will increase with 5 devices (not connected or active)  in` /etc/fstab`
Hmm .... is it possible user can active/disable the numbers of entries in  in` /etc/fstab` via whiptail menu?

So default is one or two and can be switch from 0-5?

```
#External Drives---------------------------------------------------
/dev/sda1       /mnt/usb_1      auto     defaults,noatime,nofail,x-systemd.automount  0       0
## /dev/sdb1       /mnt/usb_2      auto     defaults,noatime,nofail,x-systemd.automount  0       0
## /dev/sdc1       /mnt/usb_3      auto     defaults,noatime,nofail,x-systemd.automount  0       0
## /dev/sdd1       /mnt/usb_4      auto     defaults,noatime,nofail,x-systemd.automount  0       0
## /dev/sde1       /mnt/usb_5      auto     defaults,noatime,nofail,x-systemd.automount  0       0
```

k-plan, 2016-09-09:
@Fourdee 

> Going to need lots and lots of testing and debugging with various drives and combinations.

Yes, Sir.  😃 

hmm, I think we need  to have a look. RPi Zero `dmesg` output - if no usb storage is connected:
(only a keyboard and usb-network adapter) 
```
root@RPi-Zero:~# dmesg | grep dev-s*
[    4.769013] systemd[1]: Expecting device dev-ttySAC2.device...
[    4.799015] systemd[1]: Expecting device dev-ttyS0.device...
[    4.829005] systemd[1]: Expecting device dev-ttyAMA0.device...
[    5.098805] systemd[1]: Expecting device dev-sda1.device...
[    5.170138] systemd[1]: Expecting device dev-sdb1.device...
[    5.239501] systemd[1]: Expecting device dev-sdc1.device...
[    5.309188] systemd[1]: Expecting device dev-sdd1.device...
[    5.379203] systemd[1]: Expecting device dev-sde1.device...
[    5.449078] systemd[1]: Expecting device dev-mmcblk0p1.device...
[  195.368683] systemd[1]: Job dev-sdb1.device/start timed out.
[  376.868702] systemd[1]: Job dev-sde1.device/start timed out.
[  468.368639] systemd[1]: Job dev-sdd1.device/start timed out.
[  650.118791] systemd[1]: Job dev-sda1.device/start timed out.
[  740.888613] systemd[1]: Job dev-sde1.device/start timed out.
[  922.628695] systemd[1]: Job dev-sdc1.device/start timed out.
[ 1194.868803] systemd[1]: Job dev-sdb1.device/start timed out.
[ 1467.118781] systemd[1]: Job dev-sdb1.device/start timed out.
[ 1648.868644] systemd[1]: Job dev-sdb1.device/start timed out.
[ 1830.118683] systemd[1]: Job dev-sdd1.device/start timed out.
[ 1921.118686] systemd[1]: Job dev-sdb1.device/start timed out.
[ 2011.618731] systemd[1]: Job dev-sdb1.device/start timed out.
[ 2102.368648] systemd[1]: Job dev-sdb1.device/start timed out.
[ 2193.118671] systemd[1]: Job dev-sdb1.device/start timed out.
[ 2283.868774] systemd[1]: Job dev-sdb1.device/start timed out.
[ 2465.368673] systemd[1]: Job dev-sdb1.device/start timed out.
[ 2556.118755] systemd[1]: Job dev-sdb1.device/start timed out.
[ 2646.868676] systemd[1]: Job dev-sdb1.device/start timed out.
[ 2737.618676] systemd[1]: Job dev-sdb1.device/start timed out.
[ 2828.368733] systemd[1]: Job dev-sdb1.device/start timed out.
[ 2919.118675] systemd[1]: Job dev-sdd1.device/start timed out.
[ 3009.868673] systemd[1]: Job dev-sdb1.device/start timed out.
[ 3100.618722] systemd[1]: Job dev-sdb1.device/start timed out.
[ 3191.141041] systemd[1]: Job dev-sdb1.device/start timed out.
[ 3281.892008] systemd[1]: Job dev-sdd1.device/start timed out.
[ 3372.642011] systemd[1]: Job dev-sdb1.device/start timed out.
[ 3463.392042] systemd[1]: Job dev-sdb1.device/start timed out.
[ 3554.142019] systemd[1]: Job dev-sdd1.device/start timed out.
[ 3644.892014] systemd[1]: Job dev-sdd1.device/start timed out.
[ 3735.642045] systemd[1]: Job dev-sdb1.device/start timed out.
[ 3826.392008] systemd[1]: Job dev-sdb1.device/start timed out.
[ 3917.142011] systemd[1]: Job dev-sdb1.device/start timed out.
[ 4007.892041] systemd[1]: Job dev-sdb1.device/start timed out.
[ 4189.392008] systemd[1]: Job dev-sdb1.device/start timed out.
[ 4279.892035] systemd[1]: Job dev-sdb1.device/start timed out.
[ 4370.642073] systemd[1]: Job dev-sdb1.device/start timed out.
[ 4461.392012] systemd[1]: Job dev-sdb1.device/start timed out.
[ 4552.142014] systemd[1]: Job dev-sdb1.device/start timed out.
[ 4642.892078] systemd[1]: Job dev-sdb1.device/start timed out.
[ 4733.642008] systemd[1]: Job dev-sdb1.device/start timed out.
```

NanoPi Neo`dmesg` output - no usb device connected
```
root@NeoPi:~# dmesg | grep dev-s*
[    9.784027] systemd[1]: Expecting device dev-ttyAMA0.device...
[    9.784571] systemd[1]: Expecting device dev-ttySAC2.device...
[    9.784885] systemd[1]: Expecting device dev-ttyS0.device...
[    9.787844] systemd[1]: Expecting device dev-sda1.device...
[    9.790193] systemd[1]: Expecting device dev-sdb1.device...
[    9.799665] systemd[1]: Expecting device dev-sdc1.device...
[    9.801322] systemd[1]: Expecting device dev-sdd1.device...
[    9.802813] systemd[1]: Expecting device dev-sde1.device...
[    9.804363] systemd[1]: Expecting device dev-mmcblk0p1.device...
[  100.010269] systemd[1]: Job dev-sda1.device/start timed out.
[  351.510284] systemd[1]: Job dev-sda1.device/start timed out.
[126228.500316] systemd[1]: Job dev-sdb1.device/start timed out.
```

Fourdee, 2016-09-10:
> So default is one or two and can be switch from zero to five (0-5) to speed up boot process a little bit?

Yep. Just need to check if we can re-enable in fstab and have the auto detection working without a reboot or manual intervention.

> Yes, I think it's a good decision, where is a world aside from Windows and Linux.

Lol. No idea, but I heard if you eat one every day, keeps the doctor away :wink: 

> [  195.368683] systemd[1]: Job dev-sdb1.device/start timed out.
[  376.868702] systemd[1]: Job dev-sde1.device/start timed out.
[  468.368639] systemd[1]: Job dev-sdd1.device/start timed out.
[  650.118791] systemd[1]: Job dev-sda1.device/start timed out.
[  740.888613] systemd[1]: Job dev-sde1.device/start timed out.
[  922.628695] systemd[1]: Job dev-sdc1.device/start timed out.

Interesting its timing out. I need to:
- Check if a timed-out job prevents auto detection working in future
- Boot time with and without X auto mounts.

Fourdee, 2016-09-10:
## Ok bootup time:
- Static IP
- Performance Gov
- ```echo -e "$(cat /proc/uptime)" > /var/log/bootuptime``` at end of ```/etc/rc.local``` (Might as this as standard, pretty cool stats). "The first number is the uptime in seconds... "

===

#### NO auto mounts enabled: 
- Result = 10.52s
- Result = 10.58s
- Result = 10.57s

#### 20 auto mounts enabled | No devices attached:
```
#Create mount folders
for ((i=1;i<=20;i++)) do mkdir -p /mnt/usb_$i;done

#Add to /etc/fstab
/dev/sda1       /mnt/usb_1      auto     defaults,noatime,nofail,x-systemd.automount  0       0
.....
/dev/sdt1       /mnt/usb_20      auto     defaults,noatime,nofail,x-systemd.automount  0       0
```

- Result = 15.89s
- Result = 15.96s
- Result = 15.99s

## Results:
Each auto mount using ```x-systemd.automount``` increases boot time significantly.

Ok, so I need to disable non-active somehow and prevent longer boot times.

Fourdee, 2016-09-10:
Trying slot system:
[image]

Fourdee, 2016-09-10:
https://github.com/Fourdee/DietPi/commit/d023af2b5040686a16681564932f73e5e18b8023

Ok, so:
- 1 automount is enabled on all DietPi systems by default
- ```dietpi-drive_manager``` will automatically enable all automounts during start
- ```dietpi-drive_manager``` will automatically disable unused automounts during exit.

k-plan, 2016-09-11:
Just fresh update to testing branch.

- one usb stick with one FAT partition connected

```
root@OrangePI-ONE:~# lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
sda           8:0    1  7.5G  0 disk
└─sda1        8:1    1  7.5G  0 part /mnt/usb_1
mmcblk0     179:0    0  7.4G  0 disk
├─mmcblk0p1 179:1    0   64M  0 part /boot
└─mmcblk0p2 179:2    0  7.4G  0 part /
```

`root@OrangePI-ONE:~# dietpi-drive_manager`

[image]

- push `<ok>` 

[image]

- exit and unmount usb device:

```
root@OrangePI-ONE:~# umount /dev/sda1

root@OrangePI-ONE:~# lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
sda           8:0    1  7.5G  0 disk
└─sda1        8:1    1  7.5G  0 part
mmcblk0     179:0    0  7.4G  0 disk
├─mmcblk0p1 179:1    0   64M  0 part /boot
└─mmcblk0p2 179:2    0  7.4G  0 part /
```

`root@OrangePI-ONE:~# dietpi-drive_manager`

[image]

- push `<ok>` 

[image]

- on exit, usb device will be automounted:

```
root@OrangePI-ONE:~# lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
sda           8:0    1  7.5G  0 disk
└─sda1        8:1    1  7.5G  0 part /mnt/usb_1
mmcblk0     179:0    0  7.4G  0 disk
├─mmcblk0p1 179:1    0   64M  0 part /boot
└─mmcblk0p2 179:2    0  7.4G  0 part /

```

- `dmesg` log output:

```
[  224.361375] ehci_irq: highspeed device disconnect
[  224.361531] usb 2-1: USB disconnect, device number 2
[  224.364445] ehci_irq: highspeed device connect
[  224.523192] ehci_irq: highspeed device disconnect
[  234.284282] ehci_irq: highspeed device connect
[  234.550074] usb 2-1: new high-speed USB device number 3 using sunxi-ehci
[  234.703687] scsi1 : usb-storage 2-1:1.0
[  235.701611] scsi 1:0:0:0: Direct-Access     IS917    innostor         1.00 PQ: 0 ANSI: 6
[  235.703685] sd 1:0:0:0: [sda] 15728640 512-byte logical blocks: (8.05 GB/7.50 GiB)
[  235.704268] sd 1:0:0:0: [sda] Write Protect is off
[  235.704300] sd 1:0:0:0: [sda] Mode Sense: 23 00 00 00
[  235.704898] sd 1:0:0:0: [sda] Write cache: disabled, read cache: disabled, doesn't support DPO or FUA
[  235.722152]  sda: sda1 sda2
[  235.726878] sd 1:0:0:0: [sda] Attached SCSI removable disk
[  405.285520] ehci_irq: highspeed device disconnect
[  405.285680] usb 2-1: USB disconnect, device number 3
[  411.113401] ehci_irq: highspeed device connect
[  411.380094] usb 2-1: new high-speed USB device number 4 using sunxi-ehci
[  411.533881] scsi2 : usb-storage 2-1:1.0
[  412.531861] scsi 2:0:0:0: Direct-Access     SanDisk  Cruzer Fit       1.00 PQ: 0 ANSI: 6
[  412.534044] sd 2:0:0:0: [sda] 15630336 512-byte logical blocks: (8.00 GB/7.45 GiB)
[  412.535404] sd 2:0:0:0: [sda] Write Protect is off
[  412.535436] sd 2:0:0:0: [sda] Mode Sense: 43 00 00 00
[  412.536391] sd 2:0:0:0: [sda] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
[  412.549260]  sda: sda1
[  412.554005] sd 2:0:0:0: [sda] Attached SCSI removable disk
```

- Tested on 2 different devices with 2 different usb sticks, same result.

But first I want to test, what's happens, if usb stick has two partitions, but it fail. (see `dmesg`)
So I do this new tests with a usb stick which has only one FAT partition.

k-plan, 2016-09-12:
@Fourdee 

[image]

:smiley: 

Now I can test, what's happens, if usb stick has two partitions.

Fourdee, 2016-09-12:
@k-plan 

> Now I can test

Excellent :+1: 

> what's happens, if usb stick has two partitions.

Only the 1st partition is mounted. 
I want to try and avoid the route of supporting multiple partitions. Makes it more complicated for the end user, and, would require alot more coding to support the partitions. We'd end up with:
```
/mnt/usb_1_p1
/mnt/usb_1_p4
/mnt/usb_3_p2
/mnt/usb_4_p1
```

#### The main goal of this script is to allow users to:
- Mount/dismount drives on the fly (Allows for safe removal of the drive.)
- Allow users to add upto 5 USB drives.
- Format drives (to 1 partition, using the whole drive)
- Assign UUID to make the mount permanent.
- Move their DietPi user data from Drive A to Drive B

k-plan, 2016-09-12:
@Fourdee 

> Makes it more complicated for the end user, and, would require alot more coding to support the partitions.

Hmm, yes Sir. More work for you and really complex coding.

> We'd end up with:

Hmm, can we do this a other way?

- automount 1. partition from fist device by default. Thinking, what's good. :+1: 

```
root@NeoPi:~# lsblk -f -i
NAME        FSTYPE  LABEL    UUID                                 MOUNTPOINT
sda
|-sda1      hfsplus Mac-hfs  2f1781ec-3622-36d4-a8f3-39d75292f694 /mnt/usb_1
`-sda2      ntfs    Win-NTFS 72E6D30371378E1D
mmcblk0
|-mmcblk0p1 vfat    BOOT     FA16-1610                            /boot
`-mmcblk0p2 ext4             dcded1b7-5f4f-44dc-b738-28376a6a3b11 /

```

- if  call up `dietpi-drive_manager`, it will show it as mounted:
[image]

- now, if user hit `Refresh`, 2. partiton will be show up  e.g. in `Slot 2`, so it is possible to mount this partition via `dietpi-drive_manager`. (`not permanent` or via `UUID` fixed)
[image]

- it's end up with:

```
root@NeoPi:~# lsblk -f -i
NAME        FSTYPE  LABEL    UUID                                 MOUNTPOINT
sda
|-sda1      hfsplus Mac-hfs  2f1781ec-3622-36d4-a8f3-39d75292f694 /mnt/usb_1
`-sda2      ntfs    Win-NTFS 72E6D30371378E1D                     /mnt/usb_2
mmcblk0
|-mmcblk0p1 vfat    BOOT     FA16-1610                            /boot
`-mmcblk0p2 ext4             dcded1b7-5f4f-44dc-b738-28376a6a3b11 /

```

And your naming is correct. We have `Slots` and `USB_0`. Great, suited. :smiley: 

:question:  - but it is only a idea ....

btw: hfs and hfsplus
- if you want to do more then only read files, it will need some more package:
`hfsplus         hfsprogs        hfsutils `

Tested, works.

Tested ntfs r/w,  works out-of-the-box. :+1:

Fourdee, 2016-09-14:
+ 
[image]

[image]

k-plan, 2016-09-14:
```
HFS+:
Recommended if you plan to use this drive on a Mac OSX system.
```
:question:

Fourdee, 2016-09-14:
@k-plan 

Means installing additional packages as you mentioned earlier. Suppose we could ```apt-get install``` it when the user selects it.

Lemme take a look. Might make the selection menu too big: https://cloud.githubusercontent.com/assets/7077026/18516333/73600bf6-7a90-11e6-9416-7b1c2b7f173a.png

Fourdee, 2016-09-14:
@k-plan 
Done, HFS+ format support is in, only issue is:
- ```mkfs.hfsplus``` does not generate a UUID for drive. 

[image]

Tried:
```
root@DietPi:~# tune2fs /dev/sda -U $(cat /proc/sys/kernel/random/uuid)
tune2fs 1.42.12 (29-Aug-2014)
tune2fs: Bad magic number in super-block while trying to open /dev/sda
Couldn't find valid filesystem superblock.
```

Might be a hfsutil command, i'am yet to find it.

k-plan, 2016-09-14:
> Means installing additional packages as you mentioned earlier.

```
root@NeoPi:~# dpkg -l | grep hfs
ii  hfsplus                        1.0.4-12.1                 armhf        Tools to access HFS+ formatted volumes
ii  hfsprogs                       332.25-11                  armhf        mkfs and fsck for HFS and HFS+ file systems
ii  hfsutils                       3.2.6-13                   armhf        Tools for reading and writing Macintosh volumes
ii  libhfsp0                       1.0.4-12.1                 armhf        Shared library to access HFS+ formatted volumes

```

> Suppose we could apt-get install it when the user selects it.

Yes!  :smiley:

Zone22, 2016-09-14:
Depending on your ability would you consider adding btrfs file system?

 Although Ext4 seems to be the better choice on a desktop system since it is already present as a default file. Btrfs has many good features for server environment and and it is robust protection against data corruption.
-on write duplication
– Writable and read-only snapshots
– Checksums on data and metadata (crc32c): this is great in my view, as every stored block is checked, so it can immediately identify and correct any data corruption
– Compression (zlib and LZO)
– SSD (Flash storage) awareness: another sign of a modern filesystem. BTRFS identifies SSD devices, and changes its behaviour automatically. First, it uses TRIM/Discard for reporting free blocks for reuse, and also has some optimisations like avoiding unnecessary seek optimisations, sending writes in clusters, even if they are from unrelated files. This results in larger write operations and faster write throughput.
– Background scrub process for finding and fixing errors on files with redundant copies
– Online filesystem defragmentation. being a COW (copy-on-write) filesystem, each time a block is updated the block itself is not overwritten but written in a different location of the device, leaving the old block still in place. If the old block at some point is not needed anymore (for example if it’s not part of any snapshot) BTRFS marks the chunk as available and ready to be reused.
– In-place conversion of existing ext3/4 file systems

The btrfs file system can also be leveraged significantly by UrBackup server and client.

k-plan, 2016-09-15:
Testing:

[image]

[image]

[image]

[image]

👎 

Do it by terminal:

```
root@NeoPi:~# lsblk -i -f
NAME        FSTYPE LABEL    UUID                                 MOUNTPOINT
sda
`-sda1      hfs    Untitled                                      /mnt/usb_1
mmcblk0
|-mmcblk0p1 vfat   BOOT     FA16-1610                            /boot
`-mmcblk0p2 ext4            dcded1b7-5f4f-44dc-b738-28376a6a3b11 /

root@NeoPi:~# mkfs.hfsplus /dev/sda1
Initialized /dev/sda1 as a 7631 MB HFS Plus volume

root@NeoPi:~# lsblk -i -f
NAME        FSTYPE  LABEL    UUID                                 MOUNTPOINT
sda
`-sda1      hfsplus untitled a810c5b5-7a4f-3a97-98d4-4ba57397f799 /mnt/usb_1
mmcblk0
|-mmcblk0p1 vfat    BOOT     FA16-1610                            /boot
`-mmcblk0p2 ext4             dcded1b7-5f4f-44dc-b738-28376a6a3b11 /

```

Fourdee, 2016-09-15:
@Zone22 

> btrfs 

Sounds good :+1: I'll try and add it in.

@k-plan 

>Hmm, why and how do you format it as HFS?

You have UUID :smile:  :+1: 

https://github.com/Fourdee/DietPi/blob/531bcb011687ea3f7a71ba92daa8bceb37a91cfb/dietpi/dietpi-drive_manager#L225-L226

When I ran just ```mkfs.hfsplus /dev/sda1```, it had no filesystem. Could be I missed something.
I'll try vfat format, then just running ```mkfs.hfsplus /dev/sda1``` as you did. Will see if UUID and format is successful.

Fourdee, 2016-09-15:
@k-plan 
Yep. 
Not sure why I did that lol. Removal of ```hformat``` did the trick, UUID now working for ```hfs+``` format. Thanks @k-plan :+1:

k-plan, 2016-09-15:
@Fourdee 

> When I ran just mkfs.hfsplus /dev/sda1, it had no filesystem. Could be I missed something.

Ahh, see your problem. If you don't have a partition table with minimum 1 partition on your device, 

```
root@NeoPi:~# lsblk -f -i
NAME        FSTYPE LABEL UUID                                 MOUNTPOINT
sda
mmcblk0
|-mmcblk0p1 vfat   BOOT  FA16-1610                            /boot
`-mmcblk0p2 ext4         dcded1b7-5f4f-44dc-b738-28376a6a3b11 /
```

`mkfs.hfs` will format `sda` without a partition.

```
root@NeoPi:~# mkfs.hfs  /dev/sda
Initialized /dev/sda as a 7632 MB HFS Plus volume

root@NeoPi:~# lsblk -f -i
NAME        FSTYPE  LABEL    UUID                                 MOUNTPOINT
sda         hfsplus untitled 3b995b7a-f17a-3c0b-aba3-76408068712b
mmcblk0
|-mmcblk0p1 vfat    BOOT     FA16-1610                            /boot
`-mmcblk0p2 ext4             dcded1b7-5f4f-44dc-b738-28376a6a3b11 /

```
And now you run in trouble ....

Fourdee, 2016-09-15:
@Zone22 
BTRFS format support is in :+1: 

Had to reduce the menu desc text a little. I'll put all that info into the online doc for DietPi-Drive_Manager and link it there.
[image]

@k-plan 

> Ahh, see your problem. If you don't have a partition table with minimum 1 partition on your device,mkfs.hfs will format sda without a partition.

That must of been what I did :+1:  lol.

k-plan, 2016-09-15:
@Fourdee 

short description behind menu please, like before?  

 https://cloud.githubusercontent.com/assets/7077026/18516333/73600bf6-7a90-11e6-9416-7b1c2b7f173a.png

Fourdee, 2016-09-15:
@k-plan 

Yep, it would be nice to have, but then the height would be too big. 
I try and keep all whiptail displays under 24 rows (its 21 currently). Mainly for SSH users so they dont need a massive SSH window :)

I'll add a URL link and insert that to menu. will cover all options, similar to: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=20#p68

k-plan, 2016-09-15:
@Fourdee 

no, you missed me. :smiley: 

```
0 : EXT4 (Recommand)
1 : FAT32 (max comp - 4GB file limit) 
2 :  NTFS (Win - high CPU load)
3 :  HFS+ (Mac OS X - best OS ever!)
4 : BTRFS (only new Linux Systems)

```

Fourdee, 2016-09-15:
@k-plan 
:+1: 
[image]

Fourdee, 2016-09-15:
@k-plan 

> https://github.com/Fourdee/DietPi/issues/271#issuecomment-247310009

Works
```
#Clear partition table
dd if=/dev/zero of=/dev/sda bs=1000000 count=1
dietpi-drive_manager
#use GUI to format HFS+
```

k-plan, 2016-09-15:
@Fourdee 

< Mmmm, not sure with` | `?

Looks okay, until ...

`   3 : HFS+   |   Mac OS X (Intel Mac default file system)`

Fourdee, 2016-09-16:
Ok, I think this now completed. Stable results during multiple testing. Marking as closed. 
Please reopen if required.

closed 2016-09-16 by Fourdee