MichaIng/DietPi #273: DietPi-System | C1+ | Issue to setup WLAN device

k-plan, 2016-04-11:
http://fuzon.co.uk/phpbb/viewtopic.php?f=11&t=368

Fourdee, 2016-04-11:
@k-plan 
Thanks for the report :+1: Excellent write up.

I've got 2 different WiFi dongles, i will try them on my C1 and see if I get same results.

Fourdee, 2016-04-11:
@k-plan 
Just tried a generic chinatown WiFi adapter. It doesnt appear to work on DHCP. But setting it to static did the trick.

I cant see why this is C1 specific. ```dietpi-config``` uses the same code to control the adapters. It may be a bug in the kernel/modules, or even DietPi. I'll do some more testing and see if we can find the cause.

Fourdee, 2016-04-11:
@k-plan 
What PSU are you using for the C1+?

k-plan, 2016-04-11:
@Fourdee 

> What  PSU are you using for the C1+?

I have tried different for testing.

Normally I use: [Anker PowerPort 10](http://www.amazon.co.uk/Compact-10-Port-Anker-PowerPort-Charging/dp/B00YSA0WI8) (60W 10-Port USB) 5V - 2.4A per Port or 12A total.

For special  WiFi adapter testing: [HN Power HNP18-050](http://www.watterott.com/de/Netzteil-5V-3A) - 5 V/DC 3 A 15 W

As well tested WiFi adapter additionally active powered USB hub [Logitech H-UE5](http://www.amazon.de/Logitech-Premium-4-Port-USB-Notebook/dp/B000U1MQH4) witch PSU 5 V/DC 2.5 A

> I cant see why this is C1 specific.

It  take a long time, until I have notion, to test it on DietPi Raspberry Pi B+. But with this, it works instantly.

> a bug in the kernel/modules ...
😢  please no ....

> It doesnt appear to work on DHCP. But setting it to static did the trick.

Will test it. You will get response.

Thanks for your help. 👍

k-plan, 2016-04-11:
@Fourdee 

```
root@oDroid-C1_plus-test:~# lsusb
Bus 001 Device 003: ID 7392:7822 Edimax Technology Co., Ltd
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

root@oDroid-C1_plus-test:~# lsmod
Module                  Size  Used by
nls_cp437               5134  1
8192cu                523594  0
w1_gpio                 3178  0
wire                   20357  1 w1_gpio
aml_gpio                7315  0
fuse                   72461  1
autofs4                22325  2

```
Set WiFi to static breaks all my network connections ....

Must do a `reboot` via keyboard.

After this I tried:

```
root@oDroid-C1_plus-test:~# ifup wlan0
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan0 ; Operation not permitted.
Error for wireless request "Set Encode" (8B2A) :
    invalid argument "000test-12345".
RTNETLINK answers: File exists
Failed to bring up wlan0.
```

`000test-12345` is my wireless-key (WEP), but my WiFi network only support WPA2 PSK !

```
root@oDroid-C1_plus-test:~# dmesg
 . . .
[  332.648763] WARN::hc_xfer_timeout:2719: hc_xfer_timeout: timeout on channel 4
[  332.686034] WARN::hc_xfer_timeout:2721:      start_hcchar_val 0x00d88a00
[  332.723060] WARN::hc_xfer_timeout:2726:      chn-4,ep1-IN:type:2,speed:2,len:15360,addr3

[  370.588842] WARN::hc_xfer_timeout:2719: hc_xfer_timeout: timeout on channel 2
[  370.627114] WARN::hc_xfer_timeout:2721:      start_hcchar_val 0x00d88a00
[  370.664335] WARN::hc_xfer_timeout:2726:      chn-2,ep1-IN:type:2,speed:2,len:15360,addr3
```

Default locale for the system environment:
```
root@oDroid-C1_plus-test:~# dpkg-reconfigure locales
Generating locales (this might take a while)...
  en_GB.UTF-8... done
Generation complete.
```

Hmmm ... think I will start from the scratch.

Fourdee, 2016-04-12:
@k-plan 
Did another test on my Edimax:
```
Bus 001 Device 003: ID 7392:7811 Edimax Technology Co., Ltd EW-7811Un 802.11n Wireless Adapter [Realtek RTL8188CUS]
```

Cant get it to connect, but scans fine. Couldn't find anything obvious in ```dmesg```. It seems to fail receiving data during the DHCP request:
```
root@DietPi:~# ifup wlan0
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan0 ; Operation not permitted.
Internet Systems Consortium DHCP Client 4.3.1
Copyright 2004-2014 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/wlan0/74:da:38:26:9d:65
Sending on   LPF/wlan0/74:da:38:26:9d:65
Sending on   Socket/fallback
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 4
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 9
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 15
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 7
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 7
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 13
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 6
No DHCPOFFERS received.
No working leases in persistent database - sleeping.
```

It looks like a kernel issue and/or modules. I found this ticket on the Odroid C1 Kernel: https://github.com/hardkernel/linux/issues/179

When I get some time, I will compile the C1 kernel/modules and retest. If you want to try it also, instructions are here: http://odroid.com/dokuwiki/doku.php?id=en:c1_building_kernel

k-plan, 2016-04-12:
@Fourdee 

> Cant get it to connect, but scans fine. 

Yes, all my tested adapters do scans fine, but no WiFi link connected.

> ```
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan0 ; Operation not permitted.
```
You can test:
Comment Power Management out in `/etc/network/interfaces`

and set  ' nano /etc/modprobe.d/8192cu.conf  ' with content:

`options 8192cu rtw_power_mgnt=0 rtw_enusbss=0`

now:  `service networking restart`   ([Link](https://blog.rotzoll.net/2014/07/raspberry-pi-wlan-client-mit-edimax-ew-7811un-usb-wlan-stick/))

> When I get some time, I will compile the C1 kernel/modules and retest.
Nice, thanks! I have to waiting for this.
Sorry, but compile "my own kernel" overcharged me.No skill, hardware or software.

Where is also a [rtl8192cu-fixes](https://github.com/pvaret/rtl8192cu-fixes) USB WiFi driver, but I can't get it on work.
This hint is form Igor Pečovnik. A well tried his new[ Odroid C1 Armbian image](http://forum.armbian.com/index.php/topic/849-armbian-for-odroid-c1-and-c2/page-2#entry6992), but it will not boot for me.

k-plan, 2016-04-13:
@Fourdee 

don't think it's the module.

ubuntu-14.04.3lts-lubuntu-odroid-c1-20151020 => Edimax EW-7612UAn V2  working with dhcp 

```
root@odroid-c1-Plus-test:~# uname -a
Linux odroid-c1-Plus-test 3.10.96-148 #1 SMP PREEMPT Thu Apr 7 10:45:50 BRT 2016 armv7l armv7l armv7l GNU/Linux

root@odroid-c1-Plus-test:~# modinfo 8192cu
filename:       /lib/modules/3.10.96-148/kernel/backports/drivers/realtek/8192cu/8192cu.ko
version:        v4.0.2_9000.20130911
author:         Realtek Semiconductor Corp.
description:    Realtek Wireless Lan Driver
license:        GPL
version:        backported from Linux (next-20150129-0-g828f79f) using backports backports-20150129-0-gdd4a670
srcversion:     92ABD77EE49D06D8157B4F8

```

DietPi_v109_Odroid-C1-(Jessie) auto updatesd to v115 => Edimax EW-7612UAn V2  working with dhcp

```
root@oDroid-C1-Plus-TEST:~# uname -a
Linux oDroid-C1-Plus-TEST 3.10.80 #1 SMP PREEMPT Sun Mar 13 15:50:53 CET 2016 armv7l GNU/Linux

root@oDroid-C1-Plus-TEST:~# modinfo 8192cu
filename:       /lib/modules/3.10.80/kernel/backports/drivers/realtek/8192cu/8192cu.ko
version:        v4.0.2_9000.20130911
author:         Realtek Semiconductor Corp.
description:    Realtek Wireless Lan Driver
license:        GPL
version:        backported from Linux (next-20150129-0-g828f79f) using backports backports-20150129-0-gdd4a670
srcversion:     92ABD77EE49D06D8157B4F8

```

Fourdee, 2016-04-13:
@k-plan 

> Comment Power Management out in 

Yep, that will prevent the error occurring, but theres no harm in having it there. Its used by some adapters.

All DietPi images having the following:
```
#wifi dongles
echo -e "options 8192cu rtw_power_mgnt=0" > /etc/modprobe.d/8192cu.conf
echo -e "options 8188eu rtw_power_mgnt=0" > /etc/modprobe.d/8188eu.conf
```
Prevents those common WiFi modules from going into powersaving mode (up and down, up then down lol).

I'll compile the C1 kernel today. As per you previous post, modules appear identical.

Fourdee, 2016-04-13:
Native compile (CC not setup). Can't get uImage to compile:
```
root@DietPi:~/linux# make uImage
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
make[1]: 'include/generated/mach-types.h' is up to date.
  CALL    scripts/checksyscalls.sh
  CC      scripts/mod/devicetable-offsets.s
  GEN     scripts/mod/devicetable-offsets.h
  HOSTCC  scripts/mod/file2alias.o
  HOSTLD  scripts/mod/modpost
  CHK     include/generated/compile.h
  CHK     kernel/config_data.h
  CC      drivers/amlogic/display/osd/osd_main.o
drivers/amlogic/display/osd/osd_main.c: In function ‘osd_ioctl’:
drivers/amlogic/display/osd/osd_main.c:382:4: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
    aml_hwc_addr_t para;
    ^
drivers/amlogic/display/osd/osd_main.c:392:11: error: implicit declaration of function ‘osd_wait_vsync_hw’ [-Werror=implicit-function-declaration]
           osd_wait_vsync_hw();
           ^
cc1: some warnings being treated as errors
scripts/Makefile.build:308: recipe for target 'drivers/amlogic/display/osd/osd_main.o' failed
make[4]: *** [drivers/amlogic/display/osd/osd_main.o] Error 1
scripts/Makefile.build:455: recipe for target 'drivers/amlogic/display/osd' failed
make[3]: *** [drivers/amlogic/display/osd] Error 2
scripts/Makefile.build:455: recipe for target 'drivers/amlogic/display' failed
make[2]: *** [drivers/amlogic/display] Error 2
scripts/Makefile.build:455: recipe for target 'drivers/amlogic' failed
make[1]: *** [drivers/amlogic] Error 2
Makefile:803: recipe for target 'drivers' failed
make: *** [drivers] Error 2
root@DietPi:~/linux# find -name "*Image*"
./arch/powerpc/boot/zImage.lds.S
./arch/powerpc/boot/zImage.coff.lds.S
./arch/powerpc/boot/zImage.ps3.lds.S
```

Code to compile:
```
apt-get install build-essential git ncurses-dev lzop u-boot-tools -y

git clone --depth 1 https://github.com/hardkernel/linux.git -b odroidc-3.10.y
cd linux
make odroidc_defconfig

#make menuconfig
make -j $(nproc --all)
make -j $(nproc --all) modules

make modules_install

make uImage

cp arch/arm/boot/uImage /boot/
cp cp arch/arm/boot/dts/meson8b_odroidc.dts /boot/
```

Fourdee, 2016-04-13:
Restarted from scratch, uImage compiled fine:
```
Image Name:   Linux-3.10.96
Created:      Wed Apr 13 13:36:17 2016
Image Type:   ARM Linux Kernel Image (lzo compressed)
Data Size:    5466661 Bytes = 5338.54 kB = 5.21 MB
Load Address: 00208000
Entry Point:  00208000
  Image arch/arm/boot/uImage is ready
```

Fourdee, 2016-04-13:
Same issues with WiFi on this kernel.

k-plan, 2016-04-13:
@Fourdee 

> Same issues with WiFi on this kernel.

fu ...  :cry:  Sorry

	Anyway, many thanks for your help and support. :+1: 

Does cursor work on display with this kernel? If you like, give me your kernel. I will test it for cursor issue.

Fourdee, 2016-04-13:
@k-plan 
Running headless.

Once I finished this compile (did some testing), i'll upload it.

Fourdee, 2016-04-13:
@k-plan 
It might be worth trying Meveric's image (which DietPi for C1 is based on). Just to see if we can replicate the issue.
http://forum.odroid.com/viewtopic.php?f=96&t=17542#p114709.

k-plan, 2016-04-13:
@Fourdee 

yes, I can do so, but really Meveric's image you linked to?  
It is: Debian-Jessie-1.0~RC2-20151123-**XU3**.img.xz

Or do you like :  http://forum.odroid.com/viewtopic.php?f=114&t=17569
Debian-Jessie-1.0-20160131-**C1**.img.xz

Sorry, but only to be sure ...

Fourdee, 2016-04-13:
@k-plan 
Yep my bad, long day:
http://forum.odroid.com/viewtopic.php?f=114&t=17569

Fourdee, 2016-04-13:
@k-plan 
Uploaded zip http://dietpi.com/downloads/testing/

Not sure about ```/boot/uInitrd```. It wasn't compiled (check modified date), or mentioned anywhere in the documentation: http://odroid.com/dokuwiki/doku.php?id=en:c1_building_kernel

k-plan, 2016-04-14:
@Fourdee 

> Yep my bad, long day

No problem, here as well.

Debian-Jessie-1.0-20160131-C1.img => downloaded with 64 kb/s !!!  (what a mess)

```
root@odroid-jessie:~# uname -a
Linux odroid-jessie 3.10.80 #17 SMP PREEMPT Sat Oct 31 12:44:33 CET 2015 armv7l GNU/Linux

root@odroid-jessie:~#
Message from syslogd@localhost at Apr 14 18:11:48 ...
 kernel:[  243.938657@1] Kernel panic - not syncing: hung_task: blocked tasks
 
 [  663.414395] usb 1-1.2: new high-speed USB device number 4 using dwc_otg
[  663.667936] usbcore: registered new interface driver rtl8192cu
```

Done:   "`root@odroid-jessie:~# iwlist wlan0 scanning` "  => works!  👍 

____

Done:  " `iface wlan0 inet dhcp` "

```
root@odroid-jessie:~# ifup wlan0
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan0 ; Operation not permitted.
Error for wireless request "Set Encode" (8B2A) :
    invalid argument "Not_Set".
Internet Systems Consortium DHCP Client 4.3.1
Copyright 2004-2014 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/wlan0/74:da:38:23:1c:aa
Sending on   LPF/wlan0/74:da:38:23:1c:aa
Sending on   Socket/fallback
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 8
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 13
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 12
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 18
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 8
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 2
No DHCPOFFERS received.
No working leases in persistent database - sleeping.
```
👎 

____

Done: "  `iface wlan0 inet static` "

```
root@odroid-jessie:~# ifup wlan0
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan0 ; Operation not permitted.
RTNETLINK answers: File exists
```
👎 
____

All tests as well tested  with another WiFi-Roter (Channel 4) to connect => also fail! 👎

k-plan, 2016-04-14:
Testing kernel with DietPi v115

```
root@oDroid-C1-Plus-TEST:~# uname -a
Linux oDroid-C1-Plus-TEST 3.10.96 #2 SMP PREEMPT Wed Apr 13 15:35:19 BST 2016 armv7l GNU/Linux
```
As well missing/invisible cursor on monitor/display !

Cant get it to connect ip via dhcp or static, but scans fine. 
As well no connect to second Test WFfi-Router.

result:
forget Wifi, WiFi HotSpot or TOR HotSpot on Odroid C1+ (for the moment)
If you want/need it, better will buy a Raspberry Pi.

@Fourdee 
Sad, not to be able to get it on work.
But it is not your fault. Sorry, you have done more than enough. 
I have fiddling around with a lot images/distris for Odroid C1 in the past days.
Thus, I can tell you, DietPi is the best in my hand so far.
Forget WiFi support on C1 for v.116. 
Maybe there is another in future, perhaps armbian. We will see.

> Not sure about /boot/uInitrd.

You can have a look here, but I  don't know if it is helpful. http://forum.odroid.com/viewtopic.php?f=111&t=7965#p61873

```
root@oDroid-C1-Plus-TEST:~# /etc/kernel/postinst.d/initramfs-tools 3.10.96
update-initramfs: Generating /boot/initrd.img-3.10.96
Image Name:   uInitrd
Created:      Thu Apr 14 18:51:43 2016
Image Type:   ARM Linux RAMDisk Image (uncompressed)
Data Size:    3887133 Bytes = 3796.03 kB = 3.71 MB
Load Address: 00000000
Entry Point:  00000000
/etc/initramfs-tools/scripts/local-top/c1_init.sh: 24: /etc/initramfs-tools/scripts/local-top/c1_init.sh: cannot create /sys/class/ppmgr/ppscaler: Directory nonexistent
/etc/initramfs-tools/scripts/local-top/c1_init.sh: 83: /etc/initramfs-tools/scripts/local-top/c1_init.sh: cannot create /sys/class/net/eth0/queues/rx-0/rps_flow_cnt: Directory nonexistent
/etc/initramfs-tools/scripts/local-top/c1_init.sh: 84: /etc/initramfs-tools/scripts/local-top/c1_init.sh: cannot create /sys/class/net/eth0/queues/rx-0/rps_cpus: Directory nonexistent
/etc/initramfs-tools/scripts/local-top/c1_init.sh: 85: /etc/initramfs-tools/scripts/local-top/c1_init.sh: cannot create /sys/class/net/eth0/queues/tx-0/xps_cpus: Directory nonexistent
```

Fourdee, 2016-04-15:
@k-plan 
Found the issue :+1: 
I compared Wifi packages across C1 and C2. Seems we were missing some on the C1:

C2
```
root@DietPi:~# dpkg -l | grep wire
ii  crda                           3.13-1                    arm64        wireless Central Regulatory Domain Agent
ii  firmware-atheros               0.43                      all          Binary firmware for Atheros wireless cards
ii  firmware-brcm80211             0.43                      all          Binary firmware for Broadcom 802.11 wireless cards
ii  firmware-ralink                0.43                      all          Binary firmware for Ralink wireless cards
ii  firmware-realtek               0.43                      all          Binary firmware for Realtek wired and wireless network adapters
ii  iw                             3.17-1                    arm64        tool for configuring Linux wireless devices
ii  wireless-regdb                 2014.11.18-1              all          wireless regulatory database
ii  wireless-tools                 30~pre9-8                 arm64        Tools for manipulating Linux Wireless Extensions
root@DietPi:~# dpkg -l | grep wpa
ii  wpasupplicant                  2.3-1+deb8u3              arm64        client support for WPA and WPA2 (IEEE 802.11i)

```

Install the following missing packages on your C1, reboot, then try reconnecting wifi
```
 apt-get install wireless-regdb iw crda wpasupplicant
```

```
root@DietPi:~# ifup wlan0
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan0 ; Operation not permitted.
ioctl[SIOCSIWAP]: Operation not permitted
ioctl[SIOCSIWENCODEEXT]: Invalid argument
ioctl[SIOCSIWENCODEEXT]: Invalid argument
Internet Systems Consortium DHCP Client 4.3.1
Copyright 2004-2014 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/wlan0/74:da:38:26:9d:65
Sending on   LPF/wlan0/74:da:38:26:9d:65
Sending on   Socket/fallback
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 6
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 9
DHCPREQUEST on wlan0 to 255.255.255.255 port 67
DHCPOFFER from 192.168.0.1
DHCPACK from 192.168.0.1
bound to 192.168.0.101 -- renewal in 686767077 seconds.
```

Fourdee, 2016-04-15:
@k-plan 
Ran a 2nd test to confirm, working fine.
The missing packages will be installed during the patch in the v116 update. 

If you can confirm the issue is resolved, let me know so we can close this ticket :+1:

k-plan, 2016-04-15:
@Fourdee (Coding Beast 😄 )

> Found the issue :+1: Always the simple things, cant believe I missed it.

Will it be so easy? I can't believe. 
Clean automate install from the scratch is running now.
But  http://oph.mdrjr.net/ is so slow (max. 64 kB/s)

Yes, I have tested before to install iw and same other wireless/wifi packages without success.

On Meveric's image I have to install, iw, usb-tools, wireless-tools and same more, to get scanning on work.

wireless-regdb ,  crda I have never install. I will test it.

k-plan, 2016-04-15:
@Fourdee 

- Install v.109 from the scratch.

- install all packages you propose: `apt-get install wireless-regdb iw crda wpasupplicant`

- Plugged in the WiFi dongle, no detection with lsusb, then kernel panic and freeze. Must do  a power cycle.

Because it is v.109, configuration with keyboard and monitor as infrastructure wifi network with dhcp.

```
root@oDroid-C1-Plus-Test:~# ifup wlan0
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan0 ; Operation not permitted.
Error for wireless request "Set Encode" (8B2A) :
    invalid argument "test-12345".
ioctl[SIOCSIWAP]: Operation not permitted
ioctl[SIOCSIWENCODEEXT]: Invalid argument
ioctl[SIOCSIWENCODEEXT]: Invalid argument
Internet Systems Consortium DHCP Client 4.3.1
Copyright 2004-2014 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/wlan0/74:da:38:23:1c:aa
Sending on   LPF/wlan0/74:da:38:23:1c:aa
Sending on   Socket/fallback
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 3
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 3
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 6
DHCPREQUEST on wlan0 to 255.255.255.255 port 67
DHCPOFFER from 192.168.88.1
DHCPACK from 192.168.88.1
bound to 192.168.88.50 -- renewal in 233 seconds.

```

it works! 👍 

---

Update to v.116:

 `root@oDroid-C1-Plus-Test:~# dietpi-update`

After reboot configuration wifi with dhcp via ` dietpi-config` and scanning:

[image]

It works! 😄  👍 

```
root@oDroid-C1-Plus-Test:~# ifup wlan0
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan0 ; Operation not permitted.
Error for wireless request "Set Encode" (8B2A) :
    invalid argument "test-12345".
ioctl[SIOCSIWAP]: Operation not permitted
ioctl[SIOCSIWENCODEEXT]: Invalid argument
ioctl[SIOCSIWENCODEEXT]: Invalid argument
Internet Systems Consortium DHCP Client 4.3.1
Copyright 2004-2014 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/wlan0/74:da:38:23:1c:aa
Sending on   LPF/wlan0/74:da:38:23:1c:aa
Sending on   Socket/fallback
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 3
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 3
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 6
DHCPREQUEST on wlan0 to 255.255.255.255 port 67
DHCPOFFER from 192.168.88.1
DHCPACK from 192.168.88.1
bound to 192.168.88.50 -- renewal in 233 seconds.

```
---

Configuration wifi with static IP via ` dietpi-config` and scanning:

[image]

```
root@oDroid-C1-Plus-Test:~# ifdown wlan0
ifdown: interface wlan0 not configured

root@oDroid-C1-Plus-Test:~# ifup wlan0
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan0 ; Operation not permitted.
Error for wireless request "Set Encode" (8B2A) :
    invalid argument "test-12345".
ioctl[SIOCSIWAP]: Operation not permitted
ioctl[SIOCSIWENCODEEXT]: Invalid argument
ioctl[SIOCSIWENCODEEXT]: Invalid argument
Error: an inet prefix is expected rather than "Adresse:192.168.88.100/Maske:255.255.255.0".
Failed to bring up wlan0.

```

😢

k-plan, 2016-04-15:
@Fourdee 

Now installed DietPi WiFi HotSpot. Tested. Works!  😄     👍 

[image]

Installed DietPi  Tor WiFi Hotspot. Tested. Works!   😄      👍 

Should I test some of the other USB Wireless adapters?

Fourdee, 2016-04-15:
@k-plan 

> Configuration wifi with static IP via dietpi-config and scanning:
https://cloud.githubusercontent.com/assets/17237601/14573615/64fa46fa-0356-11e6-9390-7536ab7ee562.gif

Looks like a possible locale issue we might need to fix. Static addresses shouldn't be pulling in any text, just the IP. Can you set this to static, apply, exit, then paste results of ```/etc/network/interfaces```.

Also, what locale are you running? (eg: en_GB, de_DE)

Fourdee, 2016-04-16:
@k-plan 
Did some testing, appears we need to apply the same fix we did here: https://github.com/Fourdee/DietPi/issues/233#issuecomment-196413520.

By forcing ```en_GB```, all our grep parsing should work regardless of system locale.
```
#!/bin/bash
LANG=en_GB.UTF-8
ifconfig

test(){

ifconfig
date
}

test
```

So I'am going to look at adding ```LANG=en_GB.UTF-8``` to all the DietPi scripts.

Fourdee, 2016-04-16:
@k-plan 
Ok, we will start with the main scripts 1st:
```
dietpi-config dietpi-software dietpi-cloudshell
```
Has been added to testing branch: https://github.com/Fourdee/DietPi/commit/aaf33009c717b42cab74353c0324590dbd8af182

I'll run some tests my end. If you can do a fresh install and test the static IP's again, i'd be greatful.

Fourdee, 2016-04-16:
@k-plan 
Working fine my end using ```de_DE``` system locale (minus the UTF char visual issues with whiptail):
[image]

k-plan, 2016-04-16:
@Fourdee 

sorry for delay. Saturday = shopping and family-visit-day.

> Also, what locale are you running? (eg: en_GB, de_DE)

Last testing : **de_DE **, en_GB, C

[image]

All before: without touching  `dpkg-reconfigure locales` so it was  en_GB (Dietpi) or en_US (Meveric's image) or en_AU (Ubuntu Odroid)

> `/etc/network/interfaces`

Content is the same, equal of change in locales.

Testing with en_GB and wifi static IP (v.116) via dietpi-config:

[image]

(and where are some red messages at the end, while restarting the network. But I can't read it so quick)

```
root@oDroid-C1-Plus-Test:~# ifdown wlan0
ifdown: interface wlan0 not configured

root@oDroid-C1-Plus-Test:~# ifup wlan0
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan0 ; Operation not permitted.
Error for wireless request "Set Encode" (8B2A) :
    invalid argument "test-12345".
ioctl[SIOCSIWAP]: Operation not permitted
ioctl[SIOCSIWENCODEEXT]: Invalid argument
ioctl[SIOCSIWENCODEEXT]: Invalid argument
RTNETLINK answers: File exists
Failed to bring up wlan0.

root@oDroid-C1-Plus-Test:~# cat /etc/network/interfaces
#/etc/network/interfaces
#Please use DietPi-Config to modify network settings.

# Local
auto lo
iface lo inet loopback

# Ethernet
auto eth1
iface eth1 inet static
address 192.168.0.54
netmask 255.255.255.0
gateway 192.168.0.3
dns-nameservers 192.168.0.3 192.168.0.1

# Wifi
auto wlan0
iface wlan0 inet static
address 192.168.88.101
netmask 255.255.255.0
gateway 192.168.88.1
wireless-essid T2-Net
wireless-key test-12345
wireless-mode Managed
wireless-power off
wpa-ssid T2-Net
wpa-psk test-12345
dns-nameservers 192.168.88.1

```

TWO DNS server?

>  If you can do a fresh install and test the static IP's again, i'd be greatful.

Will do, but this will have some time, because of the slow kernel update from  http://oph.mdrjr.net/ ...

k-plan, 2016-04-16:
@Fourdee 

comment wifi dns-nameservers out be hand and restart networking:

```
root@oDroid-C1-Plus-Test:~# service networking restart

root@oDroid-C1-Plus-Test:~# ifdown wlan0
ifdown: interface wlan0 not configured

root@oDroid-C1-Plus-Test:~# ifup wlan0
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan0 ; Operation not permitted.
Error for wireless request "Set Encode" (8B2A) :
    invalid argument "test-12345".
ioctl[SIOCSIWAP]: Operation not permitted
ioctl[SIOCSIWENCODEEXT]: Invalid argument
ioctl[SIOCSIWENCODEEXT]: Invalid argument
RTNETLINK answers: File exists
Failed to bring up wlan0.
```

Staring from the scratch now ....

k-plan, 2016-04-16:
@Fourdee 

running automated installation. Kernel panic and freeze while automated apt-get update ...

manual power cycle

k-plan, 2016-04-16:
and the next  Kernel panic and freeze ....

trying now: `dietpi_check_for_updates=0`

k-plan, 2016-04-16:
install without automated to v.109. Set static IP for eth1. Finished installation.  Update to v115. Reboot.

manual setting and update to testing branch. Reboot.

After restart plugged in Edimax EW-7612UAn V2 . It will not recognize.

```
[    9.069219] [adjust link] -> eth: full-duplex
[    9.070037] [adjust link] -> eth: phy_speed <> priv_speed)
[    9.070960] [adjust link -> eth: am_adjust_link state change (new_state=true)
[    9.072049] libphy: 0:01 - Link is Up - 1000/Full
[    9.072955] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready

root@DietPi:~# lsusb
Bus 001 Device 003: ID 045e:0800 Microsoft Corp.
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

root@DietPi:~# lsmod
Module                  Size  Used by
nls_cp437               5134  1
w1_gpio                 3178  0
wire                   20357  1 w1_gpio
aml_gpio                7315  0
fuse                   72461  1
autofs4                22325  2
```

Starting `dietpi-config`:

[image]

and ... Kernel panic and freeze .... manual power cycle

Fourdee, 2016-04-16:
@k-plan 

> sorry for delay. Saturday = shopping and family-visit-day.

No worries, family comes 1st and enjoy the weekend :+1: 

If you are getting Kernel panics and freezing, sounds like your SDcard is corrupt and/or PSU instability. This shouldn't be occurring. I'd suggest trying another SDcard if you can. If not, a picture of the screen with those erorrs on would be helpful to try and debug it.

k-plan, 2016-04-16:
@Fourdee 

After a reboot with plugged in Edimax EW-7612UAn V2, starting dietpi-config:

enable WiFi, set only static IP, mask and gateway, scanning (works) but nothing.

[image]

Yes, it is all default, no changing any stetting, language, install extra software. Only things i have done, set a static IP on eth1 and update to testing branch.

```
root@DietPi:~# dpkg-reconfigure locales
Generating locales (this might take a while)...
  en_GB.UTF-8... done
Generation complete.
root@DietPi:~# uname -a
Linux DietPi 3.10.80 #1 SMP PREEMPT Sun Mar 13 15:50:53 CET 2016 armv7l GNU/Linux
root@DietPi:~# cat /DietPi/dietpi/.version
116
```

Wifi with DHCP is working, but only you done it twice via `dietpi-config`:

1.)

```
root@DietPi:~# ip a
. . .
5: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 74:da:38:73:2c:aa brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.100/24 brd 192.168.0.255 scope global wlan0
       valid_lft forever preferred_lft forever
    inet 192.168.88.100/24 brd 192.168.88.255 scope global wlan0
       valid_lft forever preferred_lft forever
    inet 192.168.88.50/24 brd 192.168.88.255 scope global secondary wlan0
       valid_lft forever preferred_lft forever
    inet6 fe80::76da:38ff:fe73:2caa/64 scope link
       valid_lft forever preferred_lft forever
```

2.)

```
root@DietPi:~# ip a
. . .
5: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 74:da:38:73:2c:aa brd ff:ff:ff:ff:ff:ff
    inet 192.168.88.50/24 brd 192.168.88.255 scope global wlan0
       valid_lft forever preferred_lft forever
    inet6 fe80::76da:38ff:fe73:2caa/64 scope link
       valid_lft forever preferred_lft forever
```

k-plan, 2016-04-16:
>  PSU instability. 

Sorry, no. (siehe oben)

> SDcard is corrupt 

No, it's the same, I used manny times before (fucking ScanDisk Ultra 16GB). And yes, I formatting the SDcard every time before writting a image on it.
But only you be sure, I will use a other now. But it is the last try ...

Fourdee, 2016-04-16:
@k-plan 
No worries, leave it with me. I'll run a few tests my end.

k-plan, 2016-04-16:
@Fourdee 

please don't miss understand me, i really want to help, to get it on work.
Sorry , a long day and English not my preferred language. So it is really difficult for me.

Please make your choice:

[image]

😄

Fourdee, 2016-04-16:
@k-plan 

Nice collection, Sandisk Extreme looking nice :+1: You want to see mine? :)

I understand and appreciate the help :).  I want to make sure you are testing the right areas, at the moment i'am doing all the DietPi images which takes time: https://github.com/Fourdee/DietPi/issues/278#issue-148344897. 
Once they are done, I will come back here and refresh my memory on what it is I would like you to test. Not to waste your time :)

k-plan, 2016-04-16:
@Fourdee 

> Sandisk Extreme looking nice

nice looking, but a lot of freeze and kernel panic  

>  You want to see mine? :)

Yes for sure. 
But then I had to turn out the rest of my sd-cards, who are in use in the other devices. 
Sorry, only  joke around.

> Not to waste your time :)

Do nothing but whine , is as well waste of time.

So I choice SD-Card Verbatin 8GB, the lowest in the sample.

1. formatting and burn DietPi v.109 image on it and decided to do all config manually (default boot.ini and dietpi.txt)
2. starting up in C1+ , dietpi-update form v109 => v.115 will started automatically
3. auto reboot, after this, finished installation without any personally choice, auto apt-get update started  and finished successful the installation with a  auto reboot
4. after restart,  `nano /DietPi/dietpi.txt` with `gitbranch=testing` , then  `dietpi-update` v.115 => v.116 
[image]

5. `Please reboot your system, using the command reboot` but done a `poweroff` and plugged in my Edimax EW-7612UAn V2
6. power cycle, and have to check some message

```
[    3.334554] checked enable EXT4 DISCARD here
[    3.336610] EXT4-fs (mmcblk0p2): mounting with "discard" option, but the device does not support discard
[    3.338645] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)
. . .
[    5.569889] usbcore: registered new interface driver rtl8192cu

root@DietPi:~# lsusb
Bus 001 Device 004: ID 045e:0800 Microsoft Corp.
Bus 001 Device 003: ID 7392:7822 Edimax Technology Co., Ltd
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

root@DietPi:~# lsmod
Module                  Size  Used by
nls_cp437               5134  1
8192cu                540446  0
w1_gpio                 3178  0
wire                   20357  1 w1_gpio
aml_gpio                7315  0
fuse                   72461  1
autofs4                22325  2

```
looks fine (for me)

8. `dietpi-config ` and enable WiFi adapter 
[image]
9. `Scan and Connect` with DHCP => works! 
[image]
10. `Scan and Connect` with static IP => nothing
[image]

```
[  571.220545] rtw_wx_set_mlme
[  571.224316] rtw_wx_set_mlme, cmd=0, reason=3
[  572.918064] rtw_wx_set_mlme
[  572.919617] rtw_wx_set_mlme, cmd=0, reason=3

root@DietPi:~# ifdown wlan0
ifdown: interface wlan0 not configured

root@DietPi:~# ifup wlan0
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan0 ; Operation not permitted.
Error for wireless request "Set Encode" (8B2A) :
    invalid argument "test-12345".
ioctl[SIOCSIWAP]: Operation not permitted
ioctl[SIOCSIWENCODEEXT]: Invalid argument
ioctl[SIOCSIWENCODEEXT]: Invalid argument
RTNETLINK answers: File exists
Failed to bring up wlan0.

root@DietPi:~# ip a
. . .
5: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether 74:da:38:73:2c:aa brd ff:ff:ff:ff:ff:ff
    inet 192.168.88.101/24 brd 192.168.88.255 scope global wlan0
       valid_lft forever preferred_lft forever
       
root@DietPi:~# cat /etc/network/interfaces
#/etc/network/interfaces
#Please use DietPi-Config to modify network settings.

# Local
auto lo
iface lo inet loopback

# Ethernet
auto eth1
iface eth1 inet dhcp
address 192.168.0.100
netmask 255.255.255.0
gateway 192.168.0.1
#dns-nameservers 8.8.8.8 8.8.4.4

# Wifi
auto wlan0
iface wlan0 inet static
address 192.168.88.101
netmask 255.255.255.0
gateway 192.168.88.1
wireless-essid T2-Net
wireless-key test-12345
wireless-mode Managed
wireless-power off
wpa-ssid T2-Net
wpa-psk test-12345
dns-nameservers 8.8.8.8 8.8.4.4root@DietPi:~#
```
Is it possible, that line break is missing in `/etc/network/interfaces` ?

> Once they are done, I will come back here and refresh my memory on what it is I would like you to test. 

Yes, please do so. I will go to make a fresh coffee now. 😄

Fourdee, 2016-04-16:
@k-plan 

> 3. Scan and Connect with static IP => nothing 

Are you able to ping the ip address?

k-plan, 2016-04-16:
@Fourdee 

> Are you able to ping the ip address?

```
root@DietPi:~# ping 192.168.88.101
PING 192.168.88.101 (192.168.88.101) 56(84) bytes of data.
64 bytes from 192.168.88.101: icmp_seq=1 ttl=64 time=0.074 ms
64 bytes from 192.168.88.101: icmp_seq=2 ttl=64 time=0.453 ms
64 bytes from 192.168.88.101: icmp_seq=3 ttl=64 time=0.738 ms
64 bytes from 192.168.88.101: icmp_seq=4 ttl=64 time=0.484 ms
64 bytes from 192.168.88.101: icmp_seq=5 ttl=64 time=0.458 ms
64 bytes from 192.168.88.101: icmp_seq=6 ttl=64 time=0.450 ms
64 bytes from 192.168.88.101: icmp_seq=7 ttl=64 time=0.444 ms
^C
--- 192.168.88.101 ping statistics ---
7 packets transmitted, 7 received, 0% packet loss, time 6009ms
rtt min/avg/max/mdev = 0.074/0.443/0.738/0.179 ms
root@DietPi:~#
```
Yes, but this indicate only that ip stack is running correct, not interface wlan0 it self. 

from outside (Wifi-Router)

```
[admin@RB_mAP-lite] > ping 192.168.88.101
  SEQ HOST                                     SIZE TTL TIME  STATUS             
    0 192.168.88.101                                          timeout            
    1 192.168.88.101                                          timeout            
    2 192.168.88.101                                          timeout            
    3 192.168.88.101                                          timeout            
    4 192.168.88.101                                          timeout            
    5 192.168.88.101                                          timeout            
    6 192.168.88.101                                          timeout            
    7 192.168.88.101                                          timeout            
    8 192.168.88.101                                          timeout            
    9 192.168.88.101                                          timeout            
   10 192.168.88.101                                          timeout            
   11 192.168.88.101                                          timeout            
    sent=12 received=0 packet-loss=100% 
```
No. Make sense, because wlan0 interface is not in registration table of AP.

Fourdee, 2016-04-16:
@k-plan 
Thanks. And just to confirm, when you reboot with static IP, it works fine?

k-plan, 2016-04-16:
@Fourdee 

```
root@DietPi:~# reboot

root@DietPi:~# ip a
. . .
5: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 74:da:38:73:2c:aa brd ff:ff:ff:ff:ff:ff
    inet 192.168.88.101/24 brd 192.168.88.255 scope global wlan0
       valid_lft forever preferred_lft forever
    inet6 fe80::76da:38ff:fe73:2caa/64 scope link
       valid_lft forever preferred_lft forever

root@DietPi:~# ping 192.168.88.1
PING 192.168.88.1 (192.168.88.1) 56(84) bytes of data.
64 bytes from 192.168.88.1: icmp_seq=1 ttl=64 time=22.7 ms
64 bytes from 192.168.88.1: icmp_seq=2 ttl=64 time=14.7 ms
64 bytes from 192.168.88.1: icmp_seq=3 ttl=64 time=14.2 ms
64 bytes from 192.168.88.1: icmp_seq=4 ttl=64 time=14.3 ms
64 bytes from 192.168.88.1: icmp_seq=5 ttl=64 time=12.9 ms
^C
--- 192.168.88.1 ping statistics ---
5 packets transmitted, 5 received, 0% packet loss, time 4009ms
rtt min/avg/max/mdev = 12.928/15.789/22.758/3.538 ms
```

Fourdee, 2016-04-16:
@k-plan 
Thanks. Think i've found the cause. Only 1 gateway entry is needed, 2 (unless in AP mode) prevents wlan0 up.

Can you:
- edit ```/etc/network/interfaces``` and comment out ```#gateway 192.168.0.1``` under ```wlan0```
- ```dietpi-config``` set dhcp, apply
- ```dietpi-config``` set static, apply

k-plan, 2016-04-16:
@Fourdee 

```
root@DietPi:~# cat /etc/network/interfaces
#/etc/network/interfaces
#Please use DietPi-Config to modify network settings.

# Local
auto lo
iface lo inet loopback

# Ethernet
auto eth1
iface eth1 inet dhcp
address 192.168.0.100
netmask 255.255.255.0
gateway 192.168.0.1
#dns-nameservers 8.8.8.8 8.8.4.4

# Wifi
auto wlan0
iface wlan0 inet static
address 192.168.88.101
netmask 255.255.255.0
## gateway 192.168.88.1
wireless-essid T2-Net
wireless-key test-12345
wireless-mode Managed
wireless-power off
wpa-ssid T2-Net
wpa-psk test-12345
dns-nameservers 8.8.8.8 8.8.4.4
```
after:

```
root@DietPi:~# cat /etc/network/interfaces
#/etc/network/interfaces
#Please use DietPi-Config to modify network settings.

# Local
auto lo
iface lo inet loopback

# Ethernet
auto eth1
iface eth1 inet dhcp
address 192.168.0.100
netmask 255.255.255.0
gateway 192.168.0.1
#dns-nameservers 8.8.8.8 8.8.4.4

# Wifi
auto wlan0
iface wlan0 inet static
address 192.168.88.101
netmask 255.255.255.0
## gateway gateway
wireless-essid T2-Net
wireless-key test-12345
wireless-mode Managed
wireless-power off
wpa-ssid T2-Net
wpa-psk test-12345
dns-nameservers 8.8.8.8 8.8.4.4
```

k-plan, 2016-04-16:
@Fourdee 

but what will happened, if I plugged eth1 out or disable it?

```
root@DietPi:~# route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         172.16.10.3     0.0.0.0         UG    0      0        0 eth1
172.16.10.0     0.0.0.0         255.255.255.0   U     0      0        0 eth1
192.168.88.0    0.0.0.0         255.255.255.0   U     0      0        0 wlan0
```

Fourdee, 2016-04-16:
> but what will happened, if I plugged eth1 out or disable it?

I assume both eth0 and wlan0 when static will share a single ```gateway=``` entry. I'll need to test it.

k-plan, 2016-04-16:
Will work in your network:

eth1: 192.168.0.100/24
wlan0: 192.168.0.200/24
GW: 192.168.0.1
DNS: 192.168.0.1, 8.8.8.8
DHCP-Server: 192.168.0.1

(Wifi) Router: 192.168.0.1/24

---
If you have a network with a Router and for example with a guest WLAN and  segregate DHCP / IP address block, you will need two gateways.

k-plan, 2016-04-16:
@Fourdee 

But for now v.116 it is okay! 👍 

leave it as it is and do your release. Let's do the rest later.

k-plan, 2016-04-16:
Note (for me or whomever):

Useful Things / Idea:

1. WiFi HotSpot
- DNS-Server selection (like new menu in interfaces) in dietpi-config
- possibility to change DHCP area and Gateway address in dietpi-config
- possibility to change 802.11g to 802.11n via diet-config, if hardware support it.

2. TOR HotSpot
- DNS-Server selection (like new menu in interfaces) in dietpi-config
- possibility to change DHCP area and Gateway address in dietpi-config
- possibility to change 802.11g to 802.11n via diet-config, if hardware support it.

k-plan, 2016-04-16:
@Fourdee 
> Edit: Yep, it does. Disabled eth0, wlan0 still using the static gateway.

Hmm ... please can you show `ip a` and `route -n` ?

> So i'll need to change the DietPi code to allow for a single shared gateway= entry when both, or 1 adapter is static.

Hmm .. if ready, please give to me. I like to test it. Thanks.

k-plan, 2016-04-17:
@Fourdee 

> Think i've found the cause. Only 1 gateway entry is needed, 2 is whats preventing wlan0 up.

hmm ... please have a look here: [Debian Network setup](https://www.debian.org/doc/manuals/debian-reference/ch05.en.html)
	especially:
https://www.debian.org/doc/manuals/debian-reference/ch05.en.html#_the_ifmetric_package

`root@oDroid-C1-Plus-Test:~# apt-get install ifmetric`
 then set an option line with "`metric 0`" just below the "`iface eth1 inet static`" line
and an option line with "`metric 1`" just below the "`iface wlan0 inet static`" line
 in "`/etc/network/interfaces`".

Controlling network interface only works via `ifdown eth1`  or `ifup wlan0` .
( ` service networking restart` fail, `dietpi-config` partially ruin the new config)

Now you can run tests (e.g. ping, trace, wget, digg, nslookup ...), switching network interfaces on and off.

You will run into problems, if you have set `dns-nameservers` in both `iface` to the same ip addresses.
(e.g.` ifup wlan0` fail , look into ` cat /etc/resolv.conf`)

I have run some tests with static ip addresses on eth1 and wlan0 then they are in the same broadcast domain (e.g. 192.168.0.0/24). I think, it's your preferred scenario, which you be able to test in your home network.

Fourdee, 2016-04-17:
@k-plan 
Did some testing today and 1 gateway entry is not a fix. Its just a workaround to get a wifi static adapter to come up when eth is up. Each device needs it own entry when static, else, the device that doesn't have it set in ```/etc/network/interfaces``` will not be able to route traffic.

I'll look into your metric suggestion and see what comes up :+1:  For now, I'll mark this as a known issue in v116 and leave this ticket open for v117.

k-plan, 2016-04-17:
@Fourdee 
>  its full of baby stuff

no thanks. 😄 

> why do you want it?

Do you want no geek stuff? :flushed:
PayPal feel too non-personal for me. But anyway, forget it. For now. :wink:

Fourdee, 2016-04-17:
@k-plan 
I main use amazon to buy nappies, wet wipes and all that general stuff lol. RS and Hardkernel for everything "gadget".

Fourdee, 2016-04-24:
@k-plan 
The ifmetric addition you mentioned seemed to have resolved the issue :+1: :
- eth0 static or dhcp
- wlan0 static (Ip addresses showing correctly)

So it looks like this resolves the issue of incorrect IP/routing when using both adapaters and wlan0 as static.

As far as I can tell, problem solved. Can you try ifmetric on your system, just to confirm it fixes the issue? All I did was add ```metric 0``` under eth, ```metric 1`` under wlan, in ```/etc/network/interfaces``` as per your suggestion.

k-plan, 2016-04-24:
@Fourdee 

only tested on Odroid C1+.

1. connect to the same network segment ( layer-2 broadcast domain) e.g. 192.168.0.0/24

- eth0 static or dhcp

- wlan0 static or dhcp

2. connect to the differnt network segment ( layer-2 broadcast domain) e.g. 192.168.0.0/24 and 10.0.0.0/24

- eth0 static or dhcp

- wlan0 static or dhcp

Not tested with WiFi HotSpot and Tor WiFi Hotspot.
Not tested on other devices, like C2, XU4 , RPis, VM, OrangePi, PINE 64

Sometimes you will run into problems, if `dietpi-config` have set (active) dns-nameservers in both iface, especially to the same ip addresses. (e.g.ifup wlan0 fail , look into cat /etc/resolv.conf)
Don´t know, why  `dietpi-config` did this sometimes.

Controlling network interface only works via ifdown eth1 or ifup wlan0 .
( service networking restart fail, `dietpi-config` partially ruin the new config)

Is it in the testing branch for now? 
Which testing scenario should we focus fist?

Fourdee, 2016-04-25:
@k-plan 
Thanks.

> Is it in the testing branch for now? 

Not currently, i'll add it in today so we can test.

> and OT dietpi-config ruin my manual ipv6 setting in /etc/network/interfaces if I have added metric to each interface

What manual setting did you use for ipv6?

[30 more replies]

closed 2016-05-01 by Fourdee