MichaIng/DietPi #274: Add documentation for using testing branch

WolfganP, 2016-04-12:
I saw several updates that I wanted to beta test before release, and couldn't find any comprehensive documentation for using it; just a couple of snippets here (https://github.com/Fourdee/DietPi/issues/214#issuecomment-208025074) and there (https://github.com/Fourdee/DietPi/issues/214#issuecomment-208366667); and few references on the fuzon forums.

Will it be worthy to add a document at root level (ie TESTING-BRANCH.txt to complement README and CHANGELOG) to use as a central go to doc for developers to help testing upcoming features?
Thx.

Fourdee, 2016-04-12:
@WolfganP 
Great suggestion :+1: I will create a new file as per your recommendation: TESTING-BRANCH.txt, with instructions to use the dev testing branch.

Just need to make users aware that this is not a "beta" branch, its active devolpment and potentially unstable at times.

Fourdee, 2016-04-12:
@WolfganP 
.md for pretty text formatting:
https://github.com/Fourdee/DietPi/blob/testing/TESTING-BRANCH.md

Will mark this as completed, great suggestion :+1: . I just need to remember to link the file next time :)

closed 2016-04-12 by Fourdee

WolfganP, 2016-04-12:
Excellent @Fourdee . I think is also worthy to add the note at https://github.com/Fourdee/DietPi/issues/214#issuecomment-208366667

> When you copy the scripts from dev branch, make sure you do NOT copy /boot/dietpi/.version. Otherwise, your system will not be updated correctly by the patch system.