MichaIng/DietPi #276: Test NETDATA install on devices

Fourdee, 2016-04-14:


Fourdee, 2016-04-14:
#### Odroid's (C2) = Failed
```
E: Package 'nodejs' has no installation candidate
```

Fourdee, 2016-04-14:
Ok, as we already have a NODEJS installation option (linux software), i've flagged ```NETDATA``` to set ```NODEJS=1```: https://github.com/Fourdee/DietPi/commit/b1b8581a87b9f6c1d86f58c8023e12e9ecd24e82

Now it appears NodeJS install/repo is no longer working:
```
W: Failed to fetch https://deb.nodesource.com/node_5.x/dists/jessie/InRelease  Unable to find expected entry 'main/binary-arm64/Packages' in Release file (Wrong sources.list entry or malformed file)

E: Some index files failed to download. They have been ignored, or old ones used instead.
```

```
deb https://deb.nodesource.com/node_5.x jessie main
```

I'll look into this further.

Fourdee, 2016-04-14:
Tried the following:
```
curl -sL https://deb.nodesource.com/setup_5.x | sudo -E bash -
sudo apt-get install -y nodejs
```

Appears there is no arm 64bit package yet for C2.

Fourdee, 2016-04-14:
I'll compile for arm64bit and host .deb on dietpi.com

Fourdee, 2016-04-14:
Compiled NodeJS for ARM64 (C2) and updated NodeJS installation code: https://github.com/Fourdee/DietPi/commit/67291022cfc383ba6909ad9cdb8dadcbdd8e9376

Fourdee, 2016-04-14:
Ok, we need to also compile netdata for ARM64 (C2):
```
root@DietPi:~# ls -lha /usr/sbin/netdata
-rwxr-xr-x 1 root root 312K Apr 12 04:19 /usr/sbin/netdata
root@DietPi:~# /usr/sbin/netdata
-bash: /usr/sbin/netdata: No such file or directory
```

@rhkean If you dont have a C2, or cross compiler, let me know and email me the steps to compile this.

Fourdee, 2016-04-14:
@rhkean 
I tested netdata on the RPi 2 with the nodeJS code changes, running fine. Looks awesome :+1: 
[image]

Fourdee, 2016-04-14:
@rhkean 
I've added the online documentation: http://fuzon.co.uk/phpbb/viewtopic.php?f=8&t=5&start=60#p1611

ps. My son is at his nans for the day, so trying to get as much done as I can :+1:

rhkean, 2016-04-14:
you're a BEAST!!!!

I forgot that you added nodejs to the other software menu.... DOH!

so, to recap:
- if NETDATA == 1, then NODEJS=1 (instead of 'AGI nodejs' in the netdata install block)
- need ARM64 build for the C2
- while you were at it, you fixed the NODEJS on ARM64 issue that we didn't know was an issue
- AND... you did the documentation

all before I got out of my morning meeting with my boss....
yup... you're a coding BEAST!!!!  :laughing:

rhkean, 2016-04-14:
I don't have a cross compiler setup, but I'm about to compile the RPi 1 version....
the steps will be the same.... I'll document them and post them here...

Fourdee, 2016-04-14:
@rhkean 
Hehe,

> if NETDATA == 1, then NODEJS=1 (instead of 'AGI nodejs' in the netdata install block)

Yep.

> need ARM64 build for the C2

I wasnt sure if you had a ARM64 system available, so i made a start based on: https://github.com/firehol/netdata/wiki/Installation#install-netdata. All we really need is the ARM64 binaries.

> while you were at it, you fixed the NODEJS on ARM64 issue that we didn't know was an issue

Yep

> AND... you did the documentation

Aye, been a busy beaver. Been storing it all up last few weeks lol.

> I don't have a cross compiler setup, but I'm about to compile the RPi 1 version

Excellent :+1:

Fourdee, 2016-04-14:
@rhkean 
Ok ARM64 is done (C2). Tested it and running fine: https://github.com/Fourdee/DietPi/commit/3816fc3507301e76a5a01acecae8613b2d2949dd

I noticed this on the Odroid C2 compile:
```
I see you have kernel memory de-duper (called Kernel Same-page Merging,
or KSM) available, but it is not currently enabled.

To enable it run:

echo 1 >/sys/kernel/mm/ksm/run
echo 1000 >/sys/kernel/mm/ksm/sleep_millisecs

If you enable it, you will save 20-60% of netdata memory.
```

Was this the same for RPi? We could add this to the existing netdata service ```/etc/systemd/system/netdata.service``` ?

rhkean, 2016-04-14:
start:  DietPi + Vim + BuildEssentials + Git + NodeJS

```
apt-get install zlib1g-dev gcc make git autoconf autogen automake pkg-config
git clone https://github.com/firehol/netdata.git --depth=1
cd netdata
export LC_ALL=C
export CFLAGS="${CFLAGS--O3}"
export ZLIB_CFLAGS=" "
export ZLIB_LIBS="-lz"

./autogen.sh
./configure --prefix="/usr" --sysconfdir="/etc" --localstatedir="/var" --with-zlib --with-math --with-user=netdata CFLAGS="${CFLAGS}"

make
DESTDIR=$(mkdir temp_install && readlink -e temp_install) make install
install -D system/netdata.service temp_install/etc/systemd/system/netdata.service
tar -czvf netdata_1.0.0_rpi1.tgz -C temp_install .
rm -rf temp_install
```
(I'll email you the RPi 1 tarball shortly and add the HW_MODEL support)

rhkean, 2016-04-14:
> Ok ARM64 is done (C2). Tested it and running fine: 3816fc3

I got the same message, but I haven't looked into it

k-plan, 2016-04-14:
Tested on Odroid C1+ ( armv7l):

install via` dietpi-software` : works! 👍 

Nice job rhkean and Fourdee . 😄 

Do a quick look.Only thing I notice , is the flashing update button in the top right corner. But no big deal.

[image]

Have to do a deeper look on netdata. Its a cool monitoring tool.

rhkean, 2016-04-14:
@k-plan 
BAH!!!!!
that's annoying!  I hadn't noticed that before.  It's apparently checking github for commits. 
So, there was no flashing right after a build.  My Odroid XU4 is flashing, but my Rpi 1 (binary built today) isn't flashing....

Let me see if I can disable that "feature"....  I'm not going to rebuild that tarball everytime there's a new commit to github.... :smiling_imp:

k-plan, 2016-04-14:
@rhkean 

> Let me see if I can disable that "feature".... I'm not going to rebuild that tarball everytime ...

👍  - best choice. 

But anyway, it will not subtract your effort. 
Really quick and smooth netdata implementation. Congratulation.

Fourdee, 2016-04-15:
@rhkean @k-plan 
Yep I agree, the update flashing is ruining this great application. Its unnecessary, especially considering it just checks for commits, and not stable release cycles.

I'd be more inclined to make a request directly to NetData, to remove the flash animation by default. Or at the very least have an option to disable this GUI blunder.

Fourdee, 2016-04-15:
If there is no way to disable the update animation button, as a last resort ,we could always fork NetData and do our own release cycles for our users.
Lets see NetData's response first.

Fourdee, 2016-04-15:
@rhkean 

#### So items left on this ticket:
- I need the armv6 compiled .tgz, so I can upload to dietpi.com and update link. If you can send this over Rob, i'd be grateful.
- Try and prevent the UI update animation flashing. Pending NetData's response: https://github.com/firehol/netdata/issues/233
- Finally, once all code is completed, test install on RPi 1 and C2 just to make sure its functional.

rhkean, 2016-04-15:
OMG... did I forget to email you the tarball.

Fourdee, 2016-04-15:
@rhkean 
I cant find it in my emails, any chance you could send it over again please?

Fourdee, 2016-04-15:
@rhkean 
Bizzare, these lines of code:
```
			for x in /var/cache/netdata /usr/share/netdata/web /etc/netdata /var/log/netdata; do
				chown -R netdata.netdata $x
				chmod 0755 $x
			done
```

is causing:
```
root@DietPi:~# dietpi-software
/DietPi/dietpi/dietpi-software: line 4330: syntax error near unexpected token `fi'
/DietPi/dietpi/dietpi-software: line 4330: `            fi'
```

I looked through the commits, cant see anything that would cause this to happen. Strange, I'll check again.

Fourdee, 2016-04-15:
Tested and confirmed:
```
echo 0 > /usr/share/netdata/web/version.txt
```

Prevents update flashing.

rhkean, 2016-04-15:
we can just change that file in all of the tarballs
or do the echo during install and keep the tarballs as-is

preference?

Fourdee, 2016-04-15:
@rhkean 
I've added this to the Apply_Config function in DietPi-Software: https://github.com/Fourdee/DietPi/commit/b611159df77fcea990d590b91f40308a620978bb

I've been updating this post with status: https://github.com/Fourdee/DietPi/issues/276#issuecomment-210417905

Apologies if I'am going a bit "full on" with this. I just want to get NetData finished today, so I can focus on the v116 re-images Saturday (son is away for the day) https://github.com/Fourdee/DietPi/issues/278 . Should also be able to get v116 released Saturday also.

rhkean, 2016-04-15:
no apologies necessary.
I'm learning in the process... I caught the need to edit the patch_file, but missed the process_tool change requirements... :smile_cat:

Fourdee, 2016-04-15:
Tested RPi 3, working fine. @rhkean , any chance you can test the RPi 1 install?

rhkean, 2016-04-15:
yes, if I can get away from the office I'll test it at lunch.  Otherwise it'll be tonight, after work.
I've been wanting to test with my son's A+ anyhow... :smile:

Fourdee, 2016-04-15:
@rhkean 
Excellent, I don't own a A+, got a Zero though. You will have to let us know what you think of it :)

rhkean, 2016-04-15:
:laughing: , I think the Zero has more RAM.

rhkean, 2016-04-15:
A+ passed

BTW... we need to add the ip address to the banner.
the single usb port had the wifi in it....  I had to look up the dhcp address on the router so that I could access it remotely.. 😄

rhkean, 2016-04-16:
B+ passed

rhkean, 2016-04-16:
Rpi 2 B+: passed

Fourdee, 2016-04-16:
@rhkean 
Excellent, thanks for the tests :+1: 

> BTW... we need to add the ip address to the banner.
the single usb port had the wifi in it.... I had to look up the dhcp address on the router so that I could access it remotely.. 

Is this in DietPi-Software, or something else?

rhkean, 2016-04-16:
the banner on the console at boot (before login) that says the version and default login information.

so, if you don't have a keyboard (but do have a monitor) you know what ip address to ssh to after the initial setup is complete.

rhkean, 2016-04-16:
I think it comes from /Dietpi/dietpi/dietpi-banner
but, I haven't looked at the code yet

Fourdee, 2016-04-16:
@rhkean 
Makes sense, good idea :+1: . ```/Dietpi/dietpi/dietpi-banner``` is correct. 
We can use the current code in ```dietpi-software``` for the same in ```/Dietpi/dietpi/dietpi-banner```

```
WHIP_BACKTITLE="DietPi-Software | IP address: $(sed -n 4p /DietPi/dietpi/.network)"
```

So, add to ```/Dietpi/dietpi/dietpi-banner```
```
echo -e "IP address: $(sed -n 4p /DietPi/dietpi/.network)"
#using existing colors as needed
```

You up for this one? Or want me to pop it in?

rhkean, 2016-04-16:
done (create #284 ) and submitted the pull request

Fourdee, 2016-04-16:
Thanks Rob. I'll mark this ticket as completed.

closed 2016-04-16 by Fourdee

WolfganP, 2016-04-19:
I just updated to v116, thx a lot for the enhancements guys. I noticed the discussions around the (almost constant) update notification and the mod to hide it, but I think one of the side effects is that it disallows manual updating (as netdata version is reflected as unknown)

[image]

Fourdee, 2016-04-19:
@WolfganP 
Yep. The downside of disabling the flashing update button (using the workaround https://github.com/firehol/netdata/issues/233#issuecomment-210468649) is the version check will not work: https://github.com/firehol/netdata/issues/233#issuecomment-210468649.

ktsaou mentions a possible option in the future that will simply disable the automatic update check. Until this is available, we will continue to use the workaround.

Its worth noting that the netdata update button only checks the version and lists changes, it does not have a facility to actually update the installation.

rhkean, 2016-04-19:
@WolfganP ,

As @Fourdee mentioned, it does not actually facilitate updating the code.  Also note, we're providing pre-compiled binaries for netdata.

We'll try to keep up with netdata's major releases, and recompile when those milestones occur.

WolfganP, 2016-04-19:
Yes, I read those issues # as well. Good to hear you're both on top of it. Thanks!