MichaIng/DietPi #277: DietPi-Software | Disable Screen blanking during installation.

Fourdee, 2016-04-14:
As per @rhkean suggestion:

Not 100% sure if user input is the only thing that prevents screenblanking (power saving). But there is no harm in adding it to our code.

Fourdee, 2016-04-14:
Tested on RPi. 
```
setterm -blank 0 -powersave off &> /dev/null
```

Added to DietPi-Software to both functions: 
- ```FirstRun_DietPi_Update```
- ```Run_Installations```
Completed.

Fourdee, 2016-04-14:
Retested, failed due to command being sent to ```/dev/null``` 

Resolved with stderr redirect:
```
setterm -blank 0 -powersave off 2> /dev/null
```

closed 2016-04-14 by Fourdee

hadifarnoud, 2016-10-02:
would be great to have this option in config as well.

Fourdee, 2016-10-02:
@hadifarnoud 

> would be great to have this option in config as well.

In ```dietpi-config``` and ```dietpi.txt```  to enable/disable screen blanking?