MichaIng/DietPi #279: Waiting for DietPi-RAMDISK to finish copying DietPi to RAM... Foreover (?)

studioai, 2016-04-14:
Hi, I'm new to diet pi.

I Installed diet pi on my Odroid-XU4 and logged in for first time.

and...

Waiting for DietPi-RAMDISK to finish copying DietPi to RAM...

for over 1 hour....

When does it ends?

Is it normal status?

Fourdee, 2016-04-14:
@studioai 

Thanks for the report.
This is not normal. You should not see the Ramdisk waiting messages at all. This could occur on a corrupted flash device, or unstable power supply.

Can you provide the following information please:
- SDcards used (make/model), or EMMC
- Power supply used
- Have you made any changes to ```dietpi.txt``` or ```boot.ini``` file before booting (if unsure, its no :) )
- Have you tried writing the DietPi image a second time to see if the issue reoccurs?

Thanks

studioai, 2016-04-14:
Thanks for response.

SDCard: TEAMGROUP 32GB (UHS1)
Power supply: Official Odroid power supply
changes to dietpi.txt or boot.ini: No
writing the DietPi image a second time: I'll try right now

studioai, 2016-04-14:
writing the DietPi image a second time

solved the Issue ! Thanks!

closed 2016-04-14 by studioai

Fourdee, 2016-04-14:
@studioai 
No worries, thanks for letting us know it works. If issues reoccur, I would highly suggest trying a known branded SD card (Sandisk/Samsung etc) to improve stability.