MichaIng/DietPi #281: DietPiCam | Fails to start

Fourdee, 2016-04-14:


Fourdee, 2016-04-14:
> note to self:
Test with RPi 1

Fourdee, 2016-04-15:
> kernowyon:
The removal and re-install has worked for me, although I note the other reporter tried before but had no joy.
77fbd864-482e-474d-bb8c-a472e721165a-0

Fourdee, 2016-04-23:
Unable to replicate the reported issue on RPi 1/3. I'am just wondering if this is a simple case of GPU memory being changed after installation:
```
root@BlueDietPi:~# raspimjpeg
start monitoring for pid: 1372
{2016/04/10 22:50:16} start monitoring for pid: 1372
RaspiMJPEG Version 5.2.12
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
Error: Could not create camera
Executing macro /var/www/dietpicam/macros/error_hard.sh "Could not create camera" 
sh: 1: /var/www/dietpicam/macros/error_hard.sh: Permission denied
```

t41photo, 2016-07-04:
Getting the same message as described above - the script "/var/www/dietpicam/macros/error_hard.sh: Permission denied"  that has no x is a reboot now??  

would love to sort it but would like to hear if you have a solution. Thanks - will followup, Wayne

[image]

Fourdee, 2016-07-04:
@t41photo 
Thanks for the report.

Which RPi model and DietPi image/version (eg: v120 Jessie) are you using?

t41photo, 2016-07-04:
Dan,

PRi model : RPi 2 Model B (armv7l)
Jessie: V122 ( I think ) straight from the root logon message.

Addition info: I am using it without screen, keyboard and mouse - wireless
only. All is good that way.

Great package and lots of effort on your part. Very cool. I did add a user
and installed "vi" ( retired sys admin - getting back into the unix like
world).

root@DietPi:/etc# cat /proc/version
Linux version 4.4.13-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3
(crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #894 SMP Mon Jun 13
13:13:27 BST 2016
root@DietPi:/etc#

root@DietPi:/etc# cat os-release
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
root@DietPi:/etc#

many thanks, Wayne

Fourdee, 2016-07-04:
@t41photo 
Thanks for the info Wayne :+1: 

I was unable to replicate the issue on a clean installation. So I'll need some more info on your system.

Can you please send a bug report, this will upload DietPi settings that I'll need and hopefully resolve this: http://dietpi.com/phpbb/viewtopic.php?f=8&t=499

t41photo, 2016-07-04:
ticket generated:
a3efb311-4481-4b5f-b421-7035b88a1c72-0

thanks again

Fourdee, 2016-07-04:
@t41photo 
System configuration is fine. ```raspimjpeg``` is also actively running on your system.

So lets check those permissions. 
```/var/www``` directory structures and file permissions are not included in the bug report, so, please can you run the following:
```
ls -lha $(find /var/www) > /root/send_to_dietpi.txt
```
Then upload (or paste)```/root/send_to_dietpi.txt``` here.

t41photo, 2016-07-04:
Here's the paste - I did unistall lighttdt in favour of nigx web service

-rw-r--r-- 1 www-data www-data  38K Jul  4 14:30 /var/www/apc.php
-rw-r--r-- 1 www-data www-data  119 Jul  3 17:32
/var/www/dietpicam/cam_get.php
lrwxrwxrwx 1 www-data www-data   22 Jul  3 17:32 /var/www/dietpicam/cam.jpg
-> /run/shm/mjpeg/cam.jpg
-rw-r--r-- 1 www-data www-data 1.2K Jul  3 17:32
/var/www/dietpicam/cam_pic_new.php
-rw-r--r-- 1 www-data www-data  238 Jul  3 17:32
/var/www/dietpicam/cam_pic.php
-rw-r--r-- 1 www-data www-data  446 Jul  3 17:32
/var/www/dietpicam/cmd_func.php
-rw-r--r-- 1 www-data www-data  177 Jul  3 17:32
/var/www/dietpicam/cmd_pipe.php
-rwxr-xr-x 1 www-data www-data 7.6K Jul  3 17:32
/var/www/dietpicam/config.php
-rw-r--r-- 1 www-data www-data   59 Jul  3 17:32
/var/www/dietpicam/convertCmd.txt
-rw-r--r-- 1 www-data www-data 1.3K Jul  3 17:32
/var/www/dietpicam/Copyright.txt
-rw-r--r-- 1 www-data www-data   39 Jul  3 17:32
/var/www/dietpicam/css/es_Default.css
-rw-r--r-- 1 www-data www-data  237 Jul  3 17:32
/var/www/dietpicam/css/es_Night.css
-rw-r--r-- 1 www-data www-data   14 Jul  3 17:32
/var/www/dietpicam/css/extrastyle.txt
-rw-r--r-- 1 www-data www-data  445 Jul  3 17:32
/var/www/dietpicam/css/preview.css
-rw-r--r-- 1 www-data www-data 109K Jul  3 17:32
/var/www/dietpicam/css/style_minified.css
-rw-r--r-- 1 www-data www-data 2.4K Jul  3 17:32
/var/www/dietpicam/delete.png
prw-r--r-- 1 www-data www-data    0 Jul  3 17:32 /var/www/dietpicam/FIFO
prw-r--r-- 1 www-data www-data    0 Jul  3 17:32 /var/www/dietpicam/FIFO1
-rw-r--r-- 1 www-data www-data  20K Jul  3 17:32
/var/www/dietpicam/fonts/glyphicons-halflings-regular.eot
-rw-r--r-- 1 www-data www-data 107K Jul  3 17:32
/var/www/dietpicam/fonts/glyphicons-halflings-regular.svg
-rw-r--r-- 1 www-data www-data  45K Jul  3 17:32
/var/www/dietpicam/fonts/glyphicons-halflings-regular.ttf
-rw-r--r-- 1 www-data www-data  23K Jul  3 17:32
/var/www/dietpicam/fonts/glyphicons-halflings-regular.woff
-rw-r--r-- 1 www-data www-data  18K Jul  3 17:32
/var/www/dietpicam/fonts/glyphicons-halflings-regular.woff2
-rw-r--r-- 1 www-data www-data 1.1K Jul  3 17:32
/var/www/dietpicam/image.png
-rw-r--r-- 1 www-data www-data  29K Jul  3 17:32
/var/www/dietpicam/index.php
-rw-r--r-- 1 www-data www-data 3.2K Jul  3 17:32
/var/www/dietpicam/js/pipan.js
-rw-r--r-- 1 www-data www-data 3.7K Jul  3 17:32
/var/www/dietpicam/js/preview.js
-rw-r--r-- 1 www-data www-data  21K Jul  3 17:32
/var/www/dietpicam/js/script.js
-rw-r--r-- 1 www-data www-data 167K Jul  3 17:32
/var/www/dietpicam/js/style_minified.js
-rw-r--r-- 1 www-data www-data  25K Jul  3 17:32
/var/www/dietpicam/loading.jpg
-rw-r--r-- 1 www-data www-data   33 Jul  3 17:32
/var/www/dietpicam/macros/error_hard.sh
-rw-r--r-- 1 www-data www-data   42 Jul  3 17:32
/var/www/dietpicam/macros/test.sh
lrwxrwxrwx 1 www-data www-data   30 Jul  3 17:32 /var/www/dietpicam/media
-> /mnt/dietpi_userdata/dietpicam
-rw-r--r-- 1 www-data www-data 5.9K Jul  3 17:32
/var/www/dietpicam/motion.php
-rw-r--r-- 1 www-data www-data    1 Jul  3 17:32
/var/www/dietpicam/pilight_off
-rw-r--r-- 1 www-data www-data    1 Jul  3 17:32
[... 113 lines cut ...]
-rw-r--r-- 1 www-data www-data  18K Jul  3 17:32
glyphicons-halflings-regular.woff2

/var/www/dietpicam/js:
total 208K
drwxr-xr-x 2 www-data www-data 4.0K Jul  3 17:32 .
drwxr-xr-x 6 www-data www-data 4.0K Jul  3 17:33 ..
-rw-r--r-- 1 www-data www-data 3.2K Jul  3 17:32 pipan.js
-rw-r--r-- 1 www-data www-data 3.7K Jul  3 17:32 preview.js
-rw-r--r-- 1 www-data www-data  21K Jul  3 17:32 script.js
-rw-r--r-- 1 www-data www-data 167K Jul  3 17:32 style_minified.js

/var/www/dietpicam/macros:
total 16K
drwxr-xr-x 2 www-data www-data 4.0K Jul  3 17:32 .
drwxr-xr-x 6 www-data www-data 4.0K Jul  3 17:33 ..
-rw-r--r-- 1 www-data www-data   33 Jul  3 17:32 error_hard.sh
-rw-r--r-- 1 www-data www-data   42 Jul  3 17:32 test.sh

/var/www/html:
total 12K
drwxr-xr-x 2 www-data www-data 4.0K Jul  4 14:25 .
drwxr-xr-x 4 www-data www-data 4.0K Jul  4 14:30 ..
-rw-r--r-- 1 www-data www-data  867 Jul  4 14:25 index.nginx-debian.html
root@DietPi:~#

t41photo, 2016-07-05:
Dan,

error message in log file agrees with permissions on Macros

Seems that there should be more macros in /var/www/dietpicam/macros  as
described in /etc/raspimjpeg (the config file) and the ones that are there
don't have +x -not able to run.

-rwxrwxrwx 1 www-data www-data 3836662 Jul  4 19:48
/var/www/dietpicam/scheduleLog.txt
wayne@DietPi:~ $ tail -n20 /var/www/dietpicam/scheduleLog.txt
{2016/07/04 19:48:33} Error: Could not create camera
{2016/07/04 19:48:33} Executing macro
/var/www/dietpicam/macros/error_hard.sh "Could not create camera"
{2016/07/04 19:48:39} RaspiMJPEG Version 5.5.2
{2016/07/04 19:48:39} Error: Could not create camera
{2016/07/04 19:48:39} Executing macro
/var/www/dietpicam/macros/error_hard.sh "Could not create camera"
{2016/07/04 19:48:44} RaspiMJPEG Version 5.5.2
{2016/07/04 19:48:44} Error: Could not create camera
{2016/07/04 19:48:44} Executing macro
/var/www/dietpicam/macros/error_hard.sh "Could not create camera"
{2016/07/04 19:48:49} RaspiMJPEG Version 5.5.2
{2016/07/04 19:48:49} Error: Could not create camera
{2016/07/04 19:48:49} Executing macro
/var/www/dietpicam/macros/error_hard.sh "Could not create camera"
{2016/07/04 19:48:54} RaspiMJPEG Version 5.5.2

uninstalled dietpicam - reboot and reinstalled package - no change

Wayne Smith

Fourdee, 2016-07-05:
@t41photo 
> Seems that there should be more macros in /var/www/dietpicam/macros  as
described in /etc/raspimjpeg (the config file) and the ones that are there
don't have +x -not able to run.

Yep spot on.

I went over the installation code, we actually apply +x to dietpicam folder recursive.

Running the below should resolve the issue.
```
chown -R www-data:www-data /var/www/dietpicam
chmod -R 777 /var/www/dietpicam
dietpi-services restart
```

> I did unistall lighttdt in favour of nigx web service

Ok, i'll look into this and see if we can find out what modified file permissions in /var/www.

Fourdee, 2016-07-05:
@t41photo 
> Seems that there should be more macros in /var/www/dietpicam/macros  as
described in /etc/raspimjpeg (the config file) and the ones that are there
don't have +x -not able to run.

EDIT:
I went over the installation code, we apply ```chmod -R 777``` to the data directory only. ```chown -R www-data:www-data``` to ```/var/www/dietpicam```

-x for me on ```/var/www/dietpicam/macros/error_hard.sh``` and still functions. Strange.
```
root@DietPi:~# ls -lha /var/www/dietpicam/macros/error_hard.sh
-rw-r--r-- 1 www-data www-data 33 Jul  4 14:07 /var/www/dietpicam/macros/error_hard.sh
```

Are you using a NTFS USB drive? and, can you paste results of following please:
```
ls -lha /mnt/dietpi_userdata
```

> I did unistall lighttdt in favour of nigx web service

Ok, i'll look into this and see if we can find out what modified file permissions in /var/www.

Fourdee, 2016-07-05:
@t41photo 

found this in your lighttpd logfile. Looks like raspimjpeg might not be starting correctly.
```
2016-07-04 09:33:40: (mod_fastcgi.c.2695) FastCGI-stderr: PHP message: PHP Warning:  readfile(/dev/shm/mjpeg/cam.jpg): failed to open stream: No such file or directory in /var/www/dietpicam/cam_pic.php on line 11
```

```/dev/shm/mjpeg/*``` is created by ```raspimjpeg``` when it runs:
```
root@DietPi:~# rm /dev/shm/mjpeg/cam*
root@DietPi:~# ls -lha /dev/shm/mjpeg/
total 0
drwxrwxrwx 2 www-data www-data 40 Jul  5 12:30 .
drwxrwxrwt 3 root     root     60 Jul  5 12:18 ..
root@DietPi:~# raspimjpeg
start monitoring for pid: 2115
{2016/07/05 12:30:47} start monitoring for pid: 2115
RaspiMJPEG Version 5.5.2
recommended video buffers 1
h264 buffers set to recommended 1
Set up internal detect width=121 height=68
MJPEG streaming, ready to receive commands
send smd 9
^CStarting command loop

root@DietPi:~# motion: no process found
SIGINT/SIGTERM received, stopping
ls -lha /dev/shraspimjpeg^C
root@DietPi:~# ls -lha /dev/shm/mjpeg/
total 40K
drwxrwxrwx 2 www-data www-data  80 Jul  5 12:30 .
drwxrwxrwt 3 root     root      60 Jul  5 12:18 ..
-rw-r--r-- 1 root     root     37K Jul  5 12:30 cam.jpg
-rw-r--r-- 1 root     root       0 Jul  5 12:30 cam.jpg.part
```

Lets check raspimijeg for outputs. Please can you run the following and paste results of raspi output:
```
dietpi-services stop
raspimjpeg
```

t41photo, 2016-07-05:
Dan,

I'll check those but I did try to stop and restart the dietpi-services as
root and the start got stuck looping and prompting for sudoers password.

I checked the sudoers config and there's a pi but no www-data - haven't
tried with a change yet.

8:01am here breakfast first.

Thanks again. Wayne

Could be different with the Pi Cam as opposed to the USB one

Fourdee, 2016-07-05:
@t41photo 

> 8:01am here breakfast first.

Early bird lol :+1: 

No rush Wayne, as and when you can. 
If your short on time, this is the command I'd like you to run 1st and paste output:
```
dietpi-services stop
raspimjpeg
#ctrl+c to exit
```

t41photo, 2016-07-05:
Dan,

Lunch break,

dietpi-services stop

result:
root@DietPi:~# dietpi-services stop

 DietPi-Services
 Mode: stop
 Please wait...
Ok | cron
Ok | nginx
Ok | php5-fpm
Ok | mysql
Ok | RaspimJpeg

root@DietPi:~# raspimjpeg
start monitoring for pid: 2376
{2016/07/05 12:00:01} start monitoring for pid: 2376
RaspiMJPEG Version 5.5.2
mmal: mmal_vc_component_create: failed to create component
'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
Error: Could not create camera
Executing macro /var/www/dietpicam/macros/error_hard.sh "Could not
create camera"
sh: 1: /var/www/dietpicam/macros/error_hard.sh: Permission denied
^C

I'm getting the Idea that this should have the pi camera attached to the board.

Like I mentioned at first I'm trying get it working with a usb one we have.

The good one is on order now.

=====  I did try another flavour of Jessie with another implementation
of RaspiMJPEG.

same results.

I love you package and want to work though it.

Let me know what next step you want. Thanks  (I'll try not to make
changes without your request)

Wayne

Fourdee, 2016-07-05:
@t41photo 
> I'm getting the Idea that this should have the pi camera attached to the board. Like I mentioned at first I'm trying get it working with a usb one we have.

Lol, yep, i missed that you had a USB camera. I automatically assumed you'd be using the RPi cam.

As far as i know raspimjpeg only supports the official RPi camera. The errors you received below confirms this:
```
root@DietPi:~# raspimjpeg
start monitoring for pid: 2376
{2016/07/05 12:00:01} start monitoring for pid: 2376
RaspiMJPEG Version 5.5.2
mmal: mmal_vc_component_create: failed to create component
'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
Error: Could not create camera
```

> The good one is on order now.

Excellent. Don't make any changes to the configs or file permissions. When you plug the camera in, i'am pretty confident it will work straight away :)

Fourdee, 2016-07-08:
I'll mark this as closed. @t41photo Please reopen if you experience issues with the official RPi camera.

closed 2016-07-08 by Fourdee