MichaIng/DietPi #284: Add IP Address to DietPi Banner to ease remote access

rhkean, 2016-04-16:
Adding for tracking purposes

Fourdee, 2016-04-16:
@rhkean 
Thanks Rob :+1:.

Looking good:
[image]

The only issue is this wont be active until the user updates their version of DietPi, or, I redo the DietPi images again for v116. Apologies for that, I should of mentioned. I had to get the images done today, only chance I'll have for a week or two: https://github.com/Fourdee/DietPi/issues/278

I'll make a note to redo the main images (c2/c1/rpi), but it may be after v116 is released.

rhkean, 2016-04-16:
😺 figures. too funny

should have thought of it before last night.... my bad.   ROFL

Fourdee, 2016-04-16:
lol. 

I could explain the image creation process if you are intrested? It's a little lengthy, but would allow you to update the DietPi images as you need :)

ps. Son is staying over nans tonight, Can finally watch Martinsville SCS tonight, cant wait :)

rhkean, 2016-04-16:
I'm game...  Not tonight though... Heading out to dinner with the Mrs.

k-plan, 2016-04-17:
hmm ... I have "Address" twice

[image]
[image]

Typo?

rhkean, 2016-04-17:
not so much a typo, but an issue with the locale... how the value gets parts and stored in the /DietPi/dietpi/.network file.

just suck it up @k-plan !!!!   😈 

no, seriously, I'll look into it.  good catch

k-plan, 2016-04-17:
https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-config#L1813

... but I don't understand what you are doing here. Sorry.

rhkean, 2016-04-17:
@k-plan can you give me the output of your ifconfig ?
the double address is a parsing error that I can fix, but I need to verify what the non-english ifconfig output looks like.

k-plan, 2016-04-17:
```
root@oDroid-C1-Plus-Test:~# ifconfig
eth1      Link encap:Ethernet  Hardware Adresse 00:1e:06:20:1b:a0
          inet Adresse:172.16.10.54  Bcast:172.16.10.255  Maske:255.255.255.0
          inet6-Adresse: fe80::21e:6ff:fe20:1ba0/64 Gültigkeitsbereich:Verbindung
          inet6-Adresse: 2001:470:1f0b:c1a:21e:6ff:fe20:1ba0/64 Gültigkeitsbereich:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metrik:1
          RX packets:79254 errors:0 dropped:64 overruns:0 frame:0
          TX packets:71248 errors:0 dropped:0 overruns:0 carrier:0
          Kollisionen:0 Sendewarteschlangenlänge:1000
          RX bytes:6067871 (5.7 MiB)  TX bytes:12547760 (11.9 MiB)
          Interrupt:40

lo        Link encap:Lokale Schleife
          inet Adresse:127.0.0.1  Maske:255.0.0.0
          inet6-Adresse: ::1/128 Gültigkeitsbereich:Maschine
          UP LOOPBACK RUNNING  MTU:65536  Metrik:1
          RX packets:6 errors:0 dropped:0 overruns:0 frame:0
          TX packets:6 errors:0 dropped:0 overruns:0 carrier:0
          Kollisionen:0 Sendewarteschlangenlänge:0
          RX bytes:984 (984.0 B)  TX bytes:984 (984.0 B)

wlan0     Link encap:Ethernet  Hardware Adresse 74:da:38:73:2c:aa
          inet Adresse:172.16.10.154  Bcast:172.16.10.255  Maske:255.255.255.0
          inet6-Adresse: fe80::76da:38ff:fe73:2caa/64 Gültigkeitsbereich:Verbindung
          inet6-Adresse: 2001:470:1f0b:c1a:76da:38ff:fe73:2caa/64 Gültigkeitsbereich:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metrik:1
          RX packets:3954 errors:0 dropped:525 overruns:0 frame:0
          TX packets:47 errors:0 dropped:5 overruns:0 carrier:0
          Kollisionen:0 Sendewarteschlangenlänge:1000
          RX bytes:569676 (556.3 KiB)  TX bytes:6169 (6.0 KiB)
```

rhkean, 2016-04-17:
bug tracking:  #287 
patched:  #286

k-plan, 2016-04-17:
@rhkean 

Sorry dump question:
Now we have to wait @Fourdee, to look over and merge your fix in testing branch.
After this, I can update to testing branch and test it. Is it correct?

rhkean, 2016-04-17:
@k-plan correct
and he just did that, so you're good to test

k-plan, 2016-04-17:
```
root@oDroid-C1-Plus-Test:~# sed -n 4p /DietPi/dietpi/.network
172.16.10.54
```

[image] 
[image]

@rhkean 💯  👍 

@Fourdee  👍  - push v.116 out (please quick - 😄 )

----

@k-plan    🇩🇪 
@Fourdee  🇬🇧 
@rhkean  :question:

rhkean, 2016-04-17:
🇺🇸

Fourdee, 2016-04-17:
@rhkean @k-plan 
Looking like v116 release for tomorrow, however, may be Tuesday at the latest. Just need to finish these jobs off:
- DietPi's hosting provider MyVirtualServer.com: https://github.com/Fourdee/DietPi/issues/288
- ```LANG=en_GB.UTF-8``` to all scripts if we can prevent ```locale-gen --purge``` from being used to remove ```en_GB.UTF-8```: https://github.com/Fourdee/DietPi/issues/287

Fourdee, 2016-04-18:
Will mark this as closed. 
I've created a new ticket to resolve non-english parsing issues throughout DietPi scripts: https://github.com/Fourdee/DietPi/issues/291

closed 2016-04-18 by Fourdee

rhkean, 2016-05-06:
Anyone else not seeing the IP Address populated with V117?

It's not getting populated on dietpi-banner call from rc.local and my RPi A+

I'm wondering if we optimized it too much... :laughing:

Fourdee, 2016-05-07:
@rhkean 
Yep, it occurred a few times for me. Seems to be when set to DHCP, and when https://github.com/Fourdee/DietPi/blob/testing/dietpi/func/obtain_network_details#L83 results have no IP entry.

I thought about creating a while loop if the IP address is empty, but this would need to be launched as another thread (process) to prevent delaying the system during boot.

For now, I added ```obtain_network_details``` to the ```login``` script: https://github.com/Fourdee/DietPi/blob/testing/dietpi/login#L128

Fourdee, 2016-05-07:
@rhkean 
Did some debugging, added the following to the ```obtain_network_details``` script:
```
        #debug
        ip r >> /root/ipr
        ifconfig "$ACTIVE_DEVICE" >> /root/ifc
        echo -e "$ACTIVE_DEVICE" >> /root/iactive
```

Results:
```
root@DietPi:~# cat iactive
eth0
root@DietPi:~# cat ipr
root@DietPi:~# cat ifc
eth0      Link encap:Ethernet  HWaddr 00:1e:06:33:15:23
          inet6 addr: fe80::21e:6ff:fe33:1523/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:4 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:600 (600.0 B)
          Interrupt:40
```

So it looks like ```ip r``` and ```ifconfig``` dont have the IP address for ipv4 at the time its called.

xenfomation, 2016-05-07:
How did I miss this?!?!  I can translate as I read/speak Spanish, French, German, and Dutch... and more.   AUGH!  I will look at  https://github.com/Fourdee/DietPi/issues/291 NOW so I can at least get something checked in!!

PR #285 (2016-04-16): Add IP Address to Banner
resolves #284