MichaIng/DietPi #287: obtain_network_details not parsing ip address properly on non-english systems

rhkean, 2016-04-17:
see comments on #284 

fixed with pull request #286

Fourdee, 2016-04-17:
@rhkean 

I recently applied:
```LANG=en_GB.UTF-8``` to our main DietPi scripts. This ensures all parsing is returned in English: https://github.com/Fourdee/DietPi/issues/273#issuecomment-210806262

I'am tempted to add this to all the DietPi scripts , which should prevent non-english parsing errors from occurring completely.

Fourdee, 2016-04-17:
@rhkean 

> we can do that, too... I still recommend the 'cut' instead of the 'sed'
also, the forcing of en_GB will work, right up until someone runs 'locale-gen --purge' with en_GB not checked.

I've merged the pull request.

```locale-gen --purge``` good spot :+1: . We could add a bash alias to ```/etc/bash.bashrc``` to prevent this from occurring.

Fourdee, 2016-04-18:
Ok will mark this as closed, created a new ticket to attempt to force english parsing throughout DietPi scripts: https://github.com/Fourdee/DietPi/issues/291

closed 2016-04-18 by Fourdee