MichaIng/DietPi #288: MyVirtualServer.com - The new web hosting provider of DietPi

Fourdee, 2016-04-17:
http://MyVirtualServer.com is now DietPi's hosting provider. http://dietpi.com has been redirected (pending ahost update), forum move will start in the week.
New IP = 185.101.92.145

#### Add following MyVirtualServer.com coverage/sponsership to DietPi as disucssed:
- :u6307:  http://dietpi.com = Header + description + image at bottom of page
- :u5408: README.md = Header + description (above DietPi team)
- :u5408: Twitter post linking our new hosting provider
- :u5408: Add hosting coverage to all DietPi forum signature accounts (RPi forum/Odroid forum etc)
- :u5408: Add hosting coverage to ```dietpi-banner 1```

k-plan, 2016-04-18:
OT:
@Fourdee 

do you ever slow down and have a break? :flushed:

Puhh, changing additional  the hoster/isp now. Is the new vServer a for-free / sponsoring offer? Or have you to pay for it? A OVZ-1 model? 

Can we  assist  you, or do something else?

Fourdee, 2016-04-18:
@k-plan 
Slowing down today, come down with flu lol.

MyVirtualServer.com are sponsoring DietPi by providing us with a VPS to host dietpi.com, and, the dietpi forums (once i've moved them over).

> Can we assist you, or do something else?

I cant think of anything at the moment, but I will let you know if I do :+1:

Fourdee, 2016-04-18:
@k-plan 
The VPS package is OVZ-3, and its blazingly fast :)

Fourdee, 2016-04-18:
Completed, marked as closed.

closed 2016-04-18 by Fourdee