MichaIng/DietPi #290: Let's Encrypt offline or unreachable

inspector71, 2016-04-18:
```
┌─────────────────────────────┤ Error ├──────────────────────────────┐
│                                                                    │
│ Unable to Install Lets Encrypt (SSL cert)                          │
│                                                                    │
│ https://github.com/letsencrypt/letsencrypt/archive/master.zip is   │
│ offline and/or unreachable.                                        │
│                                                                    │
│                                                                    │
│                                                                    │
│                                                                    │
│                                                                    │
│                                                                    │
│                                                                    │
│                               <Ok>                                 │
│                                                                    │
└────────────────────────────────────────────────────────────────────┘
This seems false since I was able to download letsencrypt-master.zip from 

https://codeload.github.com/letsencrypt/letsencrypt/zip/master 

via the same address, that is:

https://github.com/letsencrypt/letsencrypt/archive/master.zip

Fourdee, 2016-04-19:
@inspector71 

Hi, thanks for the bug report.

Could you please try the following for us and paste results:
```
date
wget https://github.com/Fourdee/DietPi/blob/master/README.md
```

If an error occurs, please run:
```
killall ntpd
ntpd -gq #Wait for this to complete, 10 seconds~
date
wget https://github.com/Fourdee/DietPi/blob/master/README.md
```

> note to self, 2 month window for date and cert failures
```
#Date of test = 19/03/2016
root@DietPi:~# date +%Y%m%d -s "20160218"
20160218
root@DietPi:~# wget https://github.com/Fourdee/DietPi/blob/master/README.md
--2016-02-18 00:00:00--  https://github.com/Fourdee/DietPi/blob/master/README.md
Resolving github.com (github.com)... 192.30.252.129
Connecting to github.com (github.com)|192.30.252.129|:443... connected.
ERROR: The certificate of ‘github.com’ is not trusted.
ERROR: The certificate of ‘github.com’ is not yet activated.
The certificate has not yet been activated
```

Fourdee, 2016-04-24:
Unable to replicate error user reported. Waiting for their response.

inspector71, 2016-04-24:
Hi 
Swear I replied to this. Might have been replying when my laptop died.
However I've upgraded since. How do I see the current version of dietpi from the command line?

Fourdee, 2016-04-25:
@inspector71 
```cat /DietPi/dietpi/.version```

inspector71, 2016-04-25:
> @inspector71 <https://github.com/inspector71>
> cat /DietPi/dietpi/.version
>
So I've upgraded to 116. Is it still relevant to do the tests?

Fourdee, 2016-04-25:
@inspector71 

> So I've upgraded to 116. Is it still relevant to do the tests?

No need. 
The DietPi update downloads from git, so if there was a cert error, if would of occurred then. Try to install letsencrypt again and let us know if the error re-appears, then we can do some tests if needed.

xenfomation, 2016-04-28:
Heads up -- I've tested 4 installs now with SSL ignored, but only after locale, timezone, and network settings are all in check.  Hope that helps?

Fourdee, 2016-05-01:
@xenfomation 
> Hope that helps? 

Yep. I'am unable to replicate this error so I will mark as closed.

@inspector71 
Please reopen if you are still experiencing issues with the letsencrypt installation, so we can assist you.

closed 2016-05-01 by Fourdee

inspector71, 2016-05-02:
@Fourdee

Thanks. My initial attempt to install LE was very exploratory. I won't be
getting back to it at any particular specific time. It's exciting to know
that Dietpi has potentially made it extremely easy to setup though :) May
be worth someone's time to porting the scripts to other Debbie-based
platforms in future.