MichaIng/DietPi #291: Ensure DietPi scripts can parse correctly, regardless of system locale

Fourdee, 2016-04-18:


Fourdee, 2016-04-18:
I have applied the following to all DietPi scripts https://github.com/Fourdee/DietPi/commit/5500ffd83ab2b2072595dd9f36dc97e9f070fc2d:
```
	#Force en_GB Locale for whole script. Prevents incorrect parsing with non-english locales.
	LANG=en_GB.UTF-8
```
This will ensure all parsing results are returned in english, regardless of active system locale.

To prevent users removing ```en_GB.UTF-8```:
I've added a message box to ```dietpi-config```, when user changes locale:
[image]

k-plan, 2016-04-18:
@Fourdee 

hmm ...  you although  keep track of DietPi-Automation and templates? 

```
#Language/Regional settings | Requires AUTO_Install_Index=1
. . .
#   Locale eg: en_GB en_US | Full list: cat /etc/locale.gen
AUTO_Locale=en_GB
```
I like dietpi-automation templates very much and often use it.

If someone set here a other language, what will happen?

Would be nice to have always set "`C`" and "`en_GB.UTF-8`" and 
e.g. `AUTO_Locale=de_DE` should only added the new extra default-language

#174

Fourdee, 2016-04-18:
@k-plan
As far as I'am aware, the current automation system for setting locale is not working correctly.

I'll make a note to test ```en_GB.UTF-8``` is not removed when I get round to fixing it.

rhkean, 2016-04-18:
Away from my computer at the moment, but I know that tool creates a text
file that lists out all the locales to build...  We could add some extra
code to always insert en-GB

Otherwise we'll need to go through everywhere that a bin util output is
parsed and correct it.

Fourdee, 2016-04-18:
@rhkean 
> but I know that tool creates a text
file that lists out all the locales to build...  We could add some extra
code to always insert en-GB

Excellent, sounds perfect :+1:

Fourdee, 2016-04-19:
@rhkean 
I'am aiming to release v116 today, this is the only item left.

Do you think it is possible to have your suggestion implemented within the next 2-4 hours? If not, I will close this ticket and we can create a new one for v117 (eg: Prevent users from deselecting/removing en_GB during locale change).

As we have a msgbox warning when users change their locale, its not critical at the moment as users will be aware. Let me know your thoughts.

rhkean, 2016-04-19:
@Fourdee 

Sorry Dan.  I can't get to this today.  I'm being pulled in about 6 directions at work at the moment.

Fourdee, 2016-04-19:
@rhkean 
No worries and no rush. :+1: 
I was just seeing if we could get this in before v116 release. Its not critical and can wait until your available and have the time.

Fourdee, 2016-04-23:
@rhkean 
I'll mark this as closed. I believe the warning message I've added to ```dietpi-config``` when changing locale should be enough.

If we get users reporting errors due to removal of ```en_GB```, then we can add in your suggested measure to force ```en_GB``` as always installed.

closed 2016-04-23 by Fourdee