MichaIng/DietPi #292: DietPi-Update | Github cert errors.

Fourdee, 2016-04-18:
> https://twitter.com/gnfrazier/status/721705125623021568

Fourdee, 2016-04-18:
Added -k (ignore cert errors) to both update check, and, .zip download.

Fourdee, 2016-04-18:
```
curl -k -L https://raw.githubusercontent.com/Fourdee/DietPi/testing/dietpi/dietpi-update > /DietPi/dietpi/dietpi-update
dietpi-update
```

Fourdee, 2016-04-19:
Confirmed resolved by user:
https://twitter.com/gnfrazier/status/722200634087358469

There may be a underlying issue with GitHub certs based on incorrect system date, as per this thread with a similar issue: https://github.com/Fourdee/DietPi/issues/290. I will investigate this further on that ticket.

As this ticket is resolved, I'll mark this one as closed.

closed 2016-04-19 by Fourdee

k-plan, 2016-04-24:
@Fourdee 

http://fuzon.co.uk/phpbb/viewtopic.php?f=11&p=1658#p1658

hmmm ... but for me:

```
root@RPi-B-Plus:~# date
Sun 24 Apr 17:15:39 CEST 2016

root@RPi-B-Plus:~# curl -k -L https://raw.githubusercontent.com/Fourdee/DietPi/testing/dietpi/dietpi-update > /DietPi/dietpi/dietpi-update
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  9960  100  9960    0     0  57717      0 --:--:-- --:--:-- --:--:-- 58245
```