MichaIng/DietPi #294: Move DietPi forums over to new VPS

Fourdee, 2016-04-19:
fuzon.co.uk > dietpi.com

rhkean, 2016-04-19:
so, are you moving the forums from http://fuzon.co.uk/phpbb to http://dietpi.com/phpbb ?

Fourdee, 2016-04-19:
```dietpi-survey``` ```dietpi-bugreport``` now uploads to dietpi.com

Fourdee, 2016-04-19:
@rhkean 
Yep, to start with will move the whole forums. Need to sort out a redirect from fuzon.co.uk to dietpi.com also, as documentation (links) in DietPi wont be updated until users install v117.

Fourdee, 2016-04-19:
ADM is a little broken lol:
[image]

rhkean, 2016-04-19:
do you still have a host for fuzon.co.uk ?

Fourdee, 2016-04-19:
At the moment yes, its still running on home server/network, at least until the move is finished.

As I will need to redirect fuzon.co.uk to dietpi.com anyway (to ensure existing links throughout the web will still work), dietpi.com will be the new home of both dietpi.com and fuzon.co.uk (forums).

In the future i'll look into clean up the forums and separate the non-dietpi stuff somewhere else. Until then, another host for fuzon.co.uk is not needed.

k-plan, 2016-04-20:
@Fourdee 

BTW: don't see it on your list, but the links on dietpi.com pointed for now as well to "http://fuzon.co.uk/phpbb/viewtopic.php?...."

And have a look for the sponsor image layout, displayed on mobile devices. 

[image]

nice-looking and better fit  if it will be embed like raspberry-pi.jpg or odroid.jpg

No big deal and take one's time.

Fourdee, 2016-04-20:
@k-plan 

> BTW: don't see it on your list, but the links on dietpi.com pointed for now as well to 

Good spot, missed that one :+1:. There's so many things to do lol. Added to list.

> And have a look for the sponsor image layout, displayed on mobile devices.

Yep, looks like we need to reduce the image size and set centre alignment. Thanks for letting me know :+1:

k-plan, 2016-04-20:
> Yep, looks like we need to reduce the image size and set centre alignment.

So far I can see it from here, this will do your nice website template in one's own right, if choose the right "`<span class>`"

But take a look, if  have some peace and quiet. Yes, it's hard for  _workerholics_. :wink:
No worry, it works well on desktops.

Fourdee, 2016-04-22:
@k-plan I couldn't figure out how to align centre (html isn't my strength), but i've resized the image for now, should be better on mobile view :)

xenfomation, 2016-04-23:
I'll gladly take a look at the HTML/CSS mark-up.  The IMG tag may be inheriting another class parameter or it's container may just neet a center/alignment adjustment.

Glad to be back into the world of non-book writing and while github is something I use for my own purposes, yeah -- I'm no pro at it -- but I'm sure we can suss something.  I will poke around to see what I can do with the centre necessity...

k-plan, 2016-04-23:
@Fourdee 
yes, better. But it will be beautiful, to use the auto-scaling potential of your html5 template.

Let @xenfomation have look. Thanks for his offer. 👍

Fourdee, 2016-04-24:
So its just the forums left now. Need to replace all ```fuzon.co.uk``` reference links with ```dietpi.com```. 

If I can find a way to "find and replace" in SQL, I'll do it that way. If not, manually replacing and checking each forum post would be a nightmare. Although as ```fuzon.co.uk``` is also redirected to ```dietpi.com```, theres no harm in leaving those old links as they are.

xenfomation, 2016-04-24:
No problem!  Just tying to sort out my account credentials and will reach out to @Fourdee to make sure he can point me to the right part of the branch.  Ah... this is the life!

k-plan, 2016-04-24:
@Fourdee 

> So its just the forums left now.

No, no ...  😄 

Have a look on: https://github.com/Fourdee/DietPi/blob/master/README.md

Your testing is fine: https://github.com/Fourdee/DietPi/blob/testing/README.md

But really no big deal. 

BTW forum: Could you perhaps change this? (could puzzled some user)
http://fuzon.co.uk/phpbb/styles/dualcore/theme/images/logo.png

But only if you like to have dietpi-logo there. Corporate design colours of you website,  black and purple/violet/pink/magenta as well  be a looker for your forum design.

rhkean, 2016-04-24:
about the only way that I've found to find/replace in SQL would be:

# mysqldump -u root -p[root_password] [database_name] > dumpfilename.sql

then find replace on the sql file

then

# *mysql -u root -p[root_password] [database_name] < dumpfilename.sql*

Fourdee, 2016-04-24:
@rhkean Yep, same method I used. :+1: 

@k-plan 
> Have a look on: https://github.com/Fourdee/DietPi/blob/master/README.md
Your testing is fine: https://github.com/Fourdee/DietPi/blob/testing/README.md
But really no big deal.

Testing > Master will be done when v117 is released :)

> BTW forum: Could you perhaps change this? (may puzzled some user)

Yep. I've been thinking about sticking DietPi logo in there. Will take a look.

k-plan, 2016-04-24:
@Fourdee 

> Testing > Master will be done when v117 is released :)

Yep. 👍   I have notice, big coding is going on in the background,  so ..

> current fuzon.co.uk server power off and take it upstairs, watch movies on it lol :)

... your home cinema have to wait for a spell. :laughing:

xenfomation, 2016-04-28:
Heads up for all -- thanks to @Fourdee I am using Firefox, FireFox Mobile Browser Test, and a Samsung Galazy S4.

- The sponsor image is inheriting responsive/css declarations in the Footer section and it's absolute size definition is all causing the slight RIGHT-JUSTIFIED bleed over

[image]

- There is a wrap around issue for the "Community Section's" button text: "Getting Started online documentation and"...

[image]

- Testing at many variable sizes, I can isolate the button text issue to wrap and I can uncover other layout design values that attempt to ruin the GLORY of the site...

[image]

- I should be able to isolate the inheritance size shortly -- before 07:00AM EST 28-APRIL-2016

Cheers!

xenfomation, 2016-04-28:
Assigned to myself.  As the code will be sent directly to @Fourdee, I will highlight items I solved (great job on this, btw!)

1.  Issue with sponsor image occurs when resolution is <= 360x480; that is resolved
- The img/link tab were nested within two classes.  Image style was not set to 100% max-width, so auto scaling from the responsive code could not do it's job

- Move link and img ref 1 level "further" from the copyright/li group; set style to max-width of 100%; RESOLVED

[image]

- Community Button's text was too long if wrap-text was used, the button doubled in size.  I shortened this text.  WORK AROUND.

[image]

- The extremely random sizes leave the Community Icons skewed -- this shouldn't happen due to some standards with mobile devices

- Emailing Dan and seeking his approval (for now until the next wave of super resolution promotes further updates)

Cheers

xenfomation, 2016-04-28:
I sent two emails, @Fourdee -- the 28-APRIL-2016-04:48AM has the non-QA site as I originally sent the QA version.  Point is, all bases are covered unless we get reports from the end-users about skewing or any oddities none of us have seen (along with their monitor, resolution, etc).  All of this feeds back into the HTML->CSS:RESPONSIVE_PAGE flow.  If further work is desired, you know I'm for it and I will be available again this evening circa 20:00 EST.

Fourdee, 2016-04-28:
@xenfomation 

The issue with alignment in the section "DietPi - features overview" has been there a while. To this day, I still dont know whats causing it lol.

I've uploaded your revised code (dietpi.com-28-APR-2016-FINAL_TEST.zip) to http://dietpi.com. Great work on this, looks much better :+1: :)

Fourdee, 2016-04-29:
Marking this as closed. I will continue to work on the forum to add DietPi logo, in the future.

closed 2016-04-29 by Fourdee

xenfomation, 2016-05-07:
"One is happy to be of service" ;)

k-plan, 2016-05-27:
@Fourdee 

https://github.com/Fourdee

[image]

Fourdee, 2016-05-27:
@k-plan 
Bloody good spot :+1: Thanking you!

Updated :)

Fourdee, 2016-05-27:
@k-plan 
Also, before I forget. I tried the DietPi logo with the current forum style weeks back, colours do not look right. So i had to revert it back to Fuzon logo. Most likely needs a new forum style to work with logo.