MichaIng/DietPi #295: Add Vim-Tiny obtion to additional software

rhkean, 2016-04-19:
http://fuzon.co.uk/phpbb/viewtopic.php?f=9&t=367

k-plan, 2016-04-19:
@rhkean 

> option to additional software

may I attached some more basic package from my personally "I-have-to-every-installation-must-add" tool-list?

like: mc, iperf, tcpdump, mtr-tiny ?

Or better open for every tool a separate issue?

rhkean, 2016-04-19:
fine with me, but let's get @Fourdee to chime in.  He has a system and it's been working well for him, so I don't want to buck it.

Dan, thoughts...  add on to this request or create separate tickets?

Fourdee, 2016-04-19:
@rhkean 
Sounds good to me, by all means add them in at your own leisure. 
The only thing I would suggest is put vim-tiny (and k-plans suggestions if you add them) in the linux software list. Basically anything that is "general and/or basic linux stuff" or shared libraries, should go there.

k-plan, 2016-04-20:
> The only thing I would suggest is put vim-tiny (and k-plans suggestions if you add them) in the linux software list.

:100: :+1: 

You can as well but them together category and/or package:
 
- Editor:
  * vim
  * mc-edit

- File Management:
  * midnight commander

- Network tools:
   * iperf, mtr-tiny, tcpdump

xenfomation, 2016-04-24:
Feature request, but I get it.  I just setup and tore down the latest build Sunday @ 03:00 EST, so if there is no progress on this, etc -- I wouldn't mind looking at this.  Like @k-plan and @Fourdee  -- it's misc Linux utils, so my last question is: should others be considered, such as I had to cover for XenServer recently (lsof, lshw, etc, etc) : https://xenfomation.wordpress.com/2016/04/20/xenserver-command-line-tools/ ?   Thanks and pardon the over excitement to contribute... it's been a year for me to clear my schedule :-)

xenfomation, 2016-04-25:
Just a quick reply to all as the CLI is one of my favorite places to be, I've got a handfull of sections with tools added and I will be testing them from branch Test, V116 (locally).

The topic areas are CLI Editors, File Managers, Web/Email, Network Tools, System Monitoring, Process Monitoring, and Administrative Tools.  It' about 60+ utilities I use all the time, but thanks to Debian -- every one is pre-package & ready to go (unlike other distros).

If you think of an additional tool, let me know!

rhkean, 2016-04-25:
@xenfomation 
sweet!!!!!
I'm with you.  I typically don't even install a GUI on my SBC's...

@Fourdee and @k-plan ,
with that many options, it's probably gonna need another menu
and, we might want to consider including approximate disk usage for each option.
thoughts?

xenfomation, 2016-04-25:
Oh -- absolutely!!!  I agree and didn't even consider that... I'll sharpen the list up this evening and see if I can't stub in a new menu.

xenfomation, 2016-04-25:
But I realized what I may have just done.... "Forgot about the Diet part of DietPi" (@Fourdee)!

xenfomation, 2016-05-01:
Final testing for VIM-TINY before checking in.

xenfomation, 2016-05-09:
Thanks, @Fourdee -- I started from addition 1 and moved forward.  Silly syntax error on my behalf.

As this is a feature request, please do let me know -- aside from 2 file manager utilities I would like to add -- how the additions I've checked into testing fit in...

Cheers!

xenfomation, 2016-05-28:
Quick question -- as this is in v118, do I need to close #295 from my fork (it points to xenfomation/DietPi#1)?

Fourdee, 2016-06-15:
As vim-tiny appears to be in dietpi-software at this time, i'll mark this as completed.

closed 2016-06-15 by Fourdee