MichaIng/DietPi #297: Nginx conf for pydio

studioai, 2016-04-24:
`    
        rewrite ^/pydio/dashboard|^/pydio/settings|^/pydio/welcome|^/ws- /pydio/index.php last;
        
   if ( !-e $request_filename ) {
      # WebDAV Rewrites
      rewrite ^/pydio/shares /pydio/dav.php last;
      # Sync client
      rewrite ^/pydio/api /pydio/rest.php last;
      # External users 
      rewrite ^/pydio/user /pydio/index.php?get_action=user_access_point last;
      # Public shares
      rewrite ^/pydio/data/public/([a-zA-Z0-9_-]+)\.php$ /pydio/data/public/share.php?hash=$1?;
   }
   rewrite ^/pydio/data/public/([a-zA-Z0-9_-]+)--([a-z]+)$ /pydio/data/public/share.php?hash=$1&lang=$2?;
   rewrite ^/pydio/data/public/([a-zA-Z0-9_-]+)$ /pydio/data/public/share.php?hash=$1?;

   # Prevent Clickjacking
   add_header X-Frame-Options "SAMEORIGIN";

   # Only allow these request methods and do not accept DELETE, SEARCH and other methods
   if ( $request_method !~ ^(GET|HEAD|POST|PROPFIND|OPTIONS)$ ) {
      return 444;
   }

   location ~* ^/pydio/(?:\.|conf|data/(?:files|personal|logs|plugins|tmp|cache)|plugins/editor.zoho/agent/files) {
      deny all;
   }

   # Enables Caching
   location ~* \.(ico|css|js)$ {
      expires 7d;
      add_header Pragma public;
      add_header Cache-Control "public, must-revalidate, proxy-revalidate";
   }
`

I wrote a nginx conf file for pydio!

pydio works fine with it in /etc/nignx/sites-dietpi, even webdav and file share link redirection are OK!

So, Please. Apply this file to next dietpi release.

Then, I'll be happy for my little contribution to this wonderful project.

Fourdee, 2016-04-24:
@studioai 
Thanks for this, although could you please explain the differences in your configuration file and its benefits, when compared to our existing one?

studioai, 2016-04-24:
As far I know there's no existing one for nginx. Only apache configuration file is given for pydio installation. So, I googled for nginx configuration for pydio and modified it.

It is automatically included if it in /etc/nignx/sites-dietpi directory with .config extention by default nginx configuration file.

this is for who installed LEMP (NginX + MySQL + Php) Stack and Installed Pydio later.

Fourdee, 2016-04-24:
@studioai 
As of DietPi version 115, all the ```dietpi-software``` webbased installations now support Lighttpd, Nginx and apache2 stacks. Rob did alot of work on getting the Nginx and Lighttpd installations and configurations working.

What version are you running?

studioai, 2016-04-24:
@Fourdee 
I have 116 Installed.

There was /etc/nginx/sites-enabled/default only. Am I missing something?

If I'm not, default configuration supports php well. but doen't go well with Pydio's WebDav service and File Sharing via URL because there is no proper rewrite options.

What I wrote is additional configuration file can be add on to default conf. and that defines rewrite rules for pydio.

Fourdee, 2016-07-23:
Outdated ticket, making as closed. Please reopen if required.

closed 2016-07-23 by Fourdee