MichaIng/DietPi #300: Update netdata to v1.1.0

WolfganP, 2016-04-25:
NetData released v1.1.0 5 days ago https://github.com/firehol/netdata/releases/tag/v1.1.0 , official tarfiles released at http://firehol.org/download/netdata/releases/v1.1.0

rhkean, 2016-04-25:
Those tarballs are source code, not binaries.

But, I'll look over the changes when I get a moment.

WolfganP, 2016-04-25:
Thx @rhkean I found a relevant discussion on debian packaging at https://github.com/firehol/netdata/pull/132

rhkean, 2016-04-25:
Wow... I thought some of our Issue conversations were long.... LOL

looks like they're adding deb package builds.  that'll make it easier...  Probably won't get to it this week, though.

Fourdee, 2016-07-09:
I'll grab this, its 1.2 currently. Time to get compiling:
:u5408: armv6
:u5408: armv7+
:u5408: arm64
:u5408: x86_64

Fourdee, 2016-07-09:
Netdata on RPi 1 really isnt a good idea lol:
[image]

Fourdee, 2016-07-09:
Completed: https://github.com/Fourdee/DietPi/commit/1ce2f5043133cd1a1f1481ba9a55affbbecdd21c

closed 2016-07-09 by Fourdee