MichaIng/DietPi #306: DietPi-Config | Add ability to enable/disable serial console

Fourdee, 2016-04-30:
Mainly for RPi, but will look into supporting Odroid's aswell.

axel-h, 2016-05-16:
Enable and disable it via a config is a nice thing. However, the serial console should be enable by default. Currently (v115, Odrioid-C1) the boot.ini has three entries (console on serial, video, serial+video), and by default the option "video" is enabled. This should be "serial+video" by default, so everybody can see what is going on during 1st and 2nd boot. Later when the config menu is presented, there could be an option to choose what to keep on.
I think a certain share of people still brings up the device using the serial console. And it is very disturbing when nothing happen there. The docs says that it re-sizes the SD card after the 1st reboot, so you are scared to turn it off if nothing happens. In my case, it turned out, it was waiting for the network to become available. But all it said in the serial console is "Starting kernel ..."

Fourdee, 2016-05-17:
@axel-h 
Fair comments and good feedback :+1: . Although I think the number of our users who actually use the serial line, its probably <1%

Once i've implemented a serial switch menu in ```dietpi-config```, I'll make a note to go through all our images and enable serial connection by default. 

The only downside is all our images will need to be redone for this the serial line to be active by default. So this change wont have any effect, at least until all our images updated with the changes. 
Updating all the DietPi images is a long and lengthy process and we try to avoid doing it unless absolutely necessary, or when they are 8+ release cycles behind the current version.

I'll leave the ticket open and i'll let you know when we are looking to update the images.

Fourdee, 2016-05-17:
@axel-h 
I've created a separate ticket for serial console enabled by default: https://github.com/Fourdee/DietPi/issues/343

axel-h, 2016-05-17:
really <1% ? Maybe in the daily life. But actually I found it quite useful to keep the serial console working. In case something is not working, this is a change to connect  to the device and figure out what is broken. Usually the serial console does not die so easy. The only reason why one would really disable this is either because they need the serial console for something else or paranoia. A weak root password might make is easy to login there - but hey, I have physical access anyway, so there are way without the root password.
So maybe there should be two options actually:
- "keep serial console active for boot log and kernel messages"
- "allow login from serial console"

Fourdee, 2016-05-24:
Notes: Tested with script
:u6307: C2
:u6e80: C1
:u6e80: XU4
:u6307: RPi Jessie
:u6e80: RPi Wheezy
:u6e80: BPi M2+
:u6e80: OrangePi PC
:u6e80: Pine A64

Fourdee, 2016-05-25:
Ok its in dietpi-config, I need to test every device with new script and PL2303 serial adapter. Fun! lol

Still not sure how to resolve 100% cpu for Rpi when serial console is enabled, yet.

axel-h, 2016-05-25:
Where do the 100% come from - is this new on V119 now? I've never seen this up to V118 on my Odroid-C1 board. There a "/sbin/agetty --keep-baud 115200 38400 9600 ttyS0 vt102" running that took 0:00.01 CPU time.

Fourdee, 2016-05-25:
@axel-h 
Not sure to be honest, although this seems to be a RPi issue (Occurs with RPi3). 

I tested Odroids with v119 today and all are fine.

Fourdee, 2016-05-26:
Looks like hciattach is causing the 100% cpu:
```
systemctl disable hciuart.service
reboot
```
fine.

Attempt to start hciuart after boot:
```
root@DietPi:~# service hciuart start
Job for hciuart.service failed. See 'systemctl status hciuart.service' and 'journalctl -xn' for details.
root@DietPi:~# systemctl status hciuart.service
● hciuart.service - Configure Bluetooth Modems connected by UART
   Loaded: loaded (/lib/systemd/system/hciuart.service; disabled)
   Active: failed (Result: exit-code) since Thu 2016-05-26 15:55:27 BST; 5s ago
  Process: 1392 ExecStart=/usr/bin/hciattach /dev/serial1 bcm43xx 921600 noflow - (code=exited, status=1/FAILURE)

May 26 15:55:27 DietPi hciattach[1392]: Failed to reset chip, invalid HCI event
May 26 15:55:27 DietPi hciattach[1392]: Can't initialize device: Success
May 26 15:55:27 DietPi hciattach[1392]: bcm43xx_init
May 26 15:55:27 DietPi systemd[1]: hciuart.service: control process exited,...=1
May 26 15:55:27 DietPi systemd[1]: Failed to start Configure Bluetooth Mode...T.
May 26 15:55:27 DietPi systemd[1]: Unit hciuart.service entered failed state.
Hint: Some lines were ellipsized, use -l to show in full.
root@DietPi:~# systemctl status hciuart.service -l
● hciuart.service - Configure Bluetooth Modems connected by UART
   Loaded: loaded (/lib/systemd/system/hciuart.service; disabled)
   Active: failed (Result: exit-code) since Thu 2016-05-26 15:55:27 BST; 8s ago
  Process: 1392 ExecStart=/usr/bin/hciattach /dev/serial1 bcm43xx 921600 noflow - (code=exited, status=1/FAILURE)

May 26 15:55:27 DietPi hciattach[1392]: Failed to reset chip, invalid HCI event
May 26 15:55:27 DietPi hciattach[1392]: Can't initialize device: Success
May 26 15:55:27 DietPi hciattach[1392]: bcm43xx_init
May 26 15:55:27 DietPi systemd[1]: hciuart.service: control process exited, code=exited status=1
May 26 15:55:27 DietPi systemd[1]: Failed to start Configure Bluetooth Modems connected by UART.
May 26 15:55:27 DietPi systemd[1]: Unit hciuart.service entered failed state.
```

Fourdee, 2016-05-28:
RPi3 notes:
- Requires core_freq=250mhz, else, corrupt characters (https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=138162)
- Uses ```serial-getty@ttyS0``` service instead of ```serial-getty@ttyAMA0.service```
- ```serial-getty@ttyAMA0.service``` must be disabled as this generates 100% cpu with hciuart.
- /boot/cmdline.txt ```console=ttyS0,115200``` and ```console=ttyS0,115200``` both work. Doesnt seem to have an effect on the active tty, just enables serial console.

Fourdee, 2016-05-29:
Add question in DietPi-Software during 1st run setup, eg: If user wants to disable serial console.

Fourdee, 2016-05-29:
Done. 

Although Pine64 will remain untested (do not have one yet), unless, @rhkean you have a serial adapter and can test it? https://github.com/Fourdee/DietPi/issues/306#issuecomment-221587312

I'll mark this as closed. will reopen if needed.

closed 2016-05-29 by Fourdee

rhkean, 2016-05-29:
Since the kernel for the Pine64 is still in flux, it's probably not a good idea to disable its serial output.

But, I'll look at when I get home tomorrow night.