MichaIng/DietPi #309: DietPi for Xenserver

Fourdee, 2016-05-01:
> Test/release the DietPi for XenServer implementation and image created by @xenfomation (JK)

@xenfomation ^^ so i dont forget :)

xenfomation, 2016-05-02:
Thanks for letting me do this, @Fourdee -- I am wrapping up the documentation as OpenSource XenServer is my primary virtualization environment.  The screenshot is from XenCenter, but I am working with the code base through SSH (of course)!
[image]

xenfomation, 2016-05-02:
The native format for XenServer VM exports are XVA files.  Sparse disks, meaning efficient use of compacting the VM, and I have tested it within XenServer 6.2 and 6.5 -- works great.

xenfomation, 2016-05-02:
Attached is an additional text - based on the mock-up I sent you previously - to explain where the XenServer resources are at.  That's all as to be blunt, the public documentation is sufficient enough and outside of scope to cover installing XenServer.  The additional information explains how to import the XVA file with XenCenter. 
[xenserverInstructions.txt](https://github.com/Fourdee/DietPi/files/244470/xenserverInstructions.txt)

xenfomation, 2016-05-07:
@Fourdee  -- Of course it sounds like you are all over this, but if I have left anything unclear -- let me know and I shall grant thee access to my private XenServer if you want to see how sweet your sandbox/X64 environment runs (as it does else where)!

Yeah - seriously, no sucking up on my part.  Best part about this project?  I use it.  :-)

Fourdee, 2016-05-07:
@xenfomation 

> Of course it sounds like you are all over this

Actually, not at the moment :). I didn't realise you can run XenServer virtualization on a Windows system lol. I also dont have a x32/x64 system available to dedicate a installation.

> let me know and I shall grant thee access to my private XenServer

This would be ideal and I'd be greatful if we could make this happen. 
All i want to do is some basic tests with the image before its released. The VirtualBox image you did was spot on, so I cant see any issues, its just something I like to do with every image.

xenfomation, 2016-05-07:
Completely understood!  Tomorrow first, EST, I you shall have full access!

Fourdee, 2016-05-13:
@xenfomation 
Sending a "prod and poke" your way for Xenserver access :)

xenfomation, 2016-05-13:
Don't blame you and thanks!!  Moved one of XenServers to my DMZ and will have that ready to go tonight. All you need will be XenCenter for XenServer 6.5 and the connection criteria.  Thanks!

Fourdee, 2016-05-13:
@xenfomation 
Thanks :+1: . If you can email me details i'd appreciate it, i'll also try it tomorrow.

Fourdee, 2016-05-15:
Going to move this outside the milestone as there is no rush. Give me a bell when your ready @xenfomation :+1:

xenfomation, 2016-05-15:
Thanks. The good news is that I have things ready to go for you, but your executive call gives us time to test your remote connection (three methods), etc. 

Thanks, @Foudree

xenfomation, 2016-05-21:
@Fourdee  --  Tested download of image and import and all looks fantastic.

[image]
[image]
[image]
[image]

Fourdee, 2016-05-21:
@xenfomation 
Excellent,, thanks for this :+1: 

I'll update the website/forums today and get this image released. Great work JK :+1:

Fourdee, 2016-05-23:
@xenfomation 
Just need the "setup guide" for XenServer and the forums when you get a chance. Then we can close this ticket :+1:

xenfomation, 2016-05-25:
Thanks for the patience, @Fourdee & @rhkean  --  most of what I am documenting can be found on XenServer's public, open source facing website ([http://xenserver.org/](http://xenserver.org/)).  However, I am consolidating this to explain what XenServer is, where documentation is available, and how to use the Windows-centric tool to import the XVA file.

Fourdee, 2016-07-23:
Minus the missing documentation, making as closed.

closed 2016-07-23 by Fourdee