MichaIng/DietPi #310: DietPi-Software | Add installation option for Emby Server

Fourdee, 2016-05-01:
> http://dietpi.com/phpbb/viewtopic.php?f=12&t=365#p1587
https://emby.media/linux-server.html

xenfomation, 2016-05-02:
Just for some info on some pain previous contributors of emby documented for public success!

[https://discourse.osmc.tv/t/howto-install-emby-server-broken/6364/23](https://discourse.osmc.tv/t/howto-install-emby-server-broken/6364/23)

Fourdee, 2016-05-02:
hmmmmmm
```
root@DietPi:~# apt-get install emby-server
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 emby-server : Depends: mono-runtime (>= 3.0~) but it is not installable
               Depends: libmono-corlib4.5-cil (>= 4.0.0~alpha1) but it is not going to be installed
               Depends: libmono-microsoft-csharp4.0-cil (>= 1.0) but it is not going to be installed
               Depends: libmono-posix4.0-cil (>= 4.0.0~alpha1) but it is not going to be installed
               Depends: libmono-security4.0-cil (>= 3.0.6) but it is not going to be installed
               Depends: libmono-system-configuration4.0-cil (>= 4.0.0~alpha1) but it is not going to be installed
               Depends: libmono-system-core4.0-cil (>= 4.0.0~alpha1) but it is not going to be installed
               Depends: libmono-system-data-linq4.0-cil (>= 1.0) but it is not going to be installed
               Depends: libmono-system-data4.0-cil (>= 4.0.0~alpha1) but it is not going to be installed
               Depends: libmono-system-drawing4.0-cil (>= 3.0.6) but it is not going to be installed
               Depends: libmono-system-io-compression4.0-cil (>= 3.2.1) but it is not going to be installed
               Depends: libmono-system-runtime-serialization4.0-cil (>= 4.0.0~alpha1) but it is not going to be installed
               Depends: libmono-system-servicemodel4.0a-cil (>= 3.2.3) but it is not going to be installed
               Depends: libmono-system-transactions4.0-cil (>= 1.0) but it is not going to be installed
               Depends: libmono-system-web4.0-cil (>= 2.10.3) but it is not going to be installed
               Depends: libmono-system-xml-linq4.0-cil (>= 3.0.6) but it is not going to be installed
               Depends: libmono-system-xml4.0-cil (>= 3.12.0) but it is not going to be installed
               Depends: libmono-system4.0-cil (>= 4.0.0~alpha1) but it is not going to be installed
               Depends: mono-devel (>= 1.0) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```

rhkean, 2016-05-02:
doesn't look like it's a supported platform yet...

There's not even a fork in the nightly tree... :frowning: 

http://download.mono-project.com/repo/debian/dists/wheezy/main/

not surprising, really.... Microsoft barely acknowledges 64-bit for the .Net framework.

Fourdee, 2016-05-02:
RPi 3:
```
root@DietPi-rpi3:~# apt-get install emby-server
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 emby-server : Depends: mono-runtime (>= 3.0~) but it is not going to be installed
               Depends: libmono-corlib4.5-cil (>= 4.0.0~alpha1) but it is not going to be installed
               Depends: libmono-microsoft-csharp4.0-cil (>= 1.0) but it is not going to be installed
               Depends: libmono-posix4.0-cil (>= 4.0.0~alpha1) but it is not going to be installed
               Depends: libmono-security4.0-cil (>= 3.0.6) but it is not going to be installed
               Depends: libmono-system-configuration4.0-cil (>= 4.0.0~alpha1) but it is not going to be installed
               Depends: libmono-system-core4.0-cil (>= 4.0.0~alpha1) but it is not going to be installed
               Depends: libmono-system-data-linq4.0-cil (>= 1.0) but it is not going to be installed
               Depends: libmono-system-data4.0-cil (>= 4.0.0~alpha1) but it is not going to be installed
               Depends: libmono-system-drawing4.0-cil (>= 3.0.6) but it is not going to be installed
               Depends: libmono-system-io-compression4.0-cil (>= 3.2.1) but it is not going to be installed
               Depends: libmono-system-runtime-serialization4.0-cil (>= 4.0.0~alpha1) but it is not going to be installed
               Depends: libmono-system-servicemodel4.0a-cil (>= 3.2.3) but it is not going to be installed
               Depends: libmono-system-transactions4.0-cil (>= 1.0) but it is not going to be installed
               Depends: libmono-system-web4.0-cil (>= 2.10.3) but it is not going to be installed
               Depends: libmono-system-xml-linq4.0-cil (>= 3.0.6) but it is not going to be installed
               Depends: libmono-system-xml4.0-cil (>= 3.12.0) but it is not going to be installed
               Depends: libmono-system4.0-cil (>= 4.0.0~alpha1) but it is not going to be installed
               Depends: mono-devel (>= 1.0) but it is not going to be installed
               Recommends: libembymagickwand-6.q8-2 but it is not installable
E: Unable to correct problems, you have held broken packages.
```

Going to be "one of those days" lol

Fourdee, 2016-05-02:
Ok , looks like an issue with the emby repo:

```
wget http://download.opensuse.org/repositories/home:emby/Debian_8.0/Release.key
apt-key add - < Release.key 

echo -e "deb http://download.opensuse.org/repositories/home:/emby/Debian_8.0/ /" > /etc/apt/sources.list.d/emby-server.list 
apt-get update
apt-get install emby-server
```

```
root@DietPi-rpi3:~# apt-get install mono-runtime
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 mono-runtime : Depends: mono-runtime-sgen (= 3.2.8+dfsg-10) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
root@DietPi-rpi3:~# rm /etc/apt/sources.list.d/emby-server.list && apt-get update

root@DietPi-rpi3:~# apt-get install mono-runtime
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following extra packages will be installed:
  binfmt-support cli-common libmono-corlib4.5-cil libmono-i18n-west4.0-cil
  libmono-i18n4.0-cil libmono-security4.0-cil
  libmono-system-configuration4.0-cil libmono-system-security4.0-cil
  libmono-system-xml4.0-cil libmono-system4.0-cil mono-4.0-gac mono-gac
  mono-runtime-common mono-runtime-sgen
Suggested packages:
  libmono-i18n4.0-all libasound2 libgamin0 xdg-utils libgnome2-0 konqueror
The following NEW packages will be installed:
  binfmt-support cli-common libmono-corlib4.5-cil libmono-i18n-west4.0-cil
  libmono-i18n4.0-cil libmono-security4.0-cil
  libmono-system-configuration4.0-cil libmono-system-security4.0-cil
  libmono-system-xml4.0-cil libmono-system4.0-cil mono-4.0-gac mono-gac
  mono-runtime mono-runtime-common mono-runtime-sgen
```

rhkean, 2016-05-02:
Here's the best 32-bit install, though....
I use this on RPi and Odroid for my Sonarr install...
this will install the latest Mono 4.2 from Xamarin's repo...

```
# Add Mono Project GPG signing key.
apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 3FA7E0328081BFF6A14DA29AA6A19B38D3D831EF

# Add Mono Project repositories.
echo "deb http://download.mono-project.com/repo/debian wheezy main" | tee /etc/apt/sources.list.d/mono-xamarin.list
echo "deb http://download.mono-project.com/repo/debian wheezy-libjpeg62-compat main" | tee -a /etc/apt/sources.list.d/mono-xamarin.list 

apt-get update
```

Fourdee, 2016-05-02:
@rhkean 
Legend :+1: 

Shes a big one lol:
```
root@DietPi-rpi3:~# apt-get install emby-server
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following extra packages will be installed:
  ca-certificates-mono dbus-x11 file fontconfig fontconfig-config
  fonts-dejavu-core gamin gconf-service gconf2 gconf2-common gnome-mime-data
  hicolor-icon-theme libart-2.0-2 libasound2 libasound2-data libatk1.0-0
  libatk1.0-data libavahi-client3 libavahi-common-data libavahi-common3
  libavahi-glib1 libbonobo2-0 libbonobo2-common libbonoboui2-0
  libbonoboui2-common libcairo2 libcanberra0 libcroco3 libcups2 libdatrie1
  libdbus-glib-1-2 libexif12 libfontconfig1 libfreetype6 libgail18 libgamin0
  libgconf-2-4 libgdiplus libgdk-pixbuf2.0-0 libgdk-pixbuf2.0-common libgif4
  libglade2-0 libgnome-2-0 libgnome-keyring-common libgnome-keyring0
  libgnome2-common libgnomecanvas2-0 libgnomecanvas2-common libgnomeui-0
  libgnomeui-common libgnomevfs2-0 libgnomevfs2-common libgnomevfs2-extra
  libgraphite2-3 libgtk2.0-0 libgtk2.0-bin libgtk2.0-common libharfbuzz0b
  libice6 libjasper1 libjbig0 libjpeg62-turbo libltdl7 libmagic1
  libmono-2.0-dev libmono-accessibility4.0-cil libmono-c5-1.1-cil
  libmono-cairo4.0-cil libmono-cecil-private-cil libmono-cil-dev
  libmono-codecontracts4.0-cil libmono-compilerservices-symbolwriter4.0-cil
  libmono-corlib4.5-cil libmono-cscompmgd0.0-cil libmono-csharp4.0c-cil
  libmono-custommarshalers4.0-cil libmono-data-tds4.0-cil libmono-db2-1.0-cil
  libmono-debugger-soft4.0a-cil libmono-http4.0-cil libmono-i18n-cjk4.0-cil
  libmono-i18n-mideast4.0-cil libmono-i18n-other4.0-cil
  libmono-i18n-rare4.0-cil libmono-i18n-west4.0-cil libmono-i18n4.0-all
  libmono-i18n4.0-cil libmono-ldap4.0-cil libmono-management4.0-cil
  libmono-messaging-rabbitmq4.0-cil libmono-messaging4.0-cil
  libmono-microsoft-build-engine4.0-cil
  libmono-microsoft-build-framework4.0-cil
  libmono-microsoft-build-tasks-v4.0-4.0-cil
  libmono-microsoft-build-utilities-v4.0-4.0-cil
  libmono-microsoft-build4.0-cil libmono-microsoft-csharp4.0-cil
  libmono-microsoft-visualc10.0-cil
  libmono-microsoft-web-infrastructure1.0-cil libmono-oracle4.0-cil
  libmono-parallel4.0-cil libmono-peapi4.0a-cil libmono-posix4.0-cil
  libmono-rabbitmq4.0-cil libmono-relaxng4.0-cil libmono-security4.0-cil
  libmono-sharpzip4.84-cil libmono-simd4.0-cil libmono-smdiagnostics0.0-cil
  libmono-sqlite4.0-cil libmono-system-componentmodel-composition4.0-cil
  libmono-system-componentmodel-dataannotations4.0-cil
  libmono-system-configuration-install4.0-cil
  libmono-system-configuration4.0-cil libmono-system-core4.0-cil
  libmono-system-data-datasetextensions4.0-cil
  libmono-system-data-entity4.0-cil libmono-system-data-linq4.0-cil
  libmono-system-data-services-client4.0-cil
  libmono-system-data-services4.0-cil libmono-system-data4.0-cil
  libmono-system-design4.0-cil libmono-system-drawing-design4.0-cil
  libmono-system-drawing4.0-cil libmono-system-dynamic4.0-cil
  libmono-system-enterpriseservices4.0-cil
  libmono-system-identitymodel-selectors4.0-cil
  libmono-system-identitymodel4.0-cil
  libmono-system-io-compression-filesystem4.0-cil
  libmono-system-io-compression4.0-cil libmono-system-json-microsoft4.0-cil
  libmono-system-json4.0-cil libmono-system-ldap-protocols4.0-cil
  libmono-system-ldap4.0-cil libmono-system-management4.0-cil
  libmono-system-messaging4.0-cil libmono-system-net-http-formatting4.0-cil
  libmono-system-net-http-webrequest4.0-cil libmono-system-net-http4.0-cil
  libmono-system-net4.0-cil libmono-system-numerics4.0-cil
  libmono-system-reactive-core2.2-cil libmono-system-reactive-debugger2.2-cil
  libmono-system-reactive-experimental2.2-cil
  libmono-system-reactive-interfaces2.2-cil
  libmono-system-reactive-linq2.2-cil
  libmono-system-reactive-observable-aliases0.0-cil
  libmono-system-reactive-platformservices2.2-cil
  libmono-system-reactive-providers2.2-cil
[... 141 lines cut ...]
  libmono-webbrowser4.0-cil libmono-webmatrix-data4.0-cil
  libmono-windowsbase4.0-cil libmono-xbuild-tasks4.0-cil libmonoboehm-2.0-1
  libmonoboehm-2.0-dev libnunit-cil-dev libnunit-console-runner2.6.3-cil
  libnunit-core-interfaces2.6.3-cil libnunit-core2.6.3-cil
  libnunit-framework2.6.3-cil libnunit-mocks2.6.3-cil libnunit-util2.6.3-cil
  libogg0 liborbit-2-0 libpango-1.0-0 libpangocairo-1.0-0 libpangoft2-1.0-0
  libpixman-1-0 libpython-stdlib libpython2.7-minimal libpython2.7-stdlib
  librsvg2-2 librsvg2-common libsm6 libsqlite3-0 libtdb1 libthai-data libthai0
  libtiff5 libvorbis0a libvorbisfile3 libx11-6 libx11-data libxau6
  libxcb-render0 libxcb-shm0 libxcb1 libxcomposite1 libxcursor1 libxdamage1
  libxdmcp6 libxext6 libxfixes3 libxi6 libxinerama1 libxml2 libxrandr2
  libxrender1 mime-support mono-csharp-shell mono-devel mono-mcs mono-xbuild
  python python-minimal python2.7 python2.7-minimal sgml-base shared-mime-info
  ucf x11-common xml-core
The following packages will be upgraded:
  libmono-corlib4.5-cil libmono-i18n-west4.0-cil libmono-i18n4.0-cil
  libmono-security4.0-cil libmono-system-configuration4.0-cil
  libmono-system-security4.0-cil libmono-system-xml4.0-cil
  libmono-system4.0-cil mono-4.0-gac mono-gac mono-runtime mono-runtime-common
  mono-runtime-sgen
13 upgraded, 248 newly installed, 0 to remove and 0 not upgraded.
Need to get 77.7 MB of archives.
After this operation, 264 MB of additional disk space will be used.
Do you want to continue? [Y/n] n
```

rhkean, 2016-05-02:
yeah...  the mono project decided to bundle most assemblies as individual packages... 
the packages seem to be 1 or 2 dll's, a changelog, and a copyright file

the mono-runtime package is mainly just a wrapper for all the base .dll's

Fourdee, 2016-05-02:
@rhkean Interesting. Never used Mono before.

Installed on RPi3 fine, but, wont play media. I need to do some digging.
[image]

rhkean, 2016-05-02:
My day job is a .Net web application developer

So, I play with mono here and there to ease cross platform development

Fourdee, 2016-05-02:
@rhkean Awesome. Any programs you've done that you are allowed to show? Be interesting to see what you do :)

```
Warn, App, ffmpeg is missing decoder h264_qsv
Warn, App, ffmpeg is missing decoder mpeg2_qsv
Warn, App, ffmpeg is missing decoder vc1_qsv
```

Installed ffmpeg, running fine :)

Fourdee, 2016-05-02:
Transcoding performance is dreadful, even on RPi3:
[image]

rhkean, 2016-05-02:
not surprising....

mono on an SBC.... LOL

Sonarr chews up cpu and memory... and it's basically just a self-hosting web application

Fourdee, 2016-05-02:
@rhkean 
Aye, looks like users should only expect audio playback through emby on SBC's. Video transcoding cant keep up, feels like 10fps transcoding speed for 24fps media.

Transcoding options are useless (intel quickstep, mmm, I even tried it whilst sending a wish to my RPi lol):
[image]

xenfomation, 2016-05-06:
Despite the hard work and my own with the while warnings of h264, etc... is emby looking a little "non DietPi-ish?"

Fourdee, 2016-05-06:
@xenfomation 
I've added the basic install code in. But for SBC you should expect audio streaming only. The VM's (depending on host spec) and possibly XU4 should run video transcoding fine.

I'll see if we can also automate the initial Emby web setup for users, once I've finished off with the OrangePi PC image, although, I have this near uncontrollable urge to throw the OPi PC in the bin lol :).

xenfomation, 2016-05-09:
Thanks, @Fourdee -- even if we can get that nudge, definitely worth the learning and DietAdmin experience.

Fourdee, 2016-05-09:
> note to self: arm64 packages not available with mono repo:
```
W: Failed to fetch http://download.mono-project.com/repo/debian/dists/wheezy/InRelease  Unable to find expected entry 'main/binary-arm64/Packages' in Release file (Wrong sources.list entry or malformed file)

W: Failed to fetch http://download.mono-project.com/repo/debian/dists/wheezy-libjpeg62-compat/InRelease  Unable to find expected entry 'main/binary-arm64/Packages' in Release file (Wrong sources.list entry or malformed file)
```

Fourdee, 2016-05-13:
Looks like no ARM64 packages are currently being built by http://download.mono-project.com/ : http://download.mono-project.com/repo/debian/dists/

Disabled C2 installation option:
```
		#Emby server (mono libary), not available for ARM64: https://github.com/Fourdee/DietPi/issues/310#issuecomment-217946474
		if (( $EMBYSERVER == 1 && $HW_MODEL == 12 )); then
			WHIP_TITLE='Emby Server'
			WHIP_QUESTION='Emby Server is not currently available for ARM64 (Odroid C2) and will not be installed. Your choice has been removed.\n\nhttps://github.com/Fourdee/DietPi/issues/310#issuecomment-217946474'
			whiptail --title "$WHIP_TITLE" --msgbox "$WHIP_QUESTION" --backtitle "$WHIP_BACKTITLE" 13 75

			EMBYSERVER=0

		fi
```

Fourdee, 2016-05-13:
Transcoding is great on VM:
AMD Athlon 5150 | 1.6ghz quad
[image]

Fourdee, 2016-05-13:
Done http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=1789#p1789

closed 2016-05-13 by Fourdee