MichaIng/DietPi #311: DietPi-Software | Several setup improvement for existing installation options

k-plan, 2016-05-01:
_1. WiFi HotSpot_ ( [see issue 273 ](https://github.com/Fourdee/DietPi/issues/273#issuecomment-215996025) )

- DNS-Server selection (like new menu under interfaces) in dietpi-config
- possibility to change DHCP area and Gateway address in dietpi-config
- possibility to change 802.11g to 802.11n via dietpi-config, if hardware support it.

_2. TOR HotSpot_

- DNS-Server selection (like new menu under interfaces) in dietpi-config
- possibility to change DHCP area and Gateway address in dietpi-config
- possibility to change 802.11g to 802.11n via dietpi-config, if hardware support it.

_3. PiHole_ ( [see issue 252](https://github.com/Fourdee/DietPi/issues/252#issuecomment-201905243) )

- DNS-Server selection (like new menu under interfaces) during installation, after question for static IP address.
- cronjob that will update gravity.sh (or the new `pihole -g` option) with update logging
- Implementing some of the "new" 2.6.2 PiHole features

@Fourdee 
don't want do start x-issues, so
you can split suit yourself or kick them out.
But it's all only ideas / suggestion, no really issues.

Fourdee, 2016-05-01:
@k-plan 
All great suggestions and improvements :+1: . I will take a look into them when I start v118.

xenfomation, 2016-05-07:
Feel free to divy out as thanks to you & @rhkean, I am getting quite cozy with my favorite distro - ARM, X64, and Virtual.  Whiptail and menus are epic.  Then again, I am possibly having too much fun!

Fourdee, 2016-05-07:
@xenfomation 
Lol, legend :+1:. We love whiptail at DietPi hehe ;)

Fourdee, 2016-05-08:
Linking for reference: https://github.com/Fourdee/DietPi/issues/313#issuecomment-217729027

Fourdee, 2016-05-13:
@k-plan 
Going to move this to v119. v118 has too many open jobs, and still lots to do before we can release v118.

k-plan, 2016-05-13:
@Fourdee 

Clean up for v118 release? :smiley: 

No problem, it's okay (for me).

If I have some spare time at weekend, I will start with WiFi HotSpot  802.11n testing (2.4 and 5 GHz) and  regulatory domain setting. (I hope so)

Fourdee, 2016-05-13:
@k-plan 

> Clean up for v118 release? 

Hehe, yes. There is so much going at the moment and I just want to make sure we can close all those tickets. I'am also not familiar with a 2+ week update cycle lol.

Fourdee, 2016-05-29:
@k-plan 

Will go through these, i'll update it as and when:

#### WiFi Hotspot - DNS-Server selection (like new menu under interfaces) in dietpi-config
Hostapd uses the DNS server of eth device. So users can change the eth DNS server to affect both.

Fourdee, 2016-06-01:
> Implementing some of the "new" 2.6.2 PiHole features

Updated Pi-Hole to latest for v119 to prep for this.

Fourdee, 2016-06-15:
@k-plan 

> PiHole DNS-Server selection (like new menu under interfaces) during installation, after question for static IP address.

In dietpi-config: 
Changing static DNS servers will now also be applied to PiHole (if installed). This needs to be completed by the user after PiHole has been installed. I'll make a documentation note for users.

Fourdee, 2016-06-24:
> In dietpi-config: 
Changing static DNS servers will now also be applied to PiHole (if installed). This needs to be completed by the user after PiHole has been installed. I'll make a documentation note for users.

Fourdee, 2016-06-30:
@k-plan 
Added cron based weekly PiHole adlist update for PiHole installations by default.

k-plan, 2016-06-30:
> > In dietpi-config: 
Changing static DNS servers will now also be applied to PiHole (if installed). This needs to be completed by the user after PiHole has been installed. I'll make a documentation note for users.

>I've reverted this change, didn't like it, and it was a bit rushed to be honest. I've updated the online docs (http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=174#p174) for now, but i'll take a fresh look after v120 is released.

Me too. Didn't like this. Better to have independent configurations for DNS server.

> PiHole will automatically update the adlist on a weekly basis via

Nice! 😄  Will have a look, after our remote session.

k-plan, 2016-07-03:
@Fourdee 

testing on fresh install Odroid C2 (v.122 with lighttp):

error message:
`Refresh lists in dnsmasq...sed: can't read /etc/dnsmasq.d/01-pihole.conf: No such file or directory
 done!`

```
root@Odroid-C2:~# gravity.sh
:::
::: You are root.
::: Neutrino emissions detected...
:::
::: No custom adlist file detected, reading from default file... done!
:::
::: Existing pihole directory found
:::
::: Getting raw.githubusercontent.com list... List updated, transport successful!
::: Getting adblock.gjtech.net list... No changes detected, transport skipped!
::: Getting mirror1.malwaredomains.com list... No changes detected, transport skipped!
::: Getting sysctl.org list... No changes detected, transport skipped!
::: Getting zeustracker.abuse.ch list... No changes detected, transport skipped!
::: Getting s3.amazonaws.com list... No changes detected, transport skipped!
::: Getting s3.amazonaws.com list... No changes detected, transport skipped!
::: Getting hosts-file.net list... List updated, transport successful!
::: Getting raw.githubusercontent.com list... List updated, transport successful!
:::
::: Aggregating list of domains... done!
::: Formatting list of domains to remove comments.... done!
::: 78957 domains being pulled in by gravity...
::: Removing duplicate domains.... done!
::: 62492 unique domains trapped in the event horizon.
::: Formatting domains into a HOSTS file...
:::
::: Adding 9 adlist sources to the whitelist... done!
::: Running whitelist script to update HOSTS file.... 7 domains whitelisted!
::: Running blacklist script to update HOSTS file.... 0 domains blacklisted!
:::
::: Cleaning up un-needed files... done!
:::
::: Refresh lists in dnsmasq...sed: can't read /etc/dnsmasq.d/01-pihole.conf: No such file or directory
 done!

root@Odroid-C2:~# ls -lah /etc/dnsmasq.d/
total 12K
drwxr-xr-x  2 root root 4.0K Jul  3 19:54 .
drwxr-xr-x 75 root root 4.0K Jul  3 20:20 ..
-rw-r--r--  1 root root  211 May  5  2015 README
```

Fourdee, 2016-07-03:
@k-plan 
From what I can tell, ```01-pihole.conf``` is written to once, but not read from in ```gravity.sh```. https://github.com/pi-hole/pi-hole/blob/master/gravity.sh#L316. So it doesn't really effect the update.

Either way, lets fix the error:
```
wget https://raw.githubusercontent.com/pi-hole/pi-hole/master/advanced/01-pihole.conf -O /etc/dnsmasq.d/01-pihole.conf
```

Run gravity again:
```
gravity.sh
```

k-plan, 2016-07-03:
@Fourdee 

```
root@Odroid-C2:~# wget https://raw.githubusercontent.com/pi-hole/pi-hole/master/advanced/01-pihole.conf -O /etc/dnsmasq.d/01-pihole.conf
--2016-07-03 21:09:49--  https://raw.githubusercontent.com/pi-hole/pi-hole/master/advanced/01-pihole.conf
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.12.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.12.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2205 (2.2K) [text/plain]
Saving to: ‘/etc/dnsmasq.d/01-pihole.conf’

/etc/dnsmasq.d/01-pihole.conf            100%[=================================================================================>]   2.15K  --.-KB/s   in 0s

2016-07-03 21:09:49 (9.83 MB/s) - ‘/etc/dnsmasq.d/01-pihole.conf’ saved [2205/2205]

root@Odroid-C2:~# gravity.sh
:::
::: You are root.
::: Neutrino emissions detected...
:::
::: No custom adlist file detected, reading from default file... done!
:::
::: Existing pihole directory found
:::
::: Getting raw.githubusercontent.com list... List updated, transport successful!
::: Getting adblock.gjtech.net list... No changes detected, transport skipped!
::: Getting mirror1.malwaredomains.com list... No changes detected, transport skipped!
::: Getting sysctl.org list... No changes detected, transport skipped!
::: Getting zeustracker.abuse.ch list... No changes detected, transport skipped!
::: Getting s3.amazonaws.com list... No changes detected, transport skipped!
::: Getting s3.amazonaws.com list... No changes detected, transport skipped!
::: Getting hosts-file.net list... List updated, transport successful!
::: Getting raw.githubusercontent.com list... List updated, transport successful!
:::
::: Aggregating list of domains... done!
::: Formatting list of domains to remove comments.... done!
::: 78957 domains being pulled in by gravity...
::: Removing duplicate domains.... done!
::: 62492 unique domains trapped in the event horizon.
::: Formatting domains into a HOSTS file...
:::
::: Adding 9 adlist sources to the whitelist... done!
::: Running whitelist script to update HOSTS file.... 7 domains whitelisted!
::: Running blacklist script to update HOSTS file.... 0 domains blacklisted!
:::
::: Cleaning up un-needed files... done!
:::
::: Refresh lists in dnsmasq... done!
```

Look okay now.  👍

Fourdee, 2016-07-03:
@k-plan 
Excellent :+1: 
I'll get this added to patch and when installed with dietpi-software.

k-plan, 2016-07-03:
@Fourdee 

after a reboot, it seem to be not so easy:

```
root@Odroid-C2:~# gravity.sh

. . .

::: Refresh lists in dnsmasq...Job for dnsmasq.service failed. See 'systemctl status dnsmasq.service' and 'journalctl -xn' for details.
 done!

root@Odroid-C2:~# systemctl status dnsmasq.service
● dnsmasq.service - dnsmasq - A lightweight DHCP and caching DNS server
   Loaded: loaded (/lib/systemd/system/dnsmasq.service; disabled)
  Drop-In: /run/systemd/generator/dnsmasq.service.d
           └─50-dnsmasq-$named.conf, 50-insserv.conf-$named.conf
   Active: failed (Result: exit-code) since Sun 2016-07-03 22:39:03 CEST; 11s ago
  Process: 2810 ExecStart=/etc/init.d/dnsmasq systemd-exec (code=exited, status=1/FAILURE)
  Process: 2807 ExecStartPre=/usr/sbin/dnsmasq --test (code=exited, status=0/SUCCESS)

Jul 03 22:39:03 Odroid-C2 dnsmasq[2807]: dnsmasq: syntax check OK.
Jul 03 22:39:03 Odroid-C2 dnsmasq[2810]: dnsmasq: bad address at line 34 of /etc/dnsmasq.d/01-pihole.conf
Jul 03 22:39:03 Odroid-C2 systemd[1]: dnsmasq.service: control process exited, code=exited status=1
Jul 03 22:39:03 Odroid-C2 systemd[1]: Failed to start dnsmasq - A lightweight DHCP and caching DNS server.
Jul 03 22:39:03 Odroid-C2 systemd[1]: Unit dnsmasq.service entered failed state.
```

Fourdee, 2016-07-04:
@k-plan 
```
bad address at line 34 of /etc/dnsmasq.d/01-pihole.conf
```
If its not too late, can you:
``` cat /etc/dnsmasq.d/01-pihole.conf | sed -n 34p```

k-plan, 2016-07-04:
@Fourdee 

> If its not too late, can you:
`cat /etc/dnsmasq.d/01-pihole.conf | sed -n 34p`

:smiley:  

Yes, have experiment yesterday night a lot, but after uncomment line 34 comes the next error messages appears ...

But anyway, have a running config now. Not sure, if is a "_proper solution_" , but after `reboot` or  `gravity.sh` or `dietpi-services restart` dnsmasq starts without error message and Pihole is working for now.

```
root@Odroid-C2:~# cat /etc/dnsmasq.d/01-pihole.conf

# Pi-hole: A black hole for Internet advertisements
# (c) 2015, 2016 by Jacob Salmela
# Network-wide ad blocking via your Raspberry Pi
# http://pi-hole.net
# dnsmasq config for Pi-hole
#
# Pi-hole is free software: you can redistribute it and/or modify
# it under the terms of the GNU General Public License as published by
# the Free Software Foundation, either version 2 of the License, or
# (at your option) any later version.

# If you want dnsmasq to read another file, as well as /etc/hosts, use
# this.
##addn-hosts=/etc/pihole/gravity.list

# The following two options make you a better netizen, since they
# tell dnsmasq to filter out queries which the public DNS cannot
# answer, and which load the servers (especially the root servers)
# unnecessarily. If you have a dial-on-demand link they also stop
# these requests from bringing up the link unnecessarily.

# Never forward plain names (without a dot or domain part)
domain-needed
# Never forward addresses in the non-routed address spaces.
bogus-priv

# If you don't want dnsmasq to read /etc/resolv.conf or any other
# file, getting its servers from this file instead (see below), then
# uncomment this.
no-resolv

# Add other name servers here, with domain specs if they are for
# non-public domains.
##server=@DNS1@
##server=@DNS2@

# If you want dnsmasq to listen for DHCP and DNS requests only on
# specified interfaces (and the loopback) give the name of the
# interface (eg eth0) here.
##interface=@INT@
# Or which to listen on by address (remember to include 127.0.0.1 if
# you use this.)
##listen-address=127.0.0.1

# Set the cachesize here.
##cache-size=10000

# For debugging purposes, log each DNS query as it passes through
# dnsmasq.
##log-queries
##log-facility=/var/log/pihole.log

# Normally responses which come from /etc/hosts and the DHCP lease
# file have Time-To-Live set as zero, which conventionally means
# do not cache further. If you are happy to trade lower load on the
# server for potentially stale date, you can set a time-to-live (in
# seconds) here.
##local-ttl=300

# This allows it to continue functioning without being blocked by syslog, and allows syslog to use dnsmasq for DNS queries without risking deadlock
log-async
```

```
root@Odroid-C2:~# cat /etc/dnsmasq.conf
#/etc/dnsmasq.conf

addn-hosts=/etc/pihole/gravity.list
domain-needed
bogus-priv
no-resolv
server=8.8.8.8
server=8.8.4.4
interface=eth0
listen-address=127.0.0.1
cache-size=10000
log-queries
log-facility=/var/log/pihole.log
local-ttl=300
```

```
root@Odroid-C2:~# systemctl status dnsmasq.service
● dnsmasq.service - dnsmasq - A lightweight DHCP and caching DNS server
   Loaded: loaded (/lib/systemd/system/dnsmasq.service; disabled)
  Drop-In: /run/systemd/generator/dnsmasq.service.d
           └─50-dnsmasq-$named.conf, 50-insserv.conf-$named.conf
   Active: active (running) since Mon 2016-07-04 01:15:47 CEST; 12h ago
 Main PID: 862 (dnsmasq)
   CGroup: /system.slice/dnsmasq.service
           └─862 /usr/sbin/dnsmasq -x /var/run/dnsmasq/dnsmasq.pid -u www-data -r /var/run/dnsmasq/resolv.conf -7 /etc/dnsmasq.d,.dpkg-dist,.dpkg-old,....

Jul 04 01:15:47 Odroid-C2 dnsmasq[856]: dnsmasq: syntax check OK.
Jul 04 01:15:47 Odroid-C2 systemd[1]: Started dnsmasq - A lightweight DHCP and caching DNS server.

```

k-plan, 2016-07-04:
@Fourdee 

Quick-test:

`root@Odroid-C2:~# cp /etc/dnsmasq.d/01-pihole.conf /etc/dnsmasq.d/01-pihole.conf.bac`

```
root@Odroid-C2:~# cat /etc/dnsmasq.d/01-pihole.conf
# Pi-hole: A black hole for Internet advertisements
# (c) 2015, 2016 by Jacob Salmela
# Network-wide ad blocking via your Raspberry Pi
# http://pi-hole.net
# dnsmasq config for Pi-hole
#
# Pi-hole is free software: you can redistribute it and/or modify
# it under the terms of the GNU General Public License as published by
# the Free Software Foundation, either version 2 of the License, or
# (at your option) any later version.

# If you want dnsmasq to read another file, as well as /etc/hosts, use
# this.
##addn-hosts=/etc/pihole/gravity.list

###### ------------------------------------
## All you needed, to work with DietPi v122
## Anything else has been deleted!
###### ------------------------------------

```

```
root@Odroid-C2:~# gravity.sh
:::
 . . . 
::: Cleaning up un-needed files... done!
:::
::: Refresh lists in dnsmasq... done!

```
```

root@Odroid-C2:~# dietpi-services restart

 DietPi-Services
 Mode: restart
 Please wait...
Ok | cron
Ok | lighttpd
Ok | php5-fpm
Ok | dnsmasq

```
```

root@Odroid-C2:~# systemctl status dnsmasq.service
● dnsmasq.service - dnsmasq - A lightweight DHCP and caching DNS server
   Loaded: loaded (/lib/systemd/system/dnsmasq.service; disabled)
  Drop-In: /run/systemd/generator/dnsmasq.service.d
           └─50-dnsmasq-$named.conf, 50-insserv.conf-$named.conf
   Active: active (running) since Mon 2016-07-04 14:37:17 CEST; 7s ago
  Process: 27393 ExecStop=/etc/init.d/dnsmasq systemd-stop-resolvconf (code=exited, status=0/SUCCESS)
  Process: 27431 ExecStartPost=/etc/init.d/dnsmasq systemd-start-resolvconf (code=exited, status=0/SUCCESS)
  Process: 27427 ExecStart=/etc/init.d/dnsmasq systemd-exec (code=exited, status=0/SUCCESS)
  Process: 27424 ExecStartPre=/usr/sbin/dnsmasq --test (code=exited, status=0/SUCCESS)
 Main PID: 27430 (dnsmasq)
   CGroup: /system.slice/dnsmasq.service
           └─27430 /usr/sbin/dnsmasq -x /var/run/dnsmasq/dnsmasq.pid -u www-data -r /var/run/dnsmasq/resolv.conf -7 /etc/dnsmasq.d,.dpkg-dist,.dpkg-old...

Jul 04 14:37:16 Odroid-C2 dnsmasq[27424]: dnsmasq: syntax check OK.
Jul 04 14:37:17 Odroid-C2 systemd[1]: Started dnsmasq - A lightweight DHCP and caching DNS server.

```

Fourdee, 2016-07-04:
@k-plan 
Great stuff :+1: 

Yep, ```/etc/dnsmasq.d/01-pihole.conf```` isn't required for the DietPi installation, mainly because we just use the normal configuration file ```/etc/dnsmasq.conf```. 

I need to update the PiHole installation to latest version, so i'll go through and check if we should add ```/etc/dnsmasq.d/01-pihole.conf```` to our installation. Either way, ```gravity.sh``` expects it to exist, so its probably best we add it.

Fourdee, 2016-07-04:
@k-plan 

We don't need the contents of the file for our installation. So resolved with:

```
echo -e "#nothing here" > /etc/dnsmasq.d/01-pihole.conf
```

```
root@DietPi:~# gravity.sh
:::
::: You are root.
::: Neutrino emissions detected...
:::
::: No custom adlist file detected, reading from default file... done!
:::
::: Existing pihole directory found
:::
::: Getting raw.githubusercontent.com list... List updated, transport successful                                                                 !
::: Getting adblock.gjtech.net list... No changes detected, transport skipped!
::: Getting mirror1.malwaredomains.com list... No changes detected, transport sk                                                                 ipped!
::: Getting sysctl.org list... No changes detected, transport skipped!
::: Getting zeustracker.abuse.ch list... No changes detected, transport skipped!
::: Getting s3.amazonaws.com list... No changes detected, transport skipped!
::: Getting s3.amazonaws.com list... No changes detected, transport skipped!
::: Getting hosts-file.net list... No changes detected, transport skipped!
::: Getting raw.githubusercontent.com list... List updated, transport successful                                                                 !
:::
::: Aggregating list of domains... done!
::: Formatting list of domains to remove comments.... done!
::: 126747 domains being pulled in by gravity...
::: Removing duplicate domains.... done!
::: 101948 unique domains trapped in the event horizon.
::: Formatting domains into a HOSTS file...
:::
::: Adding 9 adlist sources to the whitelist... done!
::: Running whitelist script to update HOSTS file.... 7 domains whitelisted!
::: Running blacklist script to update HOSTS file.... 0 domains blacklisted!
:::
::: Cleaning up un-needed files... done!
:::
::: Refresh lists in dnsmasq... done!
```

k-plan, 2016-07-05:
@Fourdee 

> I need to update the PiHole installation to latest version, so ...

hmm, don't we need it for some "_new features_" like:

> 
```
...
# If you want dnsmasq to read another file, as well as /etc/hosts, use
# this.
addn-hosts=/etc/pihole/gravity.lis
...
# This allows it to continue functioning without being blocked by syslog, and allows syslog to use dnsmasq for DNS queries without risking deadlock
log-async
...
# Add other name servers here, with domain specs if they are for
# non-public domains.
server=@DNS1@
server=@DNS2@
```

:grey_question:

Fourdee, 2016-07-05:
> # If you want dnsmasq to read another file, as well as /etc/hosts, use
# this.
addn-hosts=/etc/pihole/gravity.lis

We don't want it to read another file. /etc/pihole/gravity.list always contains the block list generated from gravity.sh.

> # Add other name servers here, with domain specs if they are for
# non-public domains.
server=@DNS1@
server=@DNS2@

Not needed, as per the documentation here: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=20#p174 we use the dnsmasq.conf file
> Optional: Change the DNS server that Pi-hole uses:
DietPi will setup Pi-hole to use external Google DNS servers. This is usually a good choice for the majority of users, but it can be changed if required.
Edit the following file /etc/dnsmasq.conf
Modify the entries server= as required (eg: server=123.456.78.9). 2 server entries maximum.
Once completed, restart Pi-hole with dietpi-services restart

> # This allows it to continue functioning without being blocked by syslog, and allows syslog to use dnsmasq for DNS queries without risking deadlock
log-async

Although rsyslog is not installed on DietPi unless you manually install it or change logging modes, i'll add this into the ```/etc/dnsmasq.conf```

Fourdee, 2016-07-05:
> ```addn-hosts=/etc/pihole/gravity.list```

We don't want it to read another file. /etc/pihole/gravity.list always contains the block list generated from gravity.sh.

> ```server=@DNS1@
server=@DNS2@```

Not needed, as per the documentation here: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=20#p174 we use the dnsmasq.conf file

> Optional: Change the DNS server that Pi-hole uses:
DietPi will setup Pi-hole to use external Google DNS servers. This is usually a good choice for the majority of users, but it can be changed if required.
Edit the following file /etc/dnsmasq.conf
Modify the entries server= as required (eg: server=123.456.78.9). 2 server entries maximum.
Once completed, restart Pi-hole with dietpi-services restart

> log-async

Although rsyslog is not installed on DietPi unless you manually install it or change logging modes, i'll add this into the ```/etc/dnsmasq.conf```

k-plan, 2016-07-05:
> Although rsyslog is not installed on DietPi unless you manually install it or change logging modes,

BTW: I forgot, rsyslog has a "lttle" Bug in Raspbian Jessie. It will spam the logfile with "_"rsyslogd-2007: action 'action 17' suspended, ..._"

Solution:

http://blog.dantup.com/2016/04/removing-rsyslog-spam-on-raspberry-pi-raspbian-jessie/

or 

[https://github.com/RPi-Distro/repo/issues/28](http://tiny.cc/3v7pcy)

Fourdee, 2016-07-14:
Note:
run this during rsyslog installation on RPi. Needs testing to verify pre and post results.
```
sed -i '/# The named pipe \/dev\/xconsole/,$d' /etc/rsyslog.conf
```

Fourdee, 2016-07-17:
@k-plan 

Looks like they fixed it in the repo.

full logging mode, or:
```
apt-get install rsyslog -y
reboot
```

RPi3 Jessie and Odroid C2 (thought may as well check official jessie repo):
```
root@DietPi:~# cat /var/log/syslog | grep 'action 17'
root@DietPi:~#
```

===

Forgot to mention, the new ```pihole``` binary commands are now in:
```
root@DietPi:~# pihole
::: Control all PiHole specific functions!
:::
::: Usage: pihole [options]
:::             Add -h after -w (whitelist), -b (blacklist), or -c (chronometer)  for more information on usage
:::
::: Options:
:::  -w, whitelist                      Whitelist domains
:::  -b, blacklist                      Blacklist domains
#:::  -d, debug                          Start a debugging session if having trouble
#:::  -f, flush                          Flush the pihole.log file
#:::  -u, updateDashboard                        Update the web dashboard manually
:::  -g, updateGravity                  Update the list of ad-serving domains
```
I've commented out the commands that should not be run on the DietPi installation, they may even break our installation as they expect the installation to be completed by ```curl -L https://install.pi-hole.net | bash```.

#### Why dont use the official installer?
- Unlike the automated installer (```curl -L https://install.pi-hole.net | bash```) which only supports lighttpd,, we support Lighttpd, nginx and apache2 webservers. 
- Our installation is slimmer and lighter than ```curl -L https://install.pi-hole.net | bash```. We dont even need to install ```git```.
- Our installation has Ramlog, unlike ```curl -L https://install.pi-hole.net | bash```.
- Our installation web interface has a sensible location ```/var/www/pihole```, unlike ```curl -L https://install.pi-hole.net | bash``` which is ```/var/www/html/admin```

Fourdee, 2016-07-17:
@k-plan 
I'll mark this as closed, if i missed anything please reopen :+1:

closed 2016-07-17 by Fourdee