MichaIng/DietPi #313: DietPi-Config | Add ability to set regulatory domain (WiFi)

k-plan, 2016-05-02:
@Fourdee 

we need a option, like in the new  [raspi-config](https://github.com/RPi-Distro/raspi-config/blob/master/debian/changelog) version, to get e.g. RPi 3 Wifi Channel 12+13 working.

[image]

Please see: [ "Test and set the correct regulatory domain"](http://dietpi.com/phpbb/viewtopic.php?p=1701#p1701)

```
root@RPi-3-test:~# iw reg get
country 00: DFS-UNSET
        (2402 - 2472 @ 40), (N/A, 20), (N/A)
        (2457 - 2482 @ 40), (N/A, 20), (N/A), NO-IR
        (2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, NO-IR
        (5170 - 5250 @ 80), (N/A, 20), (N/A), NO-IR
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, NO-IR
        (5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, NO-IR
        (5735 - 5835 @ 80), (N/A, 20), (N/A), NO-IR
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)

root@RPi-3-test:~# iw reg set DE

root@RPi-3-test:~# iw reg get
country DE: DFS-ETSI
        (2400 - 2483 @ 40), (N/A, 20), (N/A)
        (5150 - 5250 @ 80), (N/A, 20), (N/A), NO-OUTDOOR
        (5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS
        (5470 - 5725 @ 160), (N/A, 26), (0 ms), DFS
        (57000 - 66000 @ 2160), (N/A, 40), (N/A)

root@RPi-3-test:~# iwlist wlan0 channel
wlan0     13 channels in total; available frequencies :
          Channel 01 : 2.412 GHz
          Channel 02 : 2.417 GHz
          Channel 03 : 2.422 GHz
          Channel 04 : 2.427 GHz
          Channel 05 : 2.432 GHz
          Channel 06 : 2.437 GHz
          Channel 07 : 2.442 GHz
          Channel 08 : 2.447 GHz
          Channel 09 : 2.452 GHz
          Channel 10 : 2.457 GHz
          Channel 11 : 2.462 GHz
          Channel 12 : 2.467 GHz
          Channel 13 : 2.472 GHz
          Current Frequency:2.412 GHz (Channel 1)

root@RPi-3-test:~# sed -i "s/REGDOMAIN=/REGDOMAIN=DE/g" /etc/default/crda

root@RPi-3-test:~# cat /etc/default/crda
# Set REGDOMAIN to a ISO/IEC 3166-1 alpha2 country code so that iw(8) may set
# the initial regulatory domain setting for IEEE 802.11 devices which operate
# on this system.
#
# Governments assert the right to regulate usage of radio spectrum within
# their respective territories so make sure you select a ISO/IEC 3166-1 alpha2
# country code suitable for your location or you may infringe on local
# legislature. See `/usr/share/zoneinfo/zone.tab' for a table of timezone
# descriptions containing ISO/IEC 3166-1 alpha2 country codes.

REGDOMAIN=DE

```

----

Hint:
Tested on RPi 3, Odroid C1+ and C2. Working as well.
If setting regulatory domain to the correct value, and the same as the Access-Point (WiFi Router)
I will get a higher bandwidth and a more solid connection.

```
root@oDroid-C1-TEST2:~# iw reg set 00

root@oDroid-C1-TEST2:~# iw reg get
country 00: DFS-UNSET

root@oDroid-C1-TEST2:~# iperf -s
[  4]  0.0-90.1 sec   747 MBytes  69.5 Mbits/sec

```

```
root@oDroid-C1-TEST2:~# iw reg set DE

root@oDroid-C1-TEST2:~# iw reg get
country DE: DFS-ETSI

root@oDroid-C1-TEST2:~# iperf -s
[  4]  0.0-90.0 sec  1.02 GBytes  97.4 Mbits/sec
```

Fourdee, 2016-05-03:
@k-plan 
Excellent idea, will look into it :+1:

k-plan, 2016-05-03:
@Fourdee 

Verry nice! 😄  👍 

Some testing result from RPi 3:

```
root@RPi-3-test:~# uname -a
Linux RPi-3-test 4.1.19-v7+ #858 SMP Tue Mar 15 15:56:00 GMT 2016 armv7l GNU/Linux

-------------------

root@RPi-3-test:~# iw reg get
country 00: DFS-UNSET

root@RPi-3-test:~# iwlist wlan0 channel
wlan0     11 channels in total; available frequencies :

root@RPi-3-test:~# iperf -s 
. . .
[  4]  0.0-90.4 sec   109 MBytes  10.1 Mbits/sec

--------------------

root@RPi-3-test:~# iw reg get
country DE: DFS-ETSI

root@RPi-3-test:~# iwlist wlan0 channel
wlan0     11 channels in total; available frequencies :

root@RPi-3-test:~# iperf -s 
. . .
[  4]  0.0-90.2 sec   176 MBytes  16.4 Mbits/sec

=================

root@RPi-3-test:~# uname -a
Linux RPi-3-test 4.4.8-v7+ #881 SMP Sat Apr 30 12:16:50 BST 2016 armv7l GNU/Linux

-------------------

root@RPi-3-test:~# iw reg get
country 00: DFS-UNSET

root@RPi-3-test:~# iwlist wlan0 channel
wlan0     11 channels in total; available frequencies :

root@RPi-3-test:~# iperf -s 
. . .
[  4]  0.0-90.1 sec   130 MBytes  12.1 Mbits/sec

--------------------

root@RPi-3-test:~# iw reg get
country DE: DFS-ETSI

root@RPi-3-test:~# iwlist wlan0 channel
wlan0     13 channels in total; available frequencies :

root@RPi-3-test:~# iperf -s 
. . .
[  4]  0.0-90.3 sec   197 MBytes  18.3 Mbits/sec

-------------------

```

Fourdee, 2016-05-05:
Country codes: https://en.wikipedia.org/wiki/ISO_3166-1_alpha-2

Current country code
```
iw reg get | grep -m1 'country' | awk '{print $2}' | tr -d ':'
```

Menu select common codes

k-plan, 2016-05-05:
@Fourdee 

> Location:
Network Options: Adapters > Wifi ? (menu too big already?, convert to scrollable menu?)
Network Options: Adapters > WiFi Country?

Difficult question. Both have pros and cons.
If scrollable, someone can miss it.
If it will be a new list point, it will blow up the list. May be a issue, if you must have to put in new options later.

> Merge Menu_WifiHotspot into Menu_NetworkAdapters_Wifi() , create whiptail array to only show required menu options, depending on mode. Allows for #311. Possible issue with top stats being too many lines,

Hmm ... it's  the same here.

---

> Menu select common codes
-GB
-RU
-US
+manual entry option
etc

Hmm .. for 2.4 GHz (802.11b/g/n) it will be better:

Menu select common codes
-GB 
-US
-JP
+manual entry option
etc.
 
for 5 GHz (802.11a/h/j/n/ac) it will be better:

Menu select common codes
-GB
-US
-CN
+manual entry option
etc.

[WLAN Frequency](https://de.wikipedia.org/wiki/Wireless_Local_Area_Network#Frequenzen_und_Kan.C3.A4le) 

Can you please build a version with all possible option and check in via testing branch, so we can have a look?

Fourdee, 2016-05-07:
@k-plan 
Thanks for the extra info :+1: 

Started on this today. I havn't done any real "programming" for a while, so little rusty lol. Will let you know when its available for testing.

[image]

k-plan, 2016-05-07:
@Fourdee 

your screenshot is looking very handsome.  
2.4 GHz (802.11b/g/n) is a good choice to start. 
(_RPi 3 build-in WLAN module_)

Fourdee, 2016-05-08:
CRDA was not applying our saved setting during reboots, reverts to 00:
```
# - Apply to system configs, for reboots
echo -e "REGDOMAIN=$wifi_country_code_target" > /etc/default/crda
echo -e "options cfg80211 ieee80211_regdom=$wifi_country_code_target" > /etc/modprobe.d/cfg80211.conf
```
Fixed :)

Fourdee, 2016-05-08:
@k-plan 
Uploaded to testing branch. 

For now, I've just added the ability to change WiFi country code when running in non-AP mode. Also been playing around with the menu code (select "change mode") to resolve the height issue.
Country Code is available in ```Network Options: Adapters > Wifi```

Commit: https://github.com/Fourdee/DietPi/commit/49aa00bb3b272e10ec723a36cff9632dd52da332

I'll work on the AP mode menu next.

Fourdee, 2016-05-08:
As far as I can tell, country code ```iw reg set``` has no effect on Hostapd. 
- ```iw reg set US```
- Set hostapd channel to 13
- Restart hostapd and networking services
- Hotspot still available.

Fourdee, 2016-05-08:
Hostapd:

I can not get 2.4ghz N (150mbit) to work on hostapd with ```driver=rtl871xdrv``` and :
```
Bus 002 Device 002: ID 7392:7811 Edimax Technology Co., Ltd EW-7811Un 802.11n Wireless Adapter [Realtek RTL8188CUS]
```

Tried using:
```
wwm_enabled=1
ieee80211n=1
```
Either option prevents hostapd service from starting, no error codes/info from ```systemctl status hostapd -l```

Tried with RPi 3 which uses ```driver=nl80211```
```
interface=wlan0
driver=nl80211
ssid=DietPi-HotSpot
hw_mode=g
channel=3
macaddr_acl=0
auth_algs=1
ssid=DietPi-HotSpot
wpa=2
wpa_passphrase=dietpihotspot
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
#wwm_enabled=1
ieee80211n=1
```

Seems to work with N (72mbit max on this device):
[image]

k-plan, 2016-05-08:
@Fourdee 

wow, you are to quick for me. 😄  
Sorry, have to catch up.

Staring now with `gitbranch=testing`

Fourdee, 2016-05-08:
@k-plan 
I broke the RPi 3 hotspot. Turns out the internal BCM does not like any driver unless its disabled ```#driver=none```.  I've sent a new commit: https://github.com/Fourdee/DietPi/commit/6a3ad4b466a5ed500242473f2ea9174fb5ed74d7 you just need to grab ```dietpi-software``` if you are going to test RPi 3.

k-plan, 2016-05-08:
@Fourdee 

> CRDA was not applying our saved setting during reboots, reverts to 00:
Fixed :)

Hmm .. doesn't work here:

- set Country-Code manual per `dietpi-config` and ` Apply       Save Changes (Restart Networking) `

[image]

```
root@oDroid-C1-TEST:~# iw reg get
country DE: DFS-ETSI
        (2400 - 2483 @ 40), (N/A, 20), (N/A)
        (5150 - 5250 @ 80), (N/A, 20), (N/A), NO-OUTDOOR
        (5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS
        (5470 - 5725 @ 160), (N/A, 26), (0 ms), DFS
        (57000 - 66000 @ 2160), (N/A, 40), (N/A)

root@oDroid-C1-TEST:~#reboot

root@oDroid-C1-TEST:~# iw reg get
country 00: DFS-UNSET
        (2402 - 2472 @ 40), (6, 20), (N/A)
        (2457 - 2482 @ 40), (6, 20), (N/A), PASSIVE-SCAN
        (2474 - 2494 @ 20), (6, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 160), (6, 20), (N/A), PASSIVE-SCAN
        (5250 - 5330 @ 160), (6, 20), (0 ms), DFS, PASSIVE-SCAN
        (5490 - 5730 @ 160), (6, 20), (0 ms), DFS, PASSIVE-SCAN

root@oDroid-C1-TEST:~# cat /etc/default/crda
REGDOMAIN=DE

root@oDroid-C1-TEST:~# cat /etc/modprobe.d/cfg80211.conf
options cfg80211 ieee80211_regdom=DE
```

[image]

Fourdee, 2016-05-08:
@k-plan 
>
```
root@oDroid-C1-TEST:~#reboot

root@oDroid-C1-TEST:~# iw reg get
```

Strange, run it again:
```
 iw reg get
```

Fourdee, 2016-05-08:
Odroid C2 ```dmesg```. It is being applied:
```
[    7.992802] cfg80211: Calling CRDA for country: GB
[    8.005850] cfg80211: Regulatory domain changed to country: GB
[    8.006045] cfg80211:  DFS Master region: ETSI
[    8.010317] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp)
[    8.012946] cfg80211:   (2402000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm)
[    8.020294] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz), (N/A, 2000 mBm)
[    8.027643] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz), (N/A, 2000 mBm)
[    8.034937] cfg80211:   (5490000 KHz - 5710000 KHz @ 160000 KHz), (N/A, 2700 mBm)
[    8.042425] cfg80211:   (57000000 KHz - 66000000 KHz @ 2160000 KHz), (N/A, 4000 mBm)
[    8.055556] ttyS0 use xtal 24000000 change 115200 to 115200
[    8.077616] ttyS0 use xtal 24000000 change 115200 to 115200
[    8.099964] ttyS0 use xtal 24000000 change 115200 to 115200
[    8.119537] ttyS0 use xtal 24000000 change 115200 to 115200
[    8.137865] ttyS0 use xtal 24000000 change 115200 to 115200
[    8.159076] ttyS0 use xtal 24000000 change 115200 to 115200
[    8.178856] ttyS0 use xtal 24000000 change 115200 to 115200
[    8.199226] ttyS0 use xtal 24000000 change 115200 to 115200
[    8.219172] ttyS0 use xtal 24000000 change 115200 to 115200
[    8.262049] ttyS0 use xtal 24000000 change 115200 to 115200
[    8.294783] ttyS0 use xtal 24000000 change 115200 to 115200
[    8.324926] ttyS0 use xtal 24000000 change 115200 to 115200
[    9.161493] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[    9.219858] ttyS0 use xtal 24000000 change 115200 to 115200
[    9.238812] ttyS0 use xtal 24000000 change 115200 to 115200
[    9.259403] ttyS0 use xtal 24000000 change 115200 to 115200
[    9.278451] ttyS0 use xtal 24000000 change 115200 to 115200
[    9.299541] ttyS0 use xtal 24000000 change 115200 to 115200
[    9.318048] ttyS0 use xtal 24000000 change 115200 to 115200
[    9.339485] ttyS0 use xtal 24000000 change 115200 to 115200
[    9.358101] ttyS0 use xtal 24000000 change 115200 to 115200
[    9.517822] ttyS0 use xtal 24000000 change 115200 to 115200
[    9.538756] ttyS0 use xtal 24000000 change 115200 to 115200
[    9.558328] ttyS0 use xtal 24000000 change 115200 to 115200
[    9.578252] ttyS0 use xtal 24000000 change 115200 to 115200
[    9.615743] ttyS0 use xtal 24000000 change 115200 to 115200
[    9.639946] ttyS0 use xtal 24000000 change 115200 to 115200
[   10.429693] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   14.417286] libphy: stmmac-0:00 - Link is Up - 1000/Full
[   14.418071] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
```

k-plan, 2016-05-08:
```
root@oDroid-C1-TEST:~# reboot

root@oDroid-C1-TEST:~# iw reg get
country 00: DFS-UNSET
        (2402 - 2472 @ 40), (6, 20), (N/A)
        (2457 - 2482 @ 40), (6, 20), (N/A), PASSIVE-SCAN
        (2474 - 2494 @ 20), (6, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 160), (6, 20), (N/A), PASSIVE-SCAN
        (5250 - 5330 @ 160), (6, 20), (0 ms), DFS, PASSIVE-SCAN
        (5490 - 5730 @ 160), (6, 20), (0 ms), DFS, PASSIVE-SCAN
        
root@oDroid-C1-TEST:~# dmesg 
 . . .      
 [   17.081693] usbcore: registered new interface driver rtl8192cu

root@oDroid-C1-TEST:~# cat /var/log/syslog | grep cfg 
May  8 19:50:51 oDroid-C1-TEST kernel: [  440.785811@0] cfg80211: Calling CRDA for country: DE
May  8 19:50:51 oDroid-C1-TEST kernel: [  440.798204@3] cfg80211: Regulatory domain changed to country: DE
May  8 19:50:51 oDroid-C1-TEST kernel: [  440.798292@3] cfg80211:  DFS Master region: ETSI
May  8 19:50:51 oDroid-C1-TEST kernel: [  440.798340@3] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cactime)
May  8 19:50:51 oDroid-C1-TEST kernel: [  440.798440@3] cfg80211:   (2400000 KHz - 2483500 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
May  8 19:50:51 oDroid-C1-TEST kernel: [  440.798585@3] cfg80211:   (5150000 KHz - 5250000 KHz @ 80000 KHz, 200000 KHz AUTO), (N/A, 2000 mBm), N/A)
May  8 19:50:51 oDroid-C1-TEST kernel: [  440.798705@3] cfg80211:   (5250000 KHz - 5350000 KHz @ 80000 KHz, 200000 KHz AUTO), (N/A, 2000 mBm), 0 s)
May  8 19:50:51 oDroid-C1-TEST kernel: [  440.798818@3] cfg80211:   (5470000 KHz - 5725000 KHz @ 160000 KHz), (N/A, 2698 mBm), (0 s)
May  8 19:50:51 oDroid-C1-TEST kernel: [  440.798914@3] cfg80211:   (57000000 KHz - 66000000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)
May  8 19:58:50 oDroid-C1-TEST kernel: [    0.177597@0] cfg80211: Calling CRDA to update world regulatory domain
```

But I have to say, my C1+ is running DietPi from USB device at the moment and have installed "SoftEthernet"

I will make a fresh install on a new sd-card ...

k-plan, 2016-05-08:
@Fourdee 

starting with a fresh sd-card installation:

```
root@oDroid-C1-TEST:~# lsusb
Bus 001 Device 003: ID 7392:7822 Edimax Technology Co., Ltd
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

root@oDroid-C1-TEST:~# dmesg
 . . .
[    0.177589] cfg80211: Calling CRDA to update world regulatory domain
[    7.030012] usbcore: registered new interface driver rtl8192cu

root@oDroid-C1-TEST:~# cat /var/log/syslog | grep cfg
May  8 20:24:46 oDroid-C1-TEST kernel: [    0.177589@0] cfg80211: Calling CRDA to update world regulatory domain

root@oDroid-C1-TEST:~# iw reg get
country 00: DFS-UNSET
        (2402 - 2472 @ 40), (6, 20), (N/A)
        (2457 - 2482 @ 40), (6, 20), (N/A), PASSIVE-SCAN
        (2474 - 2494 @ 20), (6, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 160), (6, 20), (N/A), PASSIVE-SCAN
        (5250 - 5330 @ 160), (6, 20), (0 ms), DFS, PASSIVE-SCAN
        (5490 - 5730 @ 160), (6, 20), (0 ms), DFS, PASSIVE-SCAN

root@oDroid-C1-TEST:~# cat /etc/default/crda
# Set REGDOMAIN to a ISO/IEC 3166-1 alpha2 country code so that iw(8) may set
# the initial regulatory domain setting for IEEE 802.11 devices which operate
# on this system.
#
# Governments assert the right to regulate usage of radio spectrum within
# their respective territories so make sure you select a ISO/IEC 3166-1 alpha2
# country code suitable for your location or you may infringe on local
# legislature. See `/usr/share/zoneinfo/zone.tab' for a table of timezone
# descriptions containing ISO/IEC 3166-1 alpha2 country codes.

REGDOMAIN=

root@oDroid-C1-TEST:~# cat /etc/modprobe.d/cfg80211.conf
cat: /etc/modprobe.d/cfg80211.conf: No such file or directory

```

- now set Country per `dietpi-config` to `GB` and save

```
root@oDroid-C1-TEST:~# iw reg get
country GB: DFS-ETSI
        (2402 - 2482 @ 40), (N/A, 20), (N/A)
        (5170 - 5250 @ 80), (N/A, 20), (N/A)
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS
        (5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
        (57000 - 66000 @ 2160), (N/A, 40), (N/A)

root@oDroid-C1-TEST:~# cat /etc/default/crda
REGDOMAIN=GB

root@oDroid-C1-TEST:~# cat /etc/modprobe.d/cfg80211.conf
options cfg80211 ieee80211_regdom=GB

root@oDroid-C1-TEST:~# reboot

```

```
root@oDroid-C1-TEST:~# iw reg get
country 00: DFS-UNSET
[... 20 lines cut ...]
May  8 20:43:43 oDroid-C1-TEST kernel: [ 1140.424304@3] cfg80211:  DFS Master region: unset
May  8 20:43:43 oDroid-C1-TEST kernel: [ 1140.424357@3] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
May  8 20:43:43 oDroid-C1-TEST kernel: [ 1140.424450@3] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
May  8 20:43:43 oDroid-C1-TEST kernel: [ 1140.424533@3] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
May  8 20:43:43 oDroid-C1-TEST kernel: [ 1140.424614@3] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
May  8 20:43:43 oDroid-C1-TEST kernel: [ 1140.424691@3] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
May  8 20:43:43 oDroid-C1-TEST kernel: [ 1140.424785@3] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
May  8 20:43:43 oDroid-C1-TEST kernel: [ 1140.424878@3] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
May  8 20:43:43 oDroid-C1-TEST kernel: [ 1140.424954@3] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
May  8 20:43:43 oDroid-C1-TEST kernel: [ 1140.425040@3] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
May  8 20:43:43 oDroid-C1-TEST kernel: [ 1140.425370@3] cfg80211: Calling CRDA for country: GB
May  8 20:43:43 oDroid-C1-TEST kernel: [ 1140.437968@2] cfg80211: Regulatory domain changed to country: GB
May  8 20:43:43 oDroid-C1-TEST kernel: [ 1140.438046@2] cfg80211:  DFS Master region: ETSI
May  8 20:43:43 oDroid-C1-TEST kernel: [ 1140.438092@2] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
May  8 20:43:43 oDroid-C1-TEST kernel: [ 1140.438185@2] cfg80211:   (2402000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
May  8 20:43:43 oDroid-C1-TEST kernel: [ 1140.440388@2] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
May  8 20:43:43 oDroid-C1-TEST kernel: [ 1140.442580@2] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
May  8 20:43:43 oDroid-C1-TEST kernel: [ 1140.444730@2] cfg80211:   (5490000 KHz - 5710000 KHz @ 160000 KHz), (N/A, 2700 mBm), (0 s)
May  8 20:43:43 oDroid-C1-TEST kernel: [ 1140.446875@2] cfg80211:   (57000000 KHz - 66000000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)

May  8 20:46:08 oDroid-C1-TEST kernel: [    0.177442@0] cfg80211: Calling CRDA to update world regulatory domain

```

Will test as well 'DE' but I think it's the same.

k-plan, 2016-05-08:
@Fourdee 

is it possible we missed something (packet, config, db ..) ?

https://wireless.wiki.kernel.org/en/developers/regulatory/crda

https://wireless.wiki.kernel.org/en/developers/Regulatory

xenfomation, 2016-05-08:
My brain has been squished from LLMP, BUT I wanted to ask both of you --

Based on the dietpi-software and configure options:

- Are we caching @k-plan's setting or is cleaned up from /tmp/.... ?
- If we aren't caching this, would a work using /etc/rc.local + boot scripts for services to SET this be too... hackish?
- As for N networking, I've seen similar issues as you found @Fourdee -- good catch and reminder!
- I run N and can test this tomorrow morning as I have several routers and can wait till the family is snuggled in bed to confirm the 5Ghz range, etc

Thanks again.  I know I have not comitted much code... but getting back into the swing of real dev/commit work is something I sorely miss!

Cheers to you both!

k-plan, 2016-05-08:
Testing some tips I have read:

```
root@oDroid-C1-TEST:~# cat /etc/udev/rules.d/regulatory.rules
# WLAN-Regionseinstellungen festlegen

 KERNEL=="regulatory*", ACTION=="change", SUBSYSTEM=="platform", RUN+="/sbin/iw reg set DE"

## KERNEL=="regulatory*", ACTION=="change", SUBSYSTEM=="platform", RUN+="/sbin/crda"
```

But without success ...

xenfomation, 2016-05-08:
KUDOS for getting on the right track!!!

[https://en.wikipedia.org/wiki/ISO_3166-1_alpha-2](https://en.wikipedia.org/wiki/ISO_3166-1_alpha-2)

While @k-plan is at the core of it, I was looking into using "iw reg set DE" before the physical interface tries to come alive...

Nice work and I will keep digging into this as it should take no longer than 15-30 minutes.

xenfomation, 2016-05-08:
Despite the infinite claims for Debian/Ubuntu, I regret to report that /etc/default/crda does not promote regional codes as it should.

I was able to repeat @k-plan's issue of set manually, reboot, run iw reg get only to see "00: DFS-UNSET"

Now, as a cludge right now:

vi /etc/rc.local

```
#!/bin/sh -e
iw reg set DE
/DietPi/dietpi/dietpi-services start
/DietPi/dietpi/dietpi-banner 0
echo " Default Login:\n Username = root\n Password = dietpi\n"
exit 0
```
On reboot, I obviously retain this good config @k-plan needs:

```
iw reg get
country DE: DFS-ETSI
        (2400 - 2483 @ 40), (N/A, 20), (N/A)
        (5150 - 5250 @ 80), (N/A, 20), (N/A), NO-OUTDOOR
        (5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS
        (5470 - 5725 @ 160), (N/A, 26), (0 ms), DFS
        (57000 - 66000 @ 2160), (N/A, 40), (N/A)
```

So, looking to move this type of param-set/config option into native DietPi territory -- will post shortly.

xenfomation, 2016-05-08:
Alright --

1.  Removed hard-coded region from /etc/rc.local
2.  Added variable to /DietPi/dietpi.txt : Wifi_REGION=<ISO CODE, such as DE>

```
#Enter your Wifi Region Code (ISO_3166-1_alpha-2)
#This needs to tie into dietpi-config if agreed upon with possible links to explain ISO codes for new DietAdmins
#Default should be 00 or NULL value for scalar
Wifi_REGION=DE
```
3.  Within /DietPi/dietpi/ is our magnificent "boot" config that is called upon and I am testing the parsing of dietpi.txt to ENSURE iw retains what is set:

```
                local Wifi_SSID=$(cat /DietPi/dietpi.txt | grep -m1 '^Wifi_SSID' | sed 's/.*=//')
                local Wifi_KEY=$(cat /DietPi/dietpi.txt | grep -m1 '^Wifi_KEY' | sed 's/.*=//')
                local Wifi_REGION=$(cat /DietPi/dietpi.txt | grep -m1 '^Wifi_REGION' | sed 's/.*=//')
...

4.  CRDA expects a COUNTRY environment value to be set, so that may be why /etc/default/crda's config is being skipped

I am going to need 20 minutes to bake in v117 on a physical pie to test my code.

Thanks!

Fourdee, 2016-05-09:
@xenfomation 

Based on your ideas :+1: , I'am going to do the following:
- ```/DietPi/dietpi.txt``` | Add global var for ```WiFi_CountryCode=```. Here: https://github.com/Fourdee/DietPi/blob/master/dietpi.txt#L18
- ```/DietPi/dietpi/boot``` | Add ```iw reg set $WiFi_CountryCode```

This will allow users to declare their country code prior to first run. Also ensures country code will always be applied during boot.
I've tested ```iw reg set``` in boot script, gets applied.

Fourdee, 2016-05-09:
Decided to make a function script for applying/changing WiFi country code, so we can use it in multiple scripts.

Following applies Country code to system. with value stored in ```dietpi.txt``` if set ```wifi_country_code=``` :
```
/DietPi/dietpi/func/dietpi-set_wificountrycode
```

Following changes value in dietpi.txt and applys new country code to system
```
/DietPi/dietpi/func/dietpi-set_wificountrycode GB
```

this function is called during boot, and when changing country code in ```dietpi-config```

commit: https://github.com/Fourdee/DietPi/commit/f499ba6ce2b0c645ee109ca171f10d8e5ed683e2

Fourdee, 2016-05-09:
Tests:
```
root@DietPi:~# /DietPi/dietpi/func/dietpi-set_wificountrycode US && iw reg get && cat /DietPi/dietpi.txt | grep country_code && cat /etc/default/crda && cat /etc/modprobe.d/cfg80211.conf
country US: DFS-FCC
        (2402 - 2472 @ 40), (N/A, 30), (N/A)
        (5170 - 5250 @ 80), (N/A, 17), (N/A)
        (5250 - 5330 @ 80), (N/A, 23), (N/A), DFS
        (5735 - 5835 @ 80), (N/A, 30), (N/A)
        (57240 - 63720 @ 2160), (N/A, 40), (N/A)
wifi_country_code=US
REGDOMAIN=US
options cfg80211 ieee80211_regdom=US
root@DietPi:~# /DietPi/dietpi/func/dietpi-set_wificountrycode GB && iw reg get && cat /DietPi/dietpi.txt | grep country_code && cat /etc/default/crda && cat /etc/modprobe.d/cfg80211.conf
country GB: DFS-ETSI
        (2402 - 2482 @ 40), (N/A, 20), (N/A)
        (5170 - 5250 @ 80), (N/A, 20), (N/A)
        (5250 - 5330 @ 80), (N/A, 20), (N/A), DFS
        (5490 - 5710 @ 160), (N/A, 27), (N/A), DFS
        (57000 - 66000 @ 2160), (N/A, 40), (N/A)
wifi_country_code=GB
REGDOMAIN=GB
options cfg80211 ieee80211_regdom=GB
```

xenfomation, 2016-05-09:
@Fourdee -- slick!  Especially with it being scriptable.  Nicely done and for miscellaneous trivia, if they don't know their country code.... 00 works as a default ;-)  Kinda breaks their own ISEC/ISO standards, but I am glad you knocked this out!

Fourdee, 2016-05-09:
@xenfomation 
Yep, if only the system called ```/etc/default/crda``` during boot and used the value set there. Maybe there's something we missed.

Either way, this function script was needed so we can call it across multiple other scripts, based on a global country code value set in ```dietpi.txt```

xenfomation, 2016-05-09:
Indeed!  Sad thing about it is... I haven't found anything to even imply it was missed by anyone.  My reasoning for this is many have complained about this and it coincides with the whole "systemd" nonsense (IMHO).

I should have commits to testing for #295 and #246 in a few hours for your review.  This means I can get back to LLMP, but make a double-random sweep for CRDA.  I mean, the three of us clearly see dmesg invoking CRDA, so yes: _What are you not doing or doing ad-hoc, Mr. Debian?_

Cheers!

k-plan, 2016-05-09:
@Fourdee 

- tested your new version with `/DietPi/dietpi/boot` for changing WiFi country code with a [Edimax EW-7612UAn V2](http://www.edimax.com/edimax/merchandise/merchandise_detail/data/edimax/global/wireless_adapters_n300/ew-7612uan_v2/) (2.4 GHZ - 802.11b/g/n - 300Mbps)

Set via `dietpi-config` country code `00` -> `DE` (readout) ==>> `reboot` = `DE` (readout) setting  -> `CN` (readout) ==> `reboot` = `CN` (readout) setting -> `US` (readout) ==>> `reboot` = `US` (readout)
(I spare the log output  here)

It works, nice. 👍   

- Test with [Alfa Networks AWUS036EAC](http://www.alfa.com.tw/products_show.php?pc=137&ps=219) 2.4 / 5 GHz 802.11ac AC1200 Dual-Band USB 3.0 Adapter (AC network 867Mbps and N network  up to 300Mbps)

Here are some outputs:

```
May  9 20:14:07 oDroid-C1-TEST kernel: [  687.012837@3] RTL871X: VID = 0x0BDA, PID = 0x8812
May  9 20:14:07 oDroid-C1-TEST kernel: [  687.015625@3] RTL871X: Customer ID: 0x00, SubCustomer ID: 0xCD
May  9 20:14:07 oDroid-C1-TEST kernel: [  687.018707@3] RTL871X: hal_ReadMACAddress_8812AU MAC Address from EFUSE = 00:c0:ca:
May  9 20:14:07 oDroid-C1-TEST kernel: [  687.021876@3] RTL871X: ==>mpt_GetChnlGroup8812A in 5G, but Channel 163 in Group not found
May  9 20:14:07 oDroid-C1-TEST kernel: [  687.024869@3] RTL871X: ==>mpt_GetChnlGroup8812A in 5G, but Channel 163 in Group not found
May  9 20:14:07 oDroid-C1-TEST kernel: [  687.027777@3] RTL871X: ==>mpt_GetChnlGroup8812A in 5G, but Channel 163 in Group not found
May  9 20:14:07 oDroid-C1-TEST kernel: [  687.031541@3] RTL871X: ==>mpt_GetChnlGroup8812A in 5G, but Channel 163 in Group not found
May  9 20:14:07 oDroid-C1-TEST kernel: [  687.034350@3] RTL871X: EEPROMRegulatory = 0x0 TxPwrCalibrateRate=0x0
May  9 20:14:07 oDroid-C1-TEST kernel: [  687.036970@3] RTL871X: Board Type: 0x 0
May  9 20:14:07 oDroid-C1-TEST kernel: [  687.039595@3] RTL871X: mlmepriv.ChannelPlan = 0x7f
May  9 20:14:07 oDroid-C1-TEST kernel: [  687.042012@3] RTL871X: CrystalCap: 0x20
May  9 20:14:07 oDroid-C1-TEST kernel: [  687.043690@3] RTL871X: ThermalMeter = 0x23
May  9 20:14:07 oDroid-C1-TEST kernel: [  687.045333@3] RTL871X: SWAS: bHwAntDiv = 0, TRxAntDivType = 3
May  9 20:14:07 oDroid-C1-TEST kernel: [  687.046961@3] RTL871X: pHalData->PAType_2G is 0x3, pHalData->ExternalPA_2G = 0
May  9 20:14:07 oDroid-C1-TEST kernel: [  687.048668@3] RTL871X: pHalData->PAType_5G is 0x3, pHalData->ExternalPA_5G = 1
May  9 20:14:07 oDroid-C1-TEST kernel: [  687.050516@3] RTL871X: pHalData->LNAType_2G is 0x88, pHalData->ExternalLNA_2G = 1
May  9 20:14:07 oDroid-C1-TEST kernel: [  687.052236@3] RTL871X: pHalData->LNAType_5G is 0x88, pHalData->ExternalLNA_5G = 1
May  9 20:14:07 oDroid-C1-TEST kernel: [  687.053932@3] RTL871X: pHalData->TypeGPA = 0x0
May  9 20:14:07 oDroid-C1-TEST kernel: [  687.055590@3] RTL871X: pHalData->TypeAPA = 0x0
May  9 20:14:07 oDroid-C1-TEST kernel: [  687.057232@3] RTL871X: pHalData->TypeGLNA = 0x0
May  9 20:14:07 oDroid-C1-TEST kernel: [  687.058946@3] RTL871X: pHalData->TypeALNA = 0x0
May  9 20:14:07 oDroid-C1-TEST kernel: [  687.060755@3] RTL871X: RFE Type: 0x 3
May  9 20:14:07 oDroid-C1-TEST kernel: [  687.062404@3] RTL871X: PID= 0x8812, VID=  bda
May  9 20:14:07 oDroid-C1-TEST kernel: [  687.064038@3] RTL871X: PID= 0x8812, VID=  bda
May  9 20:14:07 oDroid-C1-TEST kernel: [  687.065646@3] RTL871X: Customer ID: 0x 0
May  9 20:14:07 oDroid-C1-TEST kernel: [  687.071109@3] RTL871X: ReadAdapterInfo8812AU <====
May  9 20:14:07 oDroid-C1-TEST kernel: [  687.072891@3] RTL871X: init_channel_set ChannelPlan ID 7f Chan num:38
May  9 20:14:07 oDroid-C1-TEST kernel: [  687.074993@3] RTL871X: rtw_register_early_suspend
May  9 20:14:07 oDroid-C1-TEST kernel: [  687.076668@3] RTL871X: rtw_macaddr_cfg MAC Address  = 00:c0:ca:
May  9 20:14:07 oDroid-C1-TEST kernel: [  687.078358@3] RTL871X: bDriverStopped:1, bSurpriseRemoved:0, bup:0, hw_init_completed:0
May  9 20:14:07 oDroid-C1-TEST kernel: [  687.080148@3] RTL871X: rtw_ndev_init(wlan0)
May  9 20:14:07 oDroid-C1-TEST kernel: [  687.083046@3] RTL871X: _rtw_drv_register_netdev, MAC Address (if1) = 00:c0:ca:
```

```
root@oDroid-C1-TEST:~# lsusb
Bus 001 Device 008: ID 0bda:8812 Realtek Semiconductor Corp.
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
root@oDroid-C1-TEST:~# iw reg get && cat /DietPi/dietpi.txt | grep country_code && cat /etc/default/crda && cat /etc/modprobe.d/cfg80211.conf
country DE: DFS-ETSI
        (2400 - 2483 @ 40), (N/A, 20), (N/A)
        (5150 - 5250 @ 80), (N/A, 20), (N/A), NO-OUTDOOR
        (5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS
        (5470 - 5725 @ 160), (N/A, 26), (0 ms), DFS
        (57000 - 66000 @ 2160), (N/A, 40), (N/A)
wifi_country_code=DE
REGDOMAIN=DE
options cfg80211 ieee80211_regdom=DE

```

[... 61 lines cut ...]
[   19.712939] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
[   19.715088] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
[   19.717534] cfg80211: Calling CRDA for country: DE
[   19.739466] cfg80211: Regulatory domain changed to country: DE
[   19.741744] cfg80211:  DFS Master region: ETSI
[   19.741803] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[   19.746577] cfg80211:   (2400000 KHz - 2483500 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[... cut ...]
Get it on work is the first time a little tricky. The 2.4 GHz USB adapter was `wlan0` you have to plug 5 GHz USB adapter in / out  and have to enable-disabel-enable WiFi network via dietpi-config because it is now `wlan1`.

But setting, reboot, switching country code works fine as well!  👍 

Here some fun. I hope, the bottleneck is USB 2.0 Port of the C1+ : 

```
root@oDroid-C1-TEST:~# iperf -s -i 12
[ ID] Interval       Transfer     Bandwidth
[  4]  0.0-90.0 sec  1.20 GBytes   114 Mbits/sec
```
😄

Fourdee, 2016-05-09:
@k-plan 
Thanking you for in-depth testing and results as always :+1: 

> And the 802.11ac AC1200 Dual-Band USB 3.0 Adapter is really chatty in dmesg !

Holy cow lol. Looks like its reporting everytime it switches from powersaving modes.

I'll mark this as closed. I will take a look into https://github.com/Fourdee/DietPi/issues/311 next and see what we can do.

closed 2016-05-09 by Fourdee

k-plan, 2016-05-16:
@Fourdee 

just a quick late hint, work as well:

 ```
cat /etc/network/interfaces
#/etc/network/interfaces
#Please use DietPi-Config to modify network settings.
 . . 
# Wifi
allow-hotplug wlan0
iface wlan0 inet dhcp
metric 1
 . . .
pre-up iw dev wlan0 set power_save off && iw reg set DE
post-down iw dev wlan0 set power_save on

```

`rsyslogd-2007: action 'action 17' suspended, next retry is Mon May  9 20:49:42 2016 [try http://www.rsyslog.com/e/2007 ]`

https://github.com/RPi-Distro/repo/issues/28#issuecomment-216640318

Fourdee, 2016-05-17:
@k-plan 

> 
```
pre-up iw dev wlan0 set power_save off
post-down iw dev wlan0 set power_save on
```

Nice find. If this works with all adapters, would be a better solution than the current:
```
echo -e "options 8192cu rtw_power_mgnt=0" > /etc/modprobe.d/8192cu.conf
echo -e "options 8188eu rtw_power_mgnt=0" > /etc/modprobe.d/8188eu.conf
```

I'll have to give it a test with a few devices.