MichaIng/DietPi #316: Phpmyadmin not working with Lighttpd Stack

Fourdee, 2016-05-06:
> http://dietpi.com/phpbb/viewtopic.php?f=11&t=401
94cc6cbb-2bce-42e5-8170-fe835eb3cb55-0

Fourdee, 2016-05-06:
> note: phpMyAdmin is not triggering webserver installation.

Fourdee, 2016-05-06:
Installed:
- LLMP
- PHPMYADMIN

Unable to replicate issue:
[image]

Go through bug report and check for other causes.

Fourdee, 2016-05-06:
Select PHPMYADMIN , install, reboot, select LLMP = fine

Select LLMP + PHPMYADMIN =
```
Run /etc/init.d/lighttpd force-reload to enable changes
dpkg: error processing package phpmyadmin (--configure):
 subprocess installed post-installation script returned error exit status 2
Processing triggers for libc-bin (2.19-18+deb8u4) ...
Processing triggers for php5-fpm (5.6.20+dfsg-0+deb8u1) ...
Errors were encountered while processing:
 phpmyadmin
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

```
debconf-set-selections <<< "phpmyadminphpmyadmin/dbconfig-install boolean true"
warning: Unknown type true, skipping line 1
```

xenfomation, 2016-05-07:
Silly question -- is this a DHCP-based install then proceeding to install LLMP + PHPMYADMIN?

xenfomation, 2016-05-07:
Hrm -- from the command line I found that there is a sort of "cart before horse" issue with MySQL deps and dbconf.  Will send logs and will start from build 117 again as this is quite important.

xenfomation, 2016-05-07:
Yeah, I know -- a whole lot of gibberish on my part, but I wanted to emphasize the soon-to-be-deprecated key_buffer_size to be removed from future releases for mysqld 5.5.xx and beyond

Fourdee, 2016-05-07:
>  is this a DHCP-based install then proceeding to install LLMP + PHPMYADMIN?

Yep, I used DHCP during my testing.

> from the command line I found that there is a sort of "cart before horse" issue with MySQL deps and dbconf. Will send logs and will start from build 117 again as this is quite important.

Sounds good :+1:

xenfomation, 2016-05-07:
I shot a bug report and am now slicking a PI as well as my sandbox (well, reverting to a snapshot).  This is a sane bug.... at least for now!

xenfomation, 2016-05-07:
Alright -- 

v117 -- unable to send bugtool after 2 identical tests
1 - Select LLMP & PhpMyAdmin
2 - Select OpenSSH
3 - Begin Install
4 - Receive Error 100 on apt-get for phpmyadmin
5 - Machine sits at last screen shot

Bug can be repro'd at will, however there are three factors I am looking at:

1 - Selecting PhpMyAdmin only (as you did, @Fourdee)
2 - From the main software menu, selecting Lighttpd as my webserver (instead of selecting both under optimized software)
3 - I have seen similar reports and will be documenting those and my next test

[image]
[image]
[image]
[image]
[image]

xenfomation, 2016-05-07:
Oh - I used both static and dynamic IPs for my test, as well.  Not helpful yet, but phpmyadmin is angry about something.

xenfomation, 2016-05-07:
So last note as I need to grab a pint and some food w/ family.  Manually, I can install this.  Using the current v117 utils, 1) phpmyadmin database does not get created and 2) it may be an order of operations issue... investigating...

xenfomation, 2016-05-07:
So, mysql_secure_install is not getting called -- this is causing the error with phpmyadmin -- working on this as we speak.

xenfomation, 2016-05-08:
Manual commands I ran from test 4 or 5:

23  netstat -an | grep 80
   24  telnet 127,0.0.1 80
   25  nestat -an | grep 3306
   26  dpkg --list | grep mysql
   27  /etc/init.d/mysql start
   28  apt-get install php5-fpm php5
   29  vi /etc/php5/fpm/php.ini
   30  systemctl restart php5-fpm.service
   31  cd /etc/lighttpd/conf-available/
   32  ls
   33  cp 15-fastcgi-php.conf 15-fastcgi-php-spawnfcgi.conf
   34  vi 15-fastcgi-php.conf
   35  lighttpd-enable-mod fastcgi
   36  lighttpd-enable-mod fastcgi-php
   37  ls -l /etc/lighttpd/conf-enabled/
   38  service lighttpd.service restart
   39  service lighttpd restart
   40  ls
   41  apt-cache search php5
   42  ifconfig
      44  apt-get install php5-mysqlnd php5-curl php5-gd php5-intl php-pear php5-imagick php5-imap php5-mcrypt php5-memcache php5-pspell php5-recode php5-sqlite php5-tidy php5-xmlrpc php5-xsl
   45  apt-get install php5-xcache
   46  service php5-fpm restart
   47  apt-get install phpmyadmin
   48  mysql_secure_installation
   49  dbconfig-load-include
   50  apt-get install phpmyadmin
   51  reboot
   52  history

xenfomation, 2016-05-08:
Due to webserver + stack if conditions in v177, trying to re-create the install (ALL CLI) to ensure I can overcome this effort.

xenfomation, 2016-05-08:
I've come to the conclusion if LLMP is selected + PhpMyAdmin, we need to generate a routine to knock this out.

xenfomation, 2016-05-08:
Took a break to think, back on the case.  Using sandbox build v177 and real-Pi DietPi.... if that makes sense...

xenfomation, 2016-05-09:
No time to be slapdash, but I am still moving on this and keeping a previous comment in this case that is probably the key to something I overlooked:

"WEBSERVER_MYADMINPHP will now trigger webserver + mysql installation" -- @Fourdee, A few days ago

xenfomation, 2016-05-10:
@Fourdee 10 hours ago I wiped an SD card, exploded DietPi onto it, and updated all the way to v117 -- just like I had over the weekend.

Using dietpi-software, I selected LLMP & PhpMyAdmin as I had before.  No fatal errors this time (aside from a minor --configure notice) and presto -- this all works.

The confusion comes as this was clearly not working before.  I will test again tomorrow, but wow -- sending a bugtool report asap.

[image]
[image]

Fourdee, 2016-05-10:
@xenfomation 

> Using dietpi-software, I selected LLMP & PhpMyAdmin as I had before. No fatal errors this time (aside from a minor --configure notice) and presto -- this all works.

Yep, this very strange. Sometimes packages/dependencies have issues and Its not unheard of with apt-get and repos. 

I'll run the same test on my C2 and see if it still occurs.

xenfomation, 2016-05-10:
Good man, sir. It is 4:21 EST and I leave the office at 6pm. I will have 2 PIs primed and may follow the install order and insert:

mysql secure install

Followed by:
apt get install phpmyadmin

It is a bit hackish, but if it works consistently it can afford us time to add sparse debug output to a file as each step is carried out.

xenfomation, 2016-05-13:
OKay -- wrapping up final code tests for WEBSERVER_MYADMINPHP  and order of operations to ensure 100% success.

xenfomation, 2016-05-13:
Hey, @Fourdee -- I am including @rhkean in this write-up as well because... I did a lot of work, appreciate the patience, and I have had continued success with one issue.  PhpMyAdmin - no matter what I do - throws error 100.  So, if either of you have to to look at what I am about to put a pull request in for and give it a go, that would be exceptional (maybe I over looked something in the dietpi-software utility).

### APT-GET and ERROR 100
I have worked with Debian for as long as I can recall and this strange error - old Deb, new Deb, Ubuntu, etc - has never been clearly defined.  Everything from transient target package permissions during apt's heavy lifting to "we don't know, just install packages individually."

My two favorite odd answers are "low disk space" (not true) and "if apt is updated and then upgraded" (no way).

I've attached screenshots, so moving on....

### NOTEABLE CHANGES
When MySQL is installed, the debconf lines have been modified to put the password for MySQL in quotes.

The order of operations - installing lighttpd, php/fastcgi, mysql, and phpmyadmin - have had subtle changes with the inclusion of the MySQL service being restarted (along with php5-fpm) before phpmyadmin is attempted to be installed.

### CODE
I will attempt a pull request as I took a fork, etc but reading up on Git and trying to make sure I am proficient, it may be smaller than my last pull request.  The key parts are the WEBSERVER stack conditions.

Cheers!
[image]
[image]
[image]
[image]
[image]

Fourdee, 2016-05-13:
> When MySQL is installed, the debconf lines have been modified to put the password for MySQL in quotes.
The order of operations - installing lighttpd, php/fastcgi, mysql, and phpmyadmin - have had subtle changes with the inclusion of the MySQL service being restarted (along with php5-fpm) before phpmyadmin is attempted to be installed.
CODE
I will attempt a pull request as I took a fork, etc but reading up on Git and trying to make sure I am proficient, it may be smaller than my last pull request. The key parts are the WEBSERVER stack conditions.

@xenfomation 
Interesting and good find :+1: , looking forward to seeing the pull request when you can.

xenfomation, 2016-05-13:
Crud!  I was up late, finally slept with my laptop open and headed into the office without it. Will be added tonight!

Fourdee, 2016-05-14:
Tests from C2:
```
Creating config file /etc/phpmyadmin/config-db.php with new version
ERROR 1045 (28000): Access denied for user 'root'@'localhost' (using password: Y                                                                 ES).
unable to connect to mysql server.
error encountered creating user:
ERROR 1045 (28000): Access denied for user 'root'@'localhost' (using password: Y                                                                 ES)
dbconfig-common: phpmyadmin configure: noninteractive fail.
dbconfig-common: phpmyadmin configure: ignoring errors from here forwards
populating database via sql...  done.
dbconfig-common: flushing administrative password
Enabling phpmyadmin: ok
Enabling auth: ok
Enabling fastcgi: ok
Met dependency: fastcgi
Enabling fastcgi-php: ok
already enabled
Run /etc/init.d/lighttpd force-reload to enable changes
dpkg: error processing package phpmyadmin (--configure):
 subprocess installed post-installation script returned error exit status 2
Processing triggers for libc-bin (2.19-18+deb8u4) ...
Processing triggers for php5-fpm (5.6.20+dfsg-0+deb8u1) ...
Errors were encountered while processing:
 phpmyadmin
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
Will take a look later today.

Fourdee, 2016-05-16:
Looks like mysql install isnt applying credentials correctly:
```
root@DietPi:~# /DietPi/dietpi/func/create_mysql_db testdb dietpi root dietpi

 DietPi: Creating MySql DB for testdb
ERROR 1045 (28000): Access denied for user 'root'@'localhost' (using password: YES)
 Create database testdb | Failed

root@DietPi:~# mysql -p
Enter password:
ERROR 1045 (28000): Access denied for user 'root'@'localhost' (using password: YES)
```

Fourdee, 2016-05-16:
Issue is quotes gets sent as password. So in this case, password is actually 'dietpi'
```
echo -e "mysql-server mysql-server/root_password password 'dietpi'" | debconf-set-selections
```

Fourdee, 2016-05-16:
As a test, I installed LLMP, rebooted, then manually installed phpmyadmin with apt-get:
```
Creating config file /etc/phpmyadmin/config-db.php with new version
granting access to database phpmyadmin for phpmyadmin@localhost: success.
verifying access for phpmyadmin@localhost: success.
creating database phpmyadmin: success.
verifying database phpmyadmin exists: success.
populating database via sql...  done.
dbconfig-common: flushing administrative password
Enabling phpmyadmin: ok
Enabling auth: ok
already enabled
already enabled
Run /etc/init.d/lighttpd force-reload to enable changes
dpkg: error processing package phpmyadmin (--configure):
 subprocess installed post-installation script returned error exit status 2
Processing triggers for libc-bin (2.19-18+deb8u4) ...
Processing triggers for php5-fpm (5.6.20+dfsg-0+deb8u1) ...
insserv: warning: current start runlevel(s) (empty) of script `php5-fpm' overrid                                                                 es LSB defaults (2 3 4 5).
insserv: warning: current stop runlevel(s) (0 1 2 3 4 5 6) of script `php5-fpm'                                                                  overrides LSB defaults (0 1 6).
insserv: warning: current start runlevel(s) (empty) of script `php5-fpm' overrid                                                                 es LSB defaults (2 3 4 5).
insserv: warning: current stop runlevel(s) (0 1 2 3 4 5 6) of script `php5-fpm'                                                                  overrides LSB defaults (0 1 6).
Errors were encountered while processing:
 phpmyadmin
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
Same issues.

Did what it asked, now installed fine:
```
root@DietPi:~# /etc/init.d/lighttpd force-reload
[ ok ] Reloading lighttpd configuration (via systemctl): lighttpd.service.
root@DietPi:~# dpkg --configure phpmyadmin
Setting up phpmyadmin (4:4.2.12-2+deb8u1) ...
dbconfig-common: writing config to /etc/dbconfig-common/phpmyadmin.conf
Replacing config file /etc/phpmyadmin/config-db.php with new version
granting access to database phpmyadmin for phpmyadmin@localhost: already exists.
creating database phpmyadmin: already exists.
populating database via sql...  done.
dbconfig-common: flushing administrative password
[ ok ] Reloading web server configuration: lighttpd.
```

Fourdee, 2016-05-16:
I just wanted to see if this is a specific DietPi issue, its not.

I did a manual ```apt-get``` installation of ```mysql-server```, ```lighttpd``` and ```phpmyadmin```, same issues. So the cause lies with the apt-get packages and their installation scripts. It is not related to DietPi in any way.

I'am personally not going to spend anymore time on this issue. 
For now, we just need to make users aware they need to install phpmyadmin separately , after LLMP and a reboot.

@xenfomation @rhkean Unless you guys have any ideas or suggestions, I'll mark this as a nofix and remove it from v118 milestone.

xenfomation, 2016-05-16:
No - I agree as the conf packages and my research (for sanity) shows this to be an on/off issue... I merely attempted to apply the manual order I always had success with + debconf for automation, so I concur with your ideas, @Fourdee

rhkean, 2016-05-16:
Can we add a check in the code for phpmyadmin that verifies that mariadb or mysql = 2 (not 1) and pop up a message?

that way...
 we can set mariadb=1 if the user just selects phpmyadmin w/out selecting a database
 Then, pop up a message that says, "please install phpmyadmin after the database is installed"

xenfomation, 2016-05-16:
I do like that idea very much.  I can use this to reconfigure stupid git desktop :)  and the warning is easy enough to pop up. 

Whatever is decided on and who should do it, I support is my point. My personal backlog is living, adjusting my personal lab, and reconfiguring git desktop tonight / installing a rack for SBCs I will begin to acquire :)

Fourdee, 2016-05-17:
@rhkean 
As per your suggestion :+1: , this should do it: https://github.com/Fourdee/DietPi/commit/4d87e824bc9d5a151b12473c24e7881cd7902cf3

> reconfiguring git desktop tonight / installing a rack for SBCs I will begin to acquire :)
@xenfomation Add a glass of wine and you've got yourself a good night in lol :)

xenfomation, 2016-05-18:
Indeed, sir!  However, being in a working mood, I will make it a goal tonight!

MichaIng, 2017-12-02:
Just rechecked the error on fresh VirtualBox Stretch v158:
```
dpkg: error processing package phpmyadmin (--configure):
 subprocess installed post-installation script returned error exit status 2
```
So issue persists.

But solution with user notification, cancelling phpmyadmin installation to rechoose after reboot, works well.
I manually removed the prevention code to force parallel installation. Reboot and reselecting phpmyadmin for installation via dietpi-software also worked without issue.
Also it recognizes mariadb `unix_socket` authentication (as mariadb will be actually installed on Stretch by `mysql-server` package) and deals great with it.

MichaIng, 2018-08-08:
- [ ] Reckeck+Solve as well for automated installs

MichaIng, 2018-08-27:
Okay PR is up to solve this as above: https://github.com/Fourdee/DietPi/pull/2040

Fourdee, 2018-08-27:
@MichaIng 

Great work fixing a 2year+ ticket 👍 🥇 

Marking as resolved.

closed 2018-08-27 by Fourdee