MichaIng/DietPi #319: RPI v117 - Bug Report Date for The Imperialists

xenfomation, 2016-05-07:
Low priority question for @Fourdee and team.

Despite my own logical preference for items, such as the dietpi-bugtool, to report in DD-MMM-YYYY (smallest to largest), is it work considering that:

- Based on localization/timezone
- Or, per logic in such scripts

... We support the Imperialistic Way of recording time?  IE, today is MAY-07-2016, but I prefer DD-MMM-YYYY?  This is just a question to ensure maximum satisfaction for the masses.

Last item is, per the bugtool, I can suss that out within the week in binary format: either DD-MMM-YYYY or MMM-DD-YYYY as needed.

Thanks, everyone!
[image]

rhkean, 2016-05-07:
from a programmatic perspective, neither is preferred
for sorting, largest-to-smallest is best
yyyy-mm-dd will numerically sort (with or without the dashes) across all dates

I, personally, also find that it reduces ambiguity.
if I see a date that starts with the year, I know that it's yyyy-mm-dd.
However, if it ends with the year and the day is less than 13, then I don't know if I'm looking at 
mm-dd-yyyy or dd-mm-yyyy

(just my 2 cents)

xenfomation, 2016-05-07:
And thanks, @rhkean  -- that is what I am looking for.

I forgot to specify as my DietFu grows stronger, have the bug tool is awesome -- and the DATE gives me an indication of the client's locale, time preferences, etc so I can dig into the logs.

BUT I APPRECIATE THE FEEDBACK, SERIOUSLY!  :-)  I know I don't get the bugreports, but if I do -- there are a handful of timezone formats to support from linux (ISO 18xx), DMY, YMD, and some dotted items.

Appreciate the feedback and If you and @Fourdee both agree it is really silly, I can halt and just attach the code to this ticket.  <Now I go into my cave and focus back on Lighttpd and the other tickets my eager fingers are eager to fix, assist with, and spit polish.

Don't flog me -- my I forgot my root 10 languages from IBM (Swahili was one), but this is a silly example if the ISO 8601 date format and that... you are probably right & we just need to jump on board.

Sigh.... trying to regain my inner perfection amidst DietPi's team <sniff> :)  

[https://en.wikipedia.org/wiki/Date_format_by_country](https://en.wikipedia.org/wiki/Date_format_by_country)

Fourdee, 2016-05-07:
For me, DD-MM-YYYY is the UK standard that I'am used to. I believe the US use MM-DD-YYYY? So I suppose it really comes down to "what you got brought up with" :)

So the current date is pulled with ```DATE_CURRENT=$(date +"%d-%m-%Y")```

After reading Rob's comment, I have agree with him. YYYY-MM-DD seems a better solution and more logical. As far as I can remember, all code in DietPi that uses dates, uses DD-MM-YYYY. So if we change the layout order, we really should do it project wide.

rhkean, 2016-05-07:
doh!

I didn't mean to open that can of worms.... 

😸

k-plan, 2016-05-07:
> neither is preferred
for sorting, largest-to-smallest is best
yyyy-mm-dd will numerically sort (with or without the dashes) across all dates

Correct for files like Bug-Report, Backups, Configs with date in the beginning of the file name, which will be store in folders.
So you have a auto sorting and more overview and organisation.

> uses DD-MM-YYYY. So if we change the layout order, we really should do it project wide.

Hmm ... no, let well enough alone. It's okay.

> I didn't mean to open that can of worms....

:no_mouth:

xenfomation, 2016-05-07:
Then this meeting is adjourned and thanks for hearing me out!  I will let @Fourdee decide how this should be closed (or wiped off the map)!  Cheers!

And yes... the US utilizes the annoying "Month Day, Year" format.  Ugh.  That's why I had to share the Wikipedia Map of ISO date adoption across the globe. :-)

xenfomation, 2016-05-07:
In the end -- if it is ever decided upon after bugs and high feature requests, a lil' recursive grep should help me help us adopt yyyy-mm-dd

Fourdee, 2016-07-23:
Outdated ticket, marking as closed. Please reopen if required.

closed 2016-07-23 by Fourdee