MichaIng/DietPi #323: DietPi-Software | Add additional linux software options

Fourdee, 2016-05-09:
As title.

xenfomation, 2016-05-09:
<bow>

xenfomation, 2016-05-09:
The following changes were made based on enhancement #295 - a request to add vim-tiny as an optional linux package one could install through DietPi instead of the command line.

The software packages I added, outside of an effort to organize/alphabatize packages, were as follows:

### Editors
vim-tiny (compact release of Vim)
GNU Emacs
Midnight Commander's mcedit (already included with Midnight Commander)
Jed

### Network Utils
iftop
iptraf
iperf
mtr-tiny
nload
tcpdump

- All packages are at the latest release level for Jessie.
- All packages exist for older versions of Debian that DietPi uses, such as for Wheezy's sake.

xenfomation, 2016-05-09:
https://github.com/Fourdee/DietPi/commit/d5f2233a0b7060cbc38d586c468ff18f2e38f949

closed 2016-05-09 by xenfomation

Fourdee, 2016-05-10:
@xenfomation 
Excellent additions, great work on this :+1: