MichaIng/DietPi #324: Can we install this package for Audio?

xenfomation, 2016-05-10:
[http://www.theverge.com/circuitbreaker/2016/5/8/11635442/raspberry-pi-tube-amp](http://www.theverge.com/circuitbreaker/2016/5/8/11635442/raspberry-pi-tube-amp)

Sorry - I had to ask.  Electrical and Audio engineering runs deep in our family.

Fourdee, 2016-05-10:
@xenfomation Aye go for it :+1:. We love our audio here at DietPi lol

> 32–300 ohms

300ohms is a little low, 600ohms would of covered the high end and studio headphones. Although it would power  the DT 990 pros @ 250ohms.
I'am just curious if this gives a "tube amp" effect (that would be awesome), or if its just a glorified headphone amp? Either way, its got me interested lol :)

xenfomation, 2016-05-10:
Yeah, sadly I think the request falls on my left ear (it doesn't work) as you are right:

It's taking clipped digital audio and trying to remove the harshness on output

Hoped all would like this and with permission, requesting to close this request/question :-)

Fourdee, 2016-05-11:
Tube amps can improve quality and depth to any music. They give a unique output that's similar to compression and maximization of the audio dynamics. It could even be classed as "Phat". At least, that was my experience with software tube amp emulation when making music: http://help.izotope.com/docs/ozone/pages/mod_harmonic_exciter.htm

Couple that with a high quality DAC, this thing is not just a sound card. Its a high end device that will give audiophiles with headphones something unique. Yes its a bit "gimmicky", but if the device keeps to its word, it could be something special.

> requesting to close this request/question :-)

If I had the $100+ to buy one, I would of said no lol. But sure, close away :)

xenfomation, 2016-05-11:
Closing until I can buy us all one

closed 2016-05-11 by xenfomation