MichaIng/DietPi #325: DietPi Idea | Sourceable script with pre-defined color codes

xenfomation, 2016-05-10:
Hey, @Fourdee --

I remember something I did for an SSH-based BBS I did that leveraged BASH and I hope it may come of use to DietPi:

A master text/console script that can be sourced, based on TTY/console capabilities and hardware, to put all the escape color codes into variable containers:

purple=\e[35;... etc.
red=...
magenta=...

As I acquired different SBCs and work from different TTY/consoles/etc, I figured it may be a shortcut in the long run (or two steps back)?

Cheers!

Fourdee, 2016-05-10:
@xenfomation 

Sounds good :+1: 

Would this be a script we could call (eg ```/DietPi/dietpi/func/thisscript blue```)? And for clearing ```/DietPi/dietpi/func/thisscript``` or ```/DietPi/dietpi/func/thisscript reset```, something like that?

xenfomation, 2016-05-10:
Almost exactly!  I will send you a copy of one I have on an old system.

I do like the idea of adding a reset option to clear variables after use!!

xenfomation, 2016-05-11:
The script is really meant to be sourced in for userspace via the bash profile.  This way, $blue is defined and can be reference by all of the scripts.

Attached is a basic sample I used forever and usually loaded on boot as not to call the script all the time.

Thanks!
[colorscript.txt](https://github.com/Fourdee/DietPi/files/258590/colorscript.txt)

Fourdee, 2016-05-11:
Awesome :+1:. 

So these vars will get loaded into current bash session, in memory? Could probably be called by our ```login``` script for ease. Although we would have to also call it during boot if ```dietpi-banner```  is using it.

Probably a bit overkill, but this could also work in an array? Would prevent having multiple vars with different names. eg:
```
	#Colour array
	#0 WHITE
	#1 RED
	#2 GREEN
	#3 YELLOW
	#4 BLUE
	#5 PURPLE
	#6 CYAN
	#7 TEST
	aCOLOUR=(
		"\e[39m"
		"\e[31m"
		"\e[32m"
		"\e[33m"
		"\e[34m"
		"\e[35m"
		"\e[36m"
		"\e[93m"
	)
```
Just throwing some ideas :)

xenfomation, 2016-05-12:
These are great ideas as my fear was occupying too much memory just for colors.

I like the array reference and with a few arguments, we could definitely give the options to:

/script/path/colors load

/script/path/colors unload

and even

/script/path/colors use  (just to use 1 color shortcut based on your idea earlier....

Fourdee, 2016-07-23:
Implemented with dietpi-notify. https://github.com/Fourdee/DietPi/blob/master/dietpi/func/dietpi-notify

closed 2016-07-23 by Fourdee