MichaIng/DietPi #326: i can't install dietpi-software (is always ethernet offline) and have problems (awk not found)

Fourdee, 2016-05-11:
> http://dietpi.com/phpbb/viewtopic.php?f=11&t=408
```/DietPi/dietpi/dietpi-config: line 1857: awk: command not found```
465e4d7f-2d25-4011-9c41-569faec28f56-1

Fourdee, 2016-05-11:
awk is installed:
```
ii  mawk                                  1.3.3-17                        armhf        a pattern scanning and text processing language
```

v117 package installations are not installed:
```
ifmetric
netplug
```

Fourdee, 2016-05-11:
Fix apt:
```
cat << _EOF_ > /etc/apt/sources.list
deb http://mirror.ox.ac.uk/sites/archive.raspbian.org/archive/raspbian jessie main contrib non-free rpi
_EOF_
/DietPi/dietpi/dietpi-apt-get_update 2
```

Check awk binary:
```
ls -lha /usr/bin/mawk
echo -e "test" | /usr/bin/mawk '{print $1}'
```

Fourdee, 2016-05-11:
Cant confirm a bug at the moment. Pending results from user: http://dietpi.com/phpbb/viewtopic.php?f=11&t=408&p=1766#p1766

vmfventura, 2016-05-11:
`` ───────────────────────────────────────
 DietPi     | 17:53 | Wed 11/05/16
 ───────────────────────────────────────
 V117       | Raspberry Pi Unknown
 ───────────────────────────────────────
 IP Address |
 ───────────────────────────────────────

 Waiting for apt-get update to finish: Please wait...

 apt-cache database size: 43320 KB

 #               processing | D I E T  P I             #

root@rpiventura:~# apt-get install -y ifmetric netplug
Reading package lists... Done
Building dependency tree
Reading state information... Done
ifmetric is already the newest version.
netplug is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 40 not upgraded.
root@rpiventura:~# ls -lha /usr/bin/mawk
-rwxr-xr-x 1 root root 93K Apr  8  2012 /usr/bin/mawk
root@rpiventura:~# echo -e "test1 test2" | /usr/bin/mawk '{print $1}'
test1
root@rpiventura:~# /DietPi/dietpi/misc/rpi_boardinfo
/DietPi/dietpi/misc/rpi_boardinfo: line 33: awk: command not found

  - RPi Board Info -
 Revision         |
 Released         |     Unknown
 Model            |     Unknown
 Memory           |     0 MB
 Manufacturer     |     Unknown

root@rpiventura:~#`

`root@rpiventura:~# dietpi-software
/DietPi/dietpi/dietpi-software: line 138: awk: command not found
 ───────────────────────────────────────
 DietPi     | 17:56 | Wed 11/05/16
 ───────────────────────────────────────
 V117       | Raspberry Pi Unknown
 ───────────────────────────────────────
 IP Address |
 ───────────────────────────────────────

 DietPi-Software

Testing connection to , please wait.....
┤ No Internet Connection! ├─────────────────────┐
     │                                                                    │
     │ Internet access is required to start the DietPi Installation       │
     │ Program.                                                           │
     │ Would you like to change your network settings and try again?      │
     │                                                                    │
     │ If problems persist:                                               │
     │  may be offline and/or unreachable.                                │
     │                                                                    │
     │                                                                    │
     │                                                                    │
     │                                                                    │
     │                  <Ok>                      <Exit> `

Fourdee, 2016-05-11:
@vmfventura 
Thanks for the results.

Looks like you have an issue with bash as it does not recognise ```awk``` as a command, yet its installed.

Can you paste the results of:
```
cat /root/.bashrc
```

vmfventura, 2016-05-11:
root@rpiventura:~# cat /root/.bashrc
cat: /root/.bashrc: No such file or directory
root@rpiventura:~#

Fourdee, 2016-05-11:
I did a diff check on your /etc/bash.bashrc seems fine: https://www.diffchecker.com/c2yysuc7

You have missing code for .bashrc. Although it probably wont fix the AWK issue, DietPi needs this to function correctly.
Please create /root/.bashrc:
```
# ~/.bashrc: executed by bash(1) for non-login shells.

# Note: PS1 and umask are already set in /etc/profile. You should not
# need this unless you want different defaults for root.
# PS1='${debian_chroot:+($debian_chroot)}\h:\w\$ '
# umask 022

# You may uncomment the following lines if you want `ls' to be colorized:
# export LS_OPTIONS='--color=auto'
# eval "`dircolors`"
# alias ls='ls $LS_OPTIONS'
# alias ll='ls $LS_OPTIONS -l'
# alias l='ls $LS_OPTIONS -lA'
#
# Some more alias to avoid making mistakes:
# alias rm='rm -i'
# alias cp='cp -i'
# alias mv='mv -i'
/DietPi/dietpi/login
```

Also, please paste results of:
```
echo -e "$PATH"
```

vmfventura, 2016-05-11:
root@rpiventura:~# nano /root/.bashrc
root@rpiventura:~# echo -e "$PATH"
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin

Fourdee, 2016-05-11:
Weird, doesn't make sense why its still reporting command not found. ```mawk``` is installed and function, your ```PATH``` is correct.

Ok, lets try a little "cheat" and see if it allows us to progress:
```
echo -e "alias awk='/usr/bin/mawk'" >> /etc/bash.bashrc
```

close ssh session and log back in, then try ```dietpi-software``` again.

@xenfomation @rhkean @k-plan
Would appreciate any assistance on this one, if any of you have some ideas to try?

vmfventura, 2016-05-11:
tried, no results posted.

logged off, login and, same. no internet connection.

> Testing connection to , please wait.....
/DietPi/dietpi/func/obtain_network_details: line 83: awk: command not found
/DietPi/dietpi/dietpi-config: line 1857: awk: command not found
/DietPi/dietpi/dietpi-config: line 1858: awk: command not found
/DietPi/dietpi/dietpi-config: line 1859: awk: command not found
/DietPi/dietpi/dietpi-config: line 1860: awk: command not found
/DietPi/dietpi/dietpi-config: line 1873: awk: command not found
/DietPi/dietpi/dietpi-config: line 1874: awk: command not found
/DietPi/dietpi/dietpi-config: line 1875: awk: command not found
/DietPi/dietpi/dietpi-config: line 1876: awk: command not found
/DietPi/dietpi/dietpi-config: line 1913: awk: command not found
/DietPi/dietpi/dietpi-config: line 1914: awk: command not found
/DietPi/dietpi/dietpi-config: line 1915: awk: command not found
/DietPi/dietpi/dietpi-config: line 1916: awk: command not found
──┤ No Internet Connection! ├─────────────────────┐
     │                                                                    │
     │ Internet access is required to start the DietPi Installation       │
     │ Program.                                                           │
     │ Would you like to change your network settings and try again?

vmfventura, 2016-05-11:
I forgot something..
after update to v117, i cannot access to transmission(:9091) and couchpotato(:5050) over the internet, and i can access to linuxdash (/linuxdash) and plex (:32400)

it's strange. in other device (odroid c2), i have the same problem (accessing services from outside).

xenfomation, 2016-05-11:
I will be diving into this after 6pm EST latest, earlier is my plan though. Will read phpbb post first as I have many questions. I have only seen this, for the record, when I attempted to build DietPi on top of Debian.

xenfomation, 2016-05-11:
Awk is in /usr/sbin/ and mawk is there as well. 

So, on my way to see what this could be related to and I have some questions:

Can you provide me the output of:

df -h

free -m

env

uname -a.

And

dpkg --list

Secondly, how long have you had DietPi and this software installed on this SD or USB card?

Lastly,

xenfomation, 2016-05-11:
My initial reaction is a software install conflict or package substitute. 

I will keep all posted.

vmfventura, 2016-05-11:
> root@rpiventura:~# df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        15G  2.6G   12G  19% /
devtmpfs        459M     0  459M   0% /dev
tmpfs           463M  4.0K  463M   1% /dev/shm
tmpfs           463M  6.7M  457M   2% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           463M     0  463M   0% /sys/fs/cgroup
tmpfs            10M  1.1M  9.0M  11% /DietPi
tmpfs           463M     0  463M   0% /tmp
tmpfs            20M  964K   20M   5% /var/log
/dev/sdb1        57G   11G   44G  21% /mnt/usb_1
/dev/mmcblk0p1   60M   22M   39M  36% /boot
/dev/sdc1       932G  801G  131G  86% /mnt/USBHDD1
/dev/sda1       1.9T  516G  1.4T  28% /mnt/USBHDD2
tmpfs            93M     0   93M   0% /run/user/111
tmpfs            93M     0   93M   0% /run/user/0

> root@rpiventura:~# free -m
             total       used       free     shared    buffers     cached
Mem:           925        913         12         18        322        402
-/+ buffers/cache:        187        738
Swap:           99          0         99

> root@rpiventura:~# env
XDG_SESSION_ID=c5
TERM=xterm
SHELL=/bin/bash
SSH_CLIENT=*.*.*.* 1027 22
SSH_TTY=/dev/pts/1
USER=root
MAIL=/var/mail/root
PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
PWD=/root
LANG=en_GB.UTF-8
SHLVL=1
HOME=/root
LOGNAME=root
SSH_CONNECTION=*.*.*.* 1027 10.0.1.6 22
XDG_RUNTIME_DIR=/run/user/0
_=/usr/bin/env

> root@rpiventura:~# uname -a
Linux rpiventura 4.1.19-v7+ #858 SMP Tue Mar 15 15:56:00 GMT 2016 armv7l GNU/Linux

> root@rpiventura:~# dpkg --list
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-=================================
ii  acl            2.2.52-2     armhf        Access control list utilities
ii  adduser        3.113+nmu3   all          add and remove users and groups
ii  adwaita-icon-t 3.14.0-2     all          default icon theme of GNOME
ii  alsa-base      1.0.27+1     all          dummy package to ease purging of
ii  alsa-utils     1.0.28-1     armhf        Utilities for configuring and usi
ii  apache2        2.4.10-10+de armhf        Apache HTTP Server
ii  apache2-bin    2.4.10-10+de armhf        Apache HTTP Server (modules and o
ii  apache2-data   2.4.10-10+de all          Apache HTTP Server (common files)
ii  apache2-utils  2.4.10-10+de armhf        Apache HTTP Server (utility progr
ii  apt            1.0.9.8.3    armhf        commandline package manager
ii  apt-transport- 1.0.9.8.3    armhf        https download transport for APT
ii  apt-utils      1.0.9.8.3    armhf        package management related utilit
ii  base-files     8+rpi1       armhf        Debian base system miscellaneous
ii  base-passwd    3.5.37       armhf        Debian base system master passwor
ii  bash           4.3-11       armhf        GNU Bourne Again SHell
ii  bash-completio 1:2.1-4      all          programmable completion for the b
ii  bc             1.06.95-9    armhf        GNU bc arbitrary precision calcul
ii  binutils       2.25-5       armhf        GNU assembler, linker and binary
ii  bluez          5.23-2+rpi1  armhf        Bluetooth tools and daemons
[... 621 lines cut ...]
ii  x11-common     1:7.7+7      all          X Window System (X.Org) infrastru
ii  x11-utils      7.7+2        armhf        X11 utilities
ii  x11-xkb-utils  7.7+1        armhf        X11 XKB utilities
ii  x11-xserver-ut 7.7+3        armhf        X server utilities
ii  xarchiver      1:0.5.4-1    armhf        GTK+ frontend for most used compr
ii  xauth          1:1.0.9-1    armhf        X authentication utility
ii  xfonts-base    1:1.0.3      all          standard fonts for X
ii  xfonts-encodin 1:1.0.4-2    all          Encodings for X.Org fonts
ii  xfonts-utils   1:7.7+2      armhf        X Window System font utility prog
ii  xinit          1.3.4-1      armhf        X server initialisation tool
ii  xkb-data       2.12-1       all          X Keyboard Extension (XKB) config
ii  xml-core       0.13+nmu2    all          XML infrastructure and XML catalo
ii  xserver-common 2:1.17.2-1+r all          common files used by various X se
ii  xserver-xorg   1:7.7+7+b1   armhf        X.Org X server
ii  xserver-xorg-c 2:1.17.2-1+r armhf        Xorg X server - core server
ii  xserver-xorg-i 1:7.7+7+b1   armhf        X.Org X server -- input driver me
ii  xserver-xorg-i 1:2.9.2-1~bp armhf        X.Org X server -- evdev input dri
ii  xserver-xorg-i 1.8.2-1~bpo8 armhf        Synaptics TouchPad driver for X.O
ii  xserver-xorg-v 1:0.4.4-1+rp armhf        X.Org X server -- fbdev display d
ii  xz-utils       5.1.1alpha+2 armhf        XZ-format compression utilities
ii  zip            3.0-8        armhf        Archiver for .zip files
ii  zlib1g:armhf   1:1.2.8.dfsg armhf        compression library - runtime
ii  zlib1g-dev:arm 1:1.2.8.dfsg armhf        compression library - development

I'm using dietpi almost 2 months +/-, i think since v112/113 (tomorrow i'll confirm)

xenfomation, 2016-05-11:
Thanks!!  I just left the office, @Fordee and will hit the data we've been provided as to ensure there is not a bug. 

Expect other updates as well for items.

xenfomation, 2016-05-11:
One concern. You've hit 99% of your swap of I read that correctly. I may have read that wrong and will confirm ASAP. 

If your swap is full, a reboot helps, but you have already done this earlier. 

My concern is that without more logs, I wonder if any services are unable to do their job fully. Ie, networking, socket based services. 

Will have more for everyone and thanks for the data!

vmfventura, 2016-05-11:
if i read correctly, swap is 0% used... 99%free...
if i backup dietpi, format sd card, reinstall and restore, solve my problem?

xenfomation, 2016-05-12:
Hey, @vmfventura  -- I did read that wrong (Gitclient from my mobile phone).

I am now at my home laptop and I see SWAP IS ACTUALLY GOOD:

```
root@rpiventura:~# free -m
             total       used       free     shared    buffers     cached
Mem:           925        913         12         18        322        402
-/+ buffers/cache:        187        738
Swap:           99          0         99
```

So forget what I was suggesting -- as I read the output backwards.

Re-reading data and looking at my local PI install of v117

xenfomation, 2016-05-12:
HI, @vmfventura --

I have been looking at the data you provided (thank you).  I have also been working to compare items with my PI 2.

###  ISSUE WITH AWK
On my system - upgrade from v115 to v117, I still have awk/mawk:
```
root@DietPi:~# which awk
/usr/bin/awk
root@DietPi:~# which mawk
/usr/bin/mawk

```

DPKG shows me I have mawk (just verifiying):
```
dpkg --list | grep -i awk
ii  mawk                           1.3.3-17                  amd64        a pattern scanning an

```

When I check the version of awk (really mawk as awk is a virtual package) that is installed, I see the following:
```
root@DietPi:~# awk -W version
mawk 1.3.3 Nov 1996, Copyright (C) Michael D. Brennan

compiled limits:
max NF             32767
sprintf buffer      2040

```

At present, I do not have an answer why the traditional "mawk" package did not install an awk package?  Did you compile anything from source, such as mawk itself before upgrading?  Was any software previously compiled before ever upgrading?

### IPTABLES

This was a good clue you have passed on:

> I forgot something..
> after update to v117, i cannot access to transmission(:9091) and couchpotato(:5050) over the internet, and i can access to linuxdash (/linuxdash) and plex (:32400)
> 
> it's strange. in other device (odroid c2), i have the same problem (accessing services from outside).

I see that you have GCC installed as well as IPTABLES.  If you shutdown IP tables, can you access :5050 and :9091?

I am honestly at a loss what is different between your DietPi v117 and mine except software.  That is why i have so many questions like this:

### Raspberry PI 3?
I see this package is installed --

ii  pi-bluetooth   0.1.0        armhf        Raspberry Pi 3 bluetooth

Since the 3 has bluetooth, I wanted to ask if you are running on a Raspberry PI 2 or a 3?  OR...
did you have to install this file for a bluetooth device?

Thank you very much as I wish we had network access to sort out your sytem in a live fashion, but please keep me updated!

Thanks @vmfventura & @Fourdee ...

vmfventura, 2016-05-12:
hi there....

```
root@rpiventura:~# which awk
root@rpiventura:~# which mawk
/usr/bin/mawk
```

```
root@rpiventura:~# dpkg --list | grep -i awk
ii  mawk                                  1.3.3-17                        armhf        a pattern scanning and text processing language
```
```
root@rpiventura:~# awk -W version
mawk 1.3.3 Nov 1996, Copyright (C) Michael D. Brennan
compiled limits:
max NF             32767
sprintf buffer      1020
```
```
root@rpiventura:~# ls -la /usr/bin/awk
lrwxrwxrwx 1 root root 21 Feb 26 00:57 /usr/bin/awk -> /etc/alternatives/awk
```
```
root@rpiventura:~# ls -la /etc/alternatives/*
ls: cannot access /etc/alternatives/*: No such file or directory
```
```
root@rpiventura:~# service iptables stop
Failed to stop iptables.service: Unit iptables.service not loaded.
```
Yes, i have 100% sure, that i have raspberry pi 2. i have 2, and 1 B.

xenfomation, 2016-05-12:
Thank you for your patience and excellent data. I promise I am not trying to torture you! :)

Two ideas...

### etc alternatives

Can you attempt to run:

update-alternatives --set awk /usr/bin/mawk 

### remote access

If you would feel comfortable, I would like to SSH to your PI and resolve this. 

@Fordee. You are the man. With the items post 117, I ask you put pressure on me to get the DietPi configuration database up. I feel this issue is not a bug, but a typical sideaffect due to the "everyone has root" access install mantra and it doesn't respect your authority!! :)

fordee, 2016-05-12:
I think you mean @fourdee.

vmfventura, 2016-05-12:
```
root@rpiventura:~# update-alternatives --set awk /usr/bin/mawk
update-alternatives: using /usr/bin/mawk to provide /usr/bin/awk (awk) in manual mode
update-alternatives: warning: skip creation of /usr/share/man/man1/awk.1.gz because associated file /usr/share/man/man1/mawk.1.gz (of link group awk) doesn't exist
update-alternatives: warning: skip creation of /usr/share/man/man1/nawk.1.gz because associated file /usr/share/man/man1/mawk.1.gz (of link group awk) doesn't exist
```

@xenfomation if i give you my ssh login, next i have to kill you...:P kidding. i can give you that... any way to give you, outside here?

xenfomation, 2016-05-12:
Thanks for the quick response!  Kill me?  But I'm already dead inside ;)

### update-alternatives
Okay, so the output you have provided is encouraging!  Now if you run the following commands and awk has magically returned, try to run dietpi-software and also let me know what happens.  SO... we are back to executing:

env

which awk

which mawk

ls -la /usr/bin/awk
ls -la /usr/bin/mawk
ls -la /etc/alternatives/awk

### Accessing your PI
It may be too complicated based on the state of your PI, but this is what I would do:

**Determine** your public IP:**
curl ipecho.net/plain; echo

**Find our your PI's IP Address**
ifconfig, right :)

**SSH PASS THROUGH**
You would need to configure your router to allow socket connections for port 22 (on your public IP) to be forwarded to your DeitPi system.

If you are up for it, let me know.  I will shoot you my public IP address in a private message and see if we can't make this finally work.

vmfventura, 2016-05-12:
@xenfomation workedddddddddddddddddddddd, i can access to dietpi-software...
you save your ass :D 

i access to my raspberry by ssh all the time.

i'm still having problems accessing to :9091(transmission) and :5050(couchpotato).
working fine owncloud (having problems and updating, but other problem), linuxdash and plexmediaserver

xenfomation, 2016-05-12:
@vmfventura EXCELLENT!  Thank you for your patience!

To make sure I am clear on next steps:

- Awk errors are gone/fixed (good)
- You can use dietpi-software, BUT you cannot do any updates (bad)
- And you still have the same problems with :9091, :5050, and Login Banner (unknown pi hardware) after updating to v117

Thanks, @vmfventura !!

xenfomation, 2016-05-12:
No problem, @vmfventura -- it's thanks to @Fourdee we all get to have fun with SBCs and DIETPI!

I am going to assign this to @Fourdee for these reasons:

- He is in charge!
- The core issue with dietpi-software was resolved by refreshing update-alternatives:
`update-alternatives --set awk /usr/bin/mawk` 
- THe issue with couchpotato is secondary, but still needs to be considered as in the end, I have no root-cause-analysis to say this is related to software "fighting each other"
- Also, I wonder if we can tie in a quick utility to help with dissonance/issues like this based on DietPi's means of storing what is installed, etc

### AS FOR COUCHPOTATO...
Once @Fourdee gives me the right direction to go, I wanted to ask if there are any specific error messages when you try to use Couchpotato?

Thanks!

Fourdee, 2016-05-13:
@xenfomation 
Excellent work on debugging this :+1: 

> Once @Fourdee gives me the right direction to go, I wanted to ask if there are any specific error messages when you try to use Couchpotato?

Ideally, we should only be supporting the software we have in ```dietpi-software```, else, we wouldn't get any work done lol. However, its always a personal choice, and time permitting, if we look into other "non-dietpi software" issues on a DietPi system.

So in other words, your call @xenfomation. 

@vmfventura, as there are so many "strange" issues on your system, I'd highly recommend a fresh installation. Install all the DietPi-Software programs you need, do a ```dietpi-backup``` then manually add the non-dietpi software you need. This way if the issue reoccurs, you can always revert back to a working system.

> after update to v117, i cannot access to transmission(:9091) and couchpotato(:5050) over the internet, and i can access to linuxdash (/linuxdash) and plex (:32400)

Are you using port forwarding per port to an IP, or, are you using a DMZ (device completely open to internet)?

vmfventura, 2016-05-13:
@Fourdee working port forwarding to ip.

any ideias to add plexmediaserver and couchpotato to your dietpi-software, as requested in forum?

i'll will reinstall everything...some that i don't want, but must have...
if i backup, reinstall and restore it work?

Fourdee, 2016-05-13:
@vmfventura 
We already have a ticket for Plex: https://github.com/Fourdee/DietPi/issues/116, I'll make a note to look into it again.

couchpotato, not seen this before, i'll take a look.

> if i backup, reinstall and restore it work?

```USBDRIVE=2``` your user data is on your USB drive ```/mnt/usb_1```, so you should be fine. Just remember to say "No" when it asks to format the drive during installation.
As for your plex/COUCHPOTATO, you'll need to figure out where your files are stored.

Either way, if your not sure, backup the USB drive and the SD card, before continuing.

vmfventura, 2016-05-13:
in /mnt/usb1 is owncloud and /mnt/USBHDD1 and 2 is everything...

i just want to know if save the config files... but i'll configure everything again (pain...)

to couchpotato and plex (search movies in torrents and usenet) i used this guides:
http://www.htpcguides.com/install-couchpotato-raspberry-pi-raspbian/
http://www.htpcguides.com/install-plex-media-server-on-raspberry-pi-2/
in raspberry. 
in odroid c2, couchpotato is the same, but plex is a little different:
```
sudo su -
apt-get install apt-transport-https
wget -O - https://dev2day.de/pms/dev2day-pms.gpg.key | apt-key add -
echo "deb https://dev2day.de/pms/ jessie main" >/etc/apt/sources.list.d/pms.list
dpkg --add-architecture armhf
apt-get update
apt-get install binutils:armhf
apt-get install --no-install-recommends plexmediaserver-installer
```
working like a charm

k-plan, 2016-05-14:
Just my 2cent:

> working like a charm

not really, or? 😄  
 `dpkg --add-architecture armhf` `apt-get install binutils:armhf`

**_Some problem are:_**
- DietPi now support so many different platforms and hardware SOC, for example
all Raspberry Pi's (different Broadcom CPU), Odroid C0, C1, C1+, XU3, XU4 , C2 (32 + 64bit arm Samsung a. Amlogic CPU), Orange Pi PC (Allwinner CPU), VMware x64, VirtualBox x64.

- DietPi want to support more different SOC and platforms (yes, here is a donations for funding, sponsoring, contribution problem)

- DietPi will be "diet" (lightweight), individually and easy configurable, stable and fast. "Get the most out of your device"

More software on more platforms = exponentially growing problems, bugs, support and amount of work and time.

**_Solution:_**
- more  donations for funding
- more contributor
- more sponsoring
- more support by user

Which part do you like to assume? 😄 
In German: "_ohne Arme - keine Kekse_" 
Sorry,  mean no harm.

rhkean, 2016-05-15:
I'd have to agree with @k-plan

xenfomation, 2016-05-15:
I agree as well.  I wish I could fix the issue, but I can't and as a developer -- my support intentions were to really help you -- and I did -- but also to make sure I found a root cause analysis to report to @Fourdee ... 

I didn't get there, but I really agree with everyone.  We do have much to work on, but also much to work for the very motto and spirit of DietPi.  With donations, support, and more adotion... as @k-plan said -- sure... we could BUILD a DietPi board.

For now, I really feel the issue is not us and while I really appreciate your interest in DietPi -- maybe start from scratch... add one software piece at a time.  If the same thing happens -- stop and note.   

Cheers everyone -- time for a pint and to hang with my kiddos!

thanks!

Fourdee, 2016-05-15:
> For now, I really feel the issue is not us and while I really appreciate your interest in DietPi -- maybe start from scratch... add one software piece at a time. If the same thing happens -- stop and note.

I also believe this is the way to move forward. It will also help us track down issues, when and if they occur.

@vmfventura I'd highly recommend a reinstall and to follow @xenfomation advice above by installing 1 program at a time. I personally want to help fix any issues our users have, however, to do that, we will need your assistance and testing to help make that possible.

I'll mark this ticket as closed for now. @vmfventura Please reopen the ticket if you encounter any further problems post-reinstall and we will try to assist you.

closed 2016-05-15 by Fourdee

cinghialino, 2017-07-30:
I tried to install samba, and this problem popped up. I see that there is no fix yet for this.