MichaIng/DietPi #329: Security problem with kernel on Orange Pi

zgoda, 2016-05-12:
And possibly elsewhere.

http://forum.armbian.com/index.php/topic/1108-security-alert-for-allwinner-sun8i-h3a83th8/

Fourdee, 2016-05-12:
@zgoda 

We have updated the OPi PC image, which is now built with ARMbian and disables the exploit: https://github.com/Fourdee/DietPi/issues/312

You can download this new image from http://dietpi.com/#five

closed 2016-05-12 by Fourdee

xenfomation, 2016-05-12:
Nicely done sir. If time allows, I wanted to speak to you @Fordee to calculate the SBC boards I should be soaking in.

Fourdee, 2016-05-12:
@xenfomation 

Without a doubt, Odroid C2 is the best all round SBC at the moment. Couple it with a EMMC module and it makes the Raspberry Pi 3 feel like a snail :) A snail thats 1+ year behind Odroid.

I use the C2 as my main test bed for all DietPi devolpment.

xenfomation, 2016-05-12:
Odroid it is!

ThomasKaiser, 2016-10-11:
@zgoda isn't that funny? A distro maintainer trying to solve security issues by deleting comments and playing censor:

- http://web.archive.org/web/20161011170340/https://github.com/Fourdee/DietPi/issues/312
- http://archive.is/xfMWR
- https://github.com/Fourdee/DietPi/issues/312

And in between it looked like this:

[image]

And here you find my first question deleted by @Fourdee:

[image]