MichaIng/DietPi #330: Update Subsonic to v6

thechildofroth, 2016-05-12:
Hi Fourdee (and the growning band of devs)

Just wanted to let you know that there is a new version of Subsonic available: http://downloads.sourceforge.net/project/subsonic/subsonic/6.0/subsonic-6.0.deb?r=https%3A%2F%2Fsourceforge.net%2Fprojects%2Fsubsonic%2Ffiles%2Fsubsonic%2F6.0%2F&ts=1463053419&use_mirror=tenet 

I just boosted it in to DP117 with dpkg -i and it seems to be up and running quite happily.

One small point (and it might be already fixed) when re directing the transcoder symlinks from var/subsonic.transcode ffmpeg is now in /usr/local/bin/... rather than /usr/local

xenfomation, 2016-05-12:
Nice!!!  I will take a look at it today (thanks for the sym link info) and report back to this ticket and @Fordee

xenfomation, 2016-05-12:
Pulling now...

xenfomation, 2016-05-12:
So far so good -- testing streaming later.  Already updated menu code if @Fourdee approves because sometimes... people want A SPECIFIC version of software :)

[image]
[image]
[image]

xenfomation, 2016-05-13:
@thechildofroth  -- that was the most painless testing i have done!  i have contributed software installer code for dietpi to @Fourdee for his review.  (Thanks!)

Off to keep tackling what I can! :-)

thechildofroth, 2016-05-13:
Thanks @xenformation might be worth adding the final v5 version too as I
believe the developer has shifted the licencing for v6 and closed the
sources, which might be an issue for some.

Fourdee, 2016-05-13:
@xenfomation 
That was fast lol, nice one :+1: 

Just need you to send a pull request when your ready. I think we will need to patch the .installed file with the ```SUBSONIC5/6``` changes, but i'll go through that in the pull request.

Fourdee, 2016-05-14:
Commit for reference: https://github.com/Fourdee/DietPi/commit/b229e757d36ec0defeb6aca02810ae6b2c86545d

Fourdee, 2016-05-14:
@xenfomation 

Added a few extra lines of code for the ```SUBSONIC5/6``` changes across DietPi scripts: 
https://github.com/Fourdee/DietPi/commit/027072e392969a75fb21335795d5c99e73d22692

xenfomation, 2016-05-15:
Despite my testing, glad you found that & thanks!

Fourdee, 2016-05-15:
Will mark this as closed.

closed 2016-05-15 by Fourdee

xenfomation, 2016-05-15:
Weekend shift done. And I saw the items you had to modify. Time to reinstall Git for Desktop as to avoid this again. Thanks!

MichaIng, 2018-01-18:
@Fourdee 
I guess we could remove SubSonic 5, don't we?

MichaIng, 2018-01-19:
@thechildofroth 
Thx for the hint. To be true, Libresonic doesn't seem to develop very fast. Last commit, even on develop branch, is from early Juli, where SubSonic had it's last release end of November. So until you see some more interest/development progress, I would stick with SubSonic. Though I am always sceptic, when prior open source software switch to closed source. Even their releases are still available on source forge, their source is not any more since 6.0 beta 🤔.

So an alternative way I just thinking about now is, to replace SubSonic 5 with current LibreSonic, for users that want to stay open source. SubSonic 6 then for those who still trust in it and want to stay on more current releases there.

thechildofroth, 2018-01-20:
Thanks.

Replacing 5 with libresonic and keeping the 6 branch updated (for those of
us who have subsonic licences) would be an ideal solution I think.

The optimist in me would like to think that perhaps some increased interest
in libresonic, thanks to it being available in DietPi, could spur some
further development over there. Anyway, it can't develop slower than the 5
branch of subsonic ;-) and being fully free it's much more in keeping with
the DP general ethos.

MichaIng, 2018-02-08:
Just found AirSonic, the next fork 😆: https://github.com/airsonic/airsonic