MichaIng/DietPi #331: Error: Obtained server version value from github is invalid.

inspector71, 2016-05-12:
```
 ───────────────────────────────────────
 DietPi     | 23:18 | Thu 05/12/16
 ───────────────────────────────────────
 V116       | Raspberry Pi 3 Model B
 ───────────────────────────────────────
 IP Address | 192.168.1.244
 ───────────────────────────────────────

Error: Obtained server version value from github is invalid. Please try running dietpi-update again.

```

xenfomation, 2016-05-12:
Yikes

What is the output of:

cat /DietPi/dietpi/.version

And

cat /DietPi/dietpi.txt

And also

cat /DietPi/dietpi/DietPi - update

Thanks!

Fourdee, 2016-05-12:
@inspector71 

This is a known issue: https://github.com/Fourdee/DietPi/issues/301

Please run:
```
sed -i '/gitbranch=master#Proxy settings/c\gitbranch=master' /DietPi/dietpi.txt

xenfomation, 2016-05-12:
You are speedy, man. Kudos.

inspector71, 2016-05-13:
that fixed the problem

thanks!

closed 2016-05-13 by inspector71