MichaIng/DietPi #333: DietPi-Software | Add installation option for Open Bazaar

Fourdee, 2016-05-13:


Fourdee, 2016-05-13:
Merged: https://github.com/Fourdee/DietPi/pull/336

Add:
- dietpi.txt automation entry

Test:
- C2

Fourdee, 2016-05-13:
> note to self:
- RPi install was fine
- Give OpenBazaar a test (currently installed on RPI v3 eth (2)) .105
- Update online documentation
- Eesolve or remove C2.

Fourdee, 2016-05-14:
@swordfish6975 

> ob.conf 
Set the username and password

To what? :) Documentation is a little "light": https://github.com/OpenBazaar/OpenBazaar-Server

swordfish6975, 2016-05-14:
Whatever you would like to enter in the client to connect with, preferably a very long password like a GUID. 

The risk to the user is someone getting in to there shop and listing stuff for sale and pocketing the money.

Fourdee, 2016-05-14:
@swordfish6975 
Thanking you. So just to confirm, the username/password can be anything. Its only purpose is to prevent the server being abused?

I installed the OpenBazaar client:
-  Is there any way to specifically connect/test our OpenBazaar server with it? 
- Or assuming there are no error logs in with ```python openbazaard.py start -d -a 0.0.0.0``` can I assume the server is fully functional and thats all that needs to be done?

Fourdee, 2016-05-14:
Service notes:

Start:
```
python openbazaard.py start -d -a 0.0.0.0
```

Stop:
```
kill $(cat /tmp/openbazaard.pid)
#rm /tmp/openbazaard.pid #OpenBazaar does this during shutdown.
```

swordfish6975, 2016-05-14:
python openbazaard.py stop

Fourdee, 2016-05-14:
@swordfish6975 

> ```python openbazaard.py stop```

Legend :+1:.

swordfish6975, 2016-05-14:
To test it I brought a steam game and I've had a order come through. One of my orders that was in progress (I migrated my shop over) had the funds released from the multisig address and it all wnet fine.

Fourdee, 2016-05-14:
Unable to run OpenBazaar from outside its own directory...

```
root@DietPi:~# python /etc/openbazaar-server/openbazaard.py start -d -a 0.0.0.0
Traceback (most recent call last):
  File "/etc/openbazaar-server/openbazaard.py", line 11, in <module>
    from api.ws import WSFactory, AuthenticatedWebSocketProtocol, AuthenticatedWebSocketFactory
  File "/etc/openbazaar-server/api/ws.py", line 16, in <module>
    from config import DATA_FOLDER, str_to_bool
  File "/etc/openbazaar-server/config.py", line 22, in <module>
    CONFIG_FILE = join(paths[0], paths[2])
IndexError: list index out of range
```

```
root@DietPi:~# cd /etc/openbazaar-server/
root@DietPi:/etc/openbazaar-server# python openbazaard.py start -d -a 0.0.0.0
________                      __________
\_____  \ ______   ____   ____\______   \_____  _____________  _____ _______
 /   |   \\____ \_/ __ \ /    \|    |  _/\__  \ \___   /\__  \ \__  \\_  __ \
/    |    \  |_> >  ___/|   |  \    |   \ / __ \_/    /  / __ \_/ __ \|  | \/
\_______  /   __/ \___  >___|  /______  /(____  /_____ \(____  (____  /__|
        \/|__|        \/     \/       \/      \/      \/     \/     \/

OpenBazaar Server v0.1 starting...
```

Fourdee, 2016-05-15:
Resolved inability to run OpenBazaar from outside its own location. Not happy with it, but its a workaround, commit: https://github.com/Fourdee/DietPi/commit/8443d01a5e63d81d16ced06047315c0b9f17b55f

Now in final testing phase.

Fourdee, 2016-05-15:
Resolved C2 failed install. Was missing package ```automake```

Fourdee, 2016-05-15:
Server tests. Running well.
```
root@DietPi:/etc/openbazaar-server# python openbazaard.py start -a 0.0.0.0
________                      __________
\_____  \ ______   ____   ____\______   \_____  _____________  _____ _______
 /   |   \\____ \_/ __ \ /    \|    |  _/\__  \ \___   /\__  \ \__  \\_  __ \
/    |    \  |_> >  ___/|   |  \    |   \ / __ \_/    /  / __ \_/ __ \|  | \/
\_______  /   __/ \___  >___|  /______  /(____  /_____ \(____  (____  /__|
        \/|__|        \/     \/       \/      \/      \/     \/     \/

OpenBazaar Server v0.1 starting...
Exception : No UPnP device discovered
2016-05-15 16:20:20+0100 [OpenBazaard] [INFO] Finding NAT Type...
2016-05-15 16:20:35+0100 [OpenBazaard] [INFO] Restric NAT on 82.44.22.31:18467
2016-05-15 16:20:36+0100 [OpenBazaard] [INFO] startup took 19.59 seconds
2016-05-15 16:20:36+0100 [Server] [INFO] bootstrapping with 20 addresses, findin                                                                 g neighbors...
2016-05-15 16:20:37+0100 [ConnHandler] [INFO] connected to 176.9.145.230:40001
2016-05-15 16:20:37+0100 [ConnHandler] [INFO] connected to 178.82.214.172:35783
2016-05-15 16:20:37+0100 [ConnHandler] [INFO] connected to 94.23.252.118:18467
2016-05-15 16:20:37+0100 [ConnHandler] [INFO] connected to 84.72.86.75:18467
2016-05-15 16:20:37+0100 [ConnHandler] [INFO] connected to 149.56.89.140:19167
2016-05-15 16:20:37+0100 [ConnHandler] [INFO] connected to 95.183.53.195:31011
2016-05-15 16:20:37+0100 [ConnHandler] [INFO] connected to 95.183.53.195:31009
2016-05-15 16:20:37+0100 [ConnHandler] [INFO] connected to 149.56.89.140:19567
2016-05-15 16:20:37+0100 [ConnHandler] [INFO] connected to 188.62.44.189:39278
2016-05-15 16:20:37+0100 [ConnHandler] [INFO] connected to 149.56.89.140:19367
2016-05-15 16:20:37+0100 [ConnHandler] [INFO] connected to 98.253.69.210:30818
2016-05-15 16:20:37+0100 [ConnHandler] [INFO] connected to 23.120.208.254:39636
2016-05-15 16:20:37+0100 [ConnHandler] [INFO] connected to 192.241.236.11:18467
2016-05-15 16:20:37+0100 [ConnHandler] [INFO] connected to 54.255.176.19:18467
2016-05-15 16:20:37+0100 [ConnHandler] [INFO] connection with 149.56.89.140:1936                                                                 7 terminated
2016-05-15 16:20:37+0100 [ConnHandler] [INFO] connected to 106.44.69.25:1843
2016-05-15 16:20:38+0100 [ConnHandler] [INFO] connected to 62.195.114.100:18467
2016-05-15 16:20:38+0100 [ConnHandler] [INFO] connection with 84.72.86.75:18467                                                                  terminated
2016-05-15 16:20:39+0100 [-] [INFO] Libbitcoin server online at tcp://libbitcoin                                                                 3.openbazaar.org:9091
2016-05-15 16:20:41+0100 [OpenBazaard] [INFO] bootstrap complete
2016-05-15 16:20:42+0100 [ConnHandler] [INFO] connected to 86.160.64.169:39260
2016-05-15 16:20:42+0100 [ConnHandler] [INFO] connected to 95.183.53.195:31003
2016-05-15 16:20:42+0100 [ConnHandler] [INFO] connected to 45.55.137.42:18467
```

Fourdee, 2016-05-15:
Documentation completed: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=1796#p1796

Fourdee, 2016-05-15:
Completed. Many thanks to @swordfish6975 for the initial installation code :+1: 

@swordfish6975 
If you want to test this prior to v118 release, please see: https://github.com/Fourdee/DietPi/blob/master/TESTING-BRANCH.md

Fourdee, 2016-05-15:
As OpenBazaar has no options for logging to a file: https://github.com/OpenBazaar/OpenBazaar-Server#usage, attempting to redirect program to file.

#### Daemon

Gets stuck in starting loop:
```
python openbazaard.py start -d -a 0.0.0.0 &> /var/log/openbazaar.log
```
[image]

#### Non-Daemon

Can not be stopped with ```python openbazaard.py stop```:
```
python openbazaard.py start -a 0.0.0.0 &> /var/log/openbazaar.log &
```

Fourdee, 2016-05-15:
Cant see a way round this for now. Users will simply have to run it as a foreground process for debugging.
I've updated documentation to explain how to achieve this: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=1796#p1796

Fourdee, 2016-05-15:
For the sake of good records lol. Commit: https://github.com/Fourdee/DietPi/commit/9457b34d27eb53b621e8a5fdeb7f8da97e02e9ee

closed 2016-05-15 by Fourdee